Paciente8159/uCNC #35: Can I send a compiled F103C8T6 HEX file? Email 459429772@qq.com

hankecnc, 2021-05-22:
Nǐ hǎo, wǒ láizì zhōngguó, shì yī míng cnc àihào zhě, běnzhí gōngzuò shì jiāgōng zhōngxīn shīfù,cnc shì yèyú àihào, wǒ zhīqián shǐyòng de shì láizì zhège gùjiàn https://Github.Com/robomechs/6-AXIS-USBCNC-GRBL, jīntiān kàn dàole nín de yuánmǎ, wǒ jiāng yuánmǎ xiàzài dào linux xìtǒng zhōng, ānzhuāng arm-none gcc, bìng shǐyòng make mìnglìng jìnxíng biānyì, dànshì xiàzài hòu, tōngguò ch340g chuànkǒu mókuài liánjiē diànnǎo, fāsòng $$ háo wú fǎnyìng, suíhòu wǒ chángshì zài boardmap_bluepill.H wénjiàn zhōng fēnbié qǔxiāo zhùshì//Setup COM pins (must match selected COM port) //#define RX_BIT 10 //#define TX_BIT 9 //#define RX_PORT A //#define TX_PORT A hé//#define COM_PORT 1 //#define TX_BIT 9 //#define TX_PORT A //#define RX_BIT 10 //#define RX_PORT A biānyì xiàzài hòu dōu méiyǒu fǎnyìng, wǒ bùshì hěn dǒng C++hé dānpiànjī zhī lèi de zhīshì, néng fǒu fāsòng wǒ yī fèn F103C8T6 de HEX wénjiàn dào 459429772@qq.Com, bìng gàozhī tōngxùn fāngshì shì chuànkǒu háishì USB xūnǐ chuànkǒu
展开
550 / 5000
翻译结果
Hello, I am from China, I am a cnc enthusiast, my job is a machining center master, cnc is a hobby, I used this firmware from https://github.com/robomechs/6-AXIS-USBCNC- GRBL, I saw your source code today. I downloaded the source code to the linux system, installed arm-none gcc, and used the make command to compile, but after downloading, I connected the computer through the ch340g serial port module and sent $$ without any response, and then I tried to uncomment them separately in the boardmap_bluepill.h file//Setup COM pins (must match selected COM port)
//#define RX_BIT 10
//#define TX_BIT 9
//#define RX_PORT A
//#define TX_PORT A

And //#define COM_PORT 1
//#define TX_BIT 9
//#define TX_PORT A
//#define RX_BIT 10
//#define RX_PORT A
There is no response after compiling and downloading. I don’t know much about C++ and microcontrollers. Can you send me a HEX file of F103C8T6 to 459429772@qq.com, and tell me whether the communication method is serial or USB virtual serial

Paciente8159, 2021-05-23:
I there. I just compiled the default version of µCNC (3 axis cartesian machine on bluepill with USB virtual COM port).
Try it and tell me if it works..

hankecnc, 2021-05-23:
[image]
 
I did not see any files available for download on this page,

I used this firmware earlier https://github.com/robomechs/6-AXIS-USBCNC-GRBL He works very well, I would like to try your firmware, but the process is not smooth,

Paciente8159, 2021-05-23:
I sent it to your e-mail. I've also added the compiled hex in the [release v1.1.0](https://github.com/Paciente8159/uCNC/releases/tag/v1.1.0).
Thanks.

hankecnc, 2021-05-24:
Thank you, I downloaded the uCNC-bluepill-usb.hex file of this page https://github.com/Paciente8159/uCNC/releases/tag/v1.1.0 to the blue f103c8t6 development board, connected via USB, the computer can The STM virtual serial port was recognized, but I did not get any reply when sending the $$ command through the serial port software, and then tried to connect the serial port PA9 and PA10 through the CH340 module, but there was no reply.
[image]

hankecnc, 2021-05-24:
Nothing happens when using other grbl-related software such as Candle

Paciente8159, 2021-05-24:
This compilation uses the bluebill built in USB. Using and external serial cable doesn't work.
Also try this:
Short PINS B5 and B6 to ground before plugin in the USB cable. These pins are the hard reset and feed hold (negative logic). If left ungrounded the uCNC will be in failsafe mode and will not respond to any command.

hankecnc, 2021-05-24:
Thank you very much for your patient guidance, I will follow your prompts
PINS B5 and B6 are grounded and have gotten a reply, thank you very much

Paciente8159, 2021-05-24:
Your welcome. And thanks for trying out µCNC.
Any problem please report and I'll get to it ASAP.
I'll keep this issue open for a few more days in case you have any doubt.

hankecnc, 2021-05-24:
I am willing to try new things.
What host computer do you use to operate uCNC?

Although uCNC can connect to the serial port software normally, it cannot use Candle2 https://github.com/Schildkroet/Candle2 The host computer communication on this page, click on the connection, no response, another laser drawing software is also, I guess whether it is a handshake protocol Has it changed? Because I am not good at programming, I can't solve this problem. Candle2 is excellent as a grbl host computer, and I don't want to give it up.
[image]

Paciente8159, 2021-05-24:
I'm using OpenCNCPilot. I like it alot and it has a very clean and simple interface.
I'll try Candle2 to see if I have any problems.
[image]

Paciente8159, 2021-05-24:
I've tested Candle2 and the original Candle. It seems to have been locked to use Grbl firmware. If I modify µCNC startup message to output:
`Grbl 1.1.1 ['$' for help]`
Candle2 and the original Candle both start communicating with µCNC.
To modify the startup message go to the grbl_interface.h file and modify the MSG_STARTUP_START from
`#define MSG_STARTUP_START "uCNC "`
to
`#define MSG_STARTUP_START "Grbl "`
Recompile the code and upload and it starts working.

Paciente8159, 2021-05-28:
Since the problem was solved I'm closing the issue.
In a near future release an option to emulate Grbl startup message will be added to improve software compatibility.

closed 2021-05-28 by Paciente8159