Paciente8159/uCNC #171: Is your F4 a F401CCU6? How much Mhz crystal is he? Your firmware usb doesn't work

hankecnc, 2022-04-22:
The program seems to be running? The PC13 light is on, but it is an unknown USB device. It is certain that there is no problem with my usb virtual serial port (I have tried grblhal's f401 firmware and he has no problem)

hankecnc, 2022-04-22:
[image]

hankecnc, 2022-04-22:
I don't have the f103c8t6 board to test anymore

Paciente8159, 2022-04-22:
How did you compiled it?
Arduino IDE, makefile or PlatformIO?

Paciente8159, 2022-04-22:
In about an hour I'll be able to test it and let you know. I think it is 25Mhz.

Paciente8159, 2022-04-22:
I've just uploaded the bin that is precompiled in github and it's working. Even without Pin B5 grounded.
I used st-link to upload the firmware. What did you use?
Here is a pic. I confirm it's a 401cdu6 at 25mhz
[image]

Paciente8159, 2022-04-22:
Also tried with STM32CubeProgrammer in DFU mode and again all worked.

hankecnc, 2022-04-22:
The led of PC13 is only on, not flashing, I try to burn the grblhal version again, his usb virtual serial port works fine, you have noticed that your chip is CDU6 instead of CCU6, I don't know what will be the difference, just flash capacity
[image]
?

hankecnc, 2022-04-22:
[image]

hankecnc, 2022-04-22:
I bought a f103c8t6 today, it's on the way, it will arrive the day after tomorrow

Paciente8159, 2022-04-23:
what did you use to upload the firmware?

hankecnc, 2022-04-23:
> what did you use to upload the firmware? If the led is not blinking that is an indication that the firmware is not running somehow.

Tried both with usb dfu and serial port with STM32CubeProgrammer

hankecnc, 2022-04-23:
> what did you use to upload the firmware? If the led is not blinking that is an indication that the firmware is not running somehow.

I have a little knowledge of microcontrollers, the PC13 LED is on means it is indeed running

hankecnc, 2022-04-23:
> what did you use to upload the firmware? If the led is not blinking that is an indication that the firmware is not running somehow.

In your program, does the PC13 LED light flash as long as it is plugged in without any communication?

Paciente8159, 2022-04-23:
The PC13 blinks because there is an internal function that runs every millisecond.

Can you try the alternative method via Arduino IDE?
[image]
[image]

Paciente8159, 2022-04-23:
If you don't have the STM32duino installed please read the [wiki tutorial](https://github.com/Paciente8159/uCNC/wiki/STM32Fx:-Compiling#method-two---arduino-ide-easiest)

hankecnc, 2022-04-23:
> If you don't have the STM32duino installed for Arduino IDE please read the [wiki tutorial](https://github.com/Paciente8159/uCNC/wiki/STM32Fx:-Compiling#method-two---arduino-ide-easiest)

Thanks, I will try again when my C8T6 arrives

closed 2022-04-23 by hankecnc

hankecnc, 2022-04-26:
> How did you compiled it? Arduino IDE, makefile or PlatformIO or used the precompiled file? I don't have my board with me right now but I will confirm it in couple of hours.
> 
> By the way... have you tried grounding PIN B5? That's the Emergency stop. If left unconnected it will lock the firmware for safety reasons. That can be modified via config file.

Hello, after experiments, the firmware burning of the F103c8t6 version I purchased works normally

hankecnc, 2022-04-26:
Depending on how you keep updating it, I'll try his

hankecnc, 2022-04-26:
I'll try him on my machine and see how he performs

Paciente8159, 2022-04-26:
> I'll try him on my machine and see how he performs

Thanks let me know if you find something off and I'll try to fix it ASAP.