Paciente8159/uCNC #172: Post Processor for uCNC

rehanzfr, 2022-04-26:
Hi,

I just reached this marvelous software. I am bit new in this field. Can you please guide about the post-processor of uCNC. I am using freecad. 

Best regards,

Paciente8159, 2022-04-26:
Hi there. µCNC was build to be compatible with Grbl protocol. You can use Grbl post-processing to generate your GCode.
µCNC also supports a subset of canned cycles (not supported by Grbl), but this feature (that can be enabled via config file) will not fit on a regular Arduino UNO board.

Paciente8159, 2022-04-26:
To complete my previous answer...µCNC also tries has much as possible to be compliant with RS274NGC version 3 standard (available in the docs folder).
Some additional custom codes are also available via config file by enabling parser extensions. These extended GCodes try in some way be similar to existing GCodes in other software/firmware (like for example LinuxCNC and Marlin).

But the original set of GCode commands supported by Grbl 1.1f is supported by µCNC as well as most of the system commands and remaining protocol commands (real-time, override, jog, etc...)

rehanzfr, 2022-04-26:
Thank you very much for detailed answers.

closed 2022-04-26 by rehanzfr