Paciente8159/uCNC #173: Dual Y Axis (One motor is working where other is not)

rehanzfr, 2022-04-27:
Hi, 

I am new in this field and I think reached to my last question. I have connected two motors on my Y-axis. In marlin firmware (on MKS Gen L V 2.1 Board = Ramps Board in pinouts (may be)), when updated for dual axis on Y axis, the  machine run smoothly. However, I could not get the Y-Axis running with both the motors (though one is running and one is not) on GRBL or uCNC. Can you please guide me in this aspect. 

Best regards,

Paciente8159, 2022-04-27:
Sure. Just need some details to better aid you.

How many dual axis are you going to run? Is it only for Y? Do you have a second Y end-stop to go with it? This is for self squaring machine right?

Paciente8159, 2022-04-27:
Ok try this.
Open src/cnc_config.h
And modify the dual axis section to this:

```
#define ENABLE_DUAL_DRIVE_AXIS
#ifdef ENABLE_DUAL_DRIVE_AXIS
// defines the first dual drive capable step output
#define DUAL_DRIVE_AXIS0 Y
//#define DUAL_DRIVE_AXIS1 Y
#endif
```

Then go to the boards ramps file in src/hal/boards/avr/boardmap_ramps14.h and modify STEPPER4 pins to STEPPER6 like this:

Line 37
```
// Setup step pins
#define STEP6_BIT 1	 // assigns STEP6 pin
#define STEP6_PORT C // assigns STEP6 port
//#define STEP4_BIT 1	 // assigns STEP4 pin
//#define STEP4_PORT C // assigns STEP4 port
...
```

Line49
```
// Setup dir pins
#define DIR6_BIT 3	// assigns DIR6 pin
#define DIR6_PORT C // assigns DIR6 port
//#define DIR4_BIT 3	// assigns DIR4 pin
//#define DIR4_PORT C // assigns DIR4 port
```

Line 98
```
// Stepper enable pin. For Grbl on Uno board a single pin is used
...
#define STEP6_EN_BIT 7
#define STEP6_EN_PORT C
//#define STEP4_EN_BIT 7
//#define STEP4_EN_PORT C
```

now Add the Y2 limit switch like this
```
// Setup limit pins
#define LIMIT_Z_BIT 3  // assigns LIMIT_Z pin
#define LIMIT_Y_BIT 1  // assigns LIMIT_Y pin
#define LIMIT_X_BIT 5  // assigns LIMIT_X pin
#define LIMIT_Z_PORT D // assigns LIMIT_Z port
#define LIMIT_Y_PORT J // assigns LIMIT_Y port
#define LIMIT_X_PORT E // assigns LIMIT_X port
#define LIMIT_Z_ISR -4 // assigns LIMIT_Z ISR
#define LIMIT_Y_ISR 1  // assigns LIMIT_Y ISR
#define LIMIT_X_ISR -6 // assigns LIMIT_X ISR
//limit switch Y2
#define LIMIT_Y2_BIT 0  // assigns LIMIT_Y2 pin
#define LIMIT_Y2_PORT J // assigns LIMIT_Y2 port
#define LIMIT_Y2_ISR 1 // assigns LIMIT_Y2 ISR
```

Compile and try. Do a small motion in Y axis before doing a home or full motion range to confirm both steppers are going in the same direction. If not turn off the board and invert one of the motors connections. Also test both switches to check that Y limit is triggered (in the status message).

I have to improve the configuration files to make them easier to customize and I will address that in the upcoming updates.
If you have any problem with this config let me know and I'll check it.

rehanzfr, 2022-04-27:
Thank you very much. I will try it tonight and will detail here the complete notes.

jimsynz, 2022-04-28:
Ahh. I have been battling with this also, I have a branch of my configuration here: https://github.com/jimsynz/uCNC/tree/add-mks-gen-l

If I am following correctly I need to set up my board map to thing that STEP6 and STEP7 have the config of STEP4 and STEP5?

Paciente8159, 2022-04-28:
Yes. I'm sorry. Some parts of the firmware are still a bit obscure due to the lack of time to update the wiki.

@rehanzfr please read the note below regarding step enable pins.

If you are using 2 dual axis (lets say X and Y) then yes. Configure Step6 and 7 shadow registers (in this case if you are using them in the place of step3 and 4 then rename them.
You will need an additional setting for X2 endstop that is not defined in the ramps board.
If using dual axis in x and Y you will have to set it up like this:

Open src/cnc_config.h
And modify the dual axis section to this:

```
#define ENABLE_DUAL_DRIVE_AXIS
#ifdef ENABLE_DUAL_DRIVE_AXIS
// defines the first dual drive capable step output
#define DUAL_DRIVE_AXIS0 X
#define DUAL_DRIVE_AXIS1 Y
#endif
```

Then go to the boards ramps file in src/hal/boards/avr/boardmap_ramps14.h and modify STEPPER4 pins to STEPPER6 like this:

Line 37
```
// Setup step pins
// Setup step pins
#define STEP7_BIT 1	 // assigns STEP4 pin
#define STEP7_PORT C // assigns STEP4 port
#define STEP6_BIT 4	 // assigns STEP3 pin
#define STEP6_PORT A // assigns STEP3 port
#define STEP2_BIT 3	 // assigns STEP2 pin
#define STEP2_PORT L // assigns STEP2 port
#define STEP1_BIT 6	 // assigns STEP1 pin
#define STEP1_PORT F // assigns STEP1 port
#define STEP0_BIT 0	 // assigns STEP0 pin
#define STEP0_PORT F // assigns STEP0 port
```

Line49
```
// Setup dir pins
#define DIR7_BIT 3	// assigns DIR2 pin
#define DIR7_PORT C // assigns DIR2 port
#define DIR6_BIT 6	// assigns DIR2 pin
#define DIR6_PORT A // assigns DIR2 port
#define DIR2_BIT 1	// assigns DIR2 pin
#define DIR2_PORT L // assigns DIR2 port
#define DIR1_BIT 7	// assigns DIR1 pin
#define DIR1_PORT F // assigns DIR1 port
#define DIR0_BIT 1	// assigns DIR0 pin
#define DIR0_PORT F // assigns DIR0 port
```

**DO NOT CHANGE STEP ENABLE PINS**
There is a bug in the firmware I just noticed now. STEP6_EN and STEP7_EN are not defined the HAL. Keep this as it is and it should work never the less.
```
// Stepper enable pin. For Grbl on Uno board a single pin is used
...
#define STEP0_EN_BIT 7
#define STEP0_EN_PORT D
#define STEP1_EN_BIT 2
#define STEP1_EN_PORT F
#define STEP2_EN_BIT 0
#define STEP2_EN_PORT K
#define STEP3_EN_BIT 2
#define STEP3_EN_PORT A
#define STEP4_EN_BIT 7
#define STEP4_EN_PORT C
```

now Add the Y2 limit switch like this
```
// Setup limit pins
#define LIMIT_Z_BIT 3  // assigns LIMIT_Z pin
#define LIMIT_Y_BIT 1  // assigns LIMIT_Y pin
#define LIMIT_X_BIT 5  // assigns LIMIT_X pin
#define LIMIT_Z_PORT D // assigns LIMIT_Z port
#define LIMIT_Y_PORT J // assigns LIMIT_Y port
#define LIMIT_X_PORT E // assigns LIMIT_X port
#define LIMIT_Z_ISR -4 // assigns LIMIT_Z ISR
#define LIMIT_Y_ISR 1  // assigns LIMIT_Y ISR
#define LIMIT_X_ISR -6 // assigns LIMIT_X ISR
//limit switch X2
#define LIMIT_X2_BIT 4  // assigns LIMIT_X2 pin
#define LIMIT_X2_PORT E // assigns LIMIT_X2 port
#define LIMIT_X2_ISR -3 // assigns LIMIT_X2 ISR
//limit switch Y2
#define LIMIT_Y2_BIT 0  // assigns LIMIT_Y2 pin
#define LIMIT_Y2_PORT J // assigns LIMIT_Y2 port
#define LIMIT_Y2_ISR 1 // assigns LIMIT_Y2 ISR
```

rehanzfr, 2022-04-28:
Very fruitful discussion. I will try it and mention all the steps.

Paciente8159, 2022-04-28:
@rehanzfr @jimsynz

I've looked at the code and there is also the same issue replicated in the DIR pins.
The issue is that all the initial testings for the dual axis were developed around the UNO board. In the UNO board there is only a single enable pin for all axis. Also for pin economy the same dir pin was shared by both dual drived axis. On RAMPS or MKS GEN L this is not the case.

I'm addressing this issue and fixing the code that I thing I'll be able to push later today.
So please hold the testings for a bit while I address this issue. Thank you for your patience.

rehanzfr, 2022-04-28:
Everything went well. Y axis with dual motors are working fine. 

What I did, as per suggestions (for other users as well):

1- Copy the new branch (https://github.com/Paciente8159/uCNC/tree/dual-axis-new-config) to my desktop. 
2- Opened the file boarddefs.h (from location C:\Users\rehan\Desktop\uCNC-dual-axis-new-config\uCNC\src\hal\boards) and copy the board name that is "BOARD_RAMPS14".
3- Opened the file "cnc_config.h" in main folder and update the following code where "BOARD_RAMPS14" is updated: 

`#ifndef BOARD
#define BOARD BOARD_RAMPS14
#endif` 

4- Opened the file "cnc_hal_config.h" and uncomment the following line at first
`#define LIMIT_X2_DISABLE`
however, it gave me errors during compiling. Therefore, I commented it again. 
5- Later I updated the following code in the same file:

`#define ENABLE_DUAL_DRIVE_AXIS
#ifdef ENABLE_DUAL_DRIVE_AXIS
// defines the first dual drive capable axis
#define DUAL_DRIVE0_AXIS Y
// by default this will be rewired to STEPPER6 (if available on the board)
// this can be uncommented to re-wire to an available (unused stepper other then 6)
#define DUAL_DRIVE0_STEPPER 4`

6- It got compiled and now it is running perfectly. 

7- Tried same on MKS Gen L V2,1 where the second Y stepper motor is attached to E1. It is running smottly there as well. 

8- In CNCJS, however, I am getting the "error:8 not idle", which, I will look into. 

Thank you again for explanation and updates in the complete branch. If you suggest, let's close this issue.

Paciente8159, 2022-04-28:
Thanks for the info and testing. Did you try homing with this setup?

jimsynz, 2022-04-28:
Thanks @Paciente8159 I'll give it a go today and report back.

rehanzfr, 2022-04-28:
Let me do that as well.

rehanzfr, 2022-04-28:
I am using Arduino IDE :)

Paciente8159, 2022-04-28:
> 4- Opened the file "cnc_hal_config.h" and uncomment the following line at firs
#define LIMIT_X2_DISABLE
However, it gave me errors during compiling. É Therefore, I commented it again.

I'll also check this before merging

Paciente8159, 2022-04-28:
> 4- Opened the file "cnc_hal_config.h" and uncomment the following line at first
> `#define LIMIT_X2_DISABLE`
> However, it gave me errors during compiling. Therefore, I commented it agai

@rehanzfr  I've fixed this one. You have to enable it or else you will probably have errors homing because X2 will trigger the X limit-switch if left unconected.

rehanzfr, 2022-04-28:
Ok, in the previous branch my homing was not working.

jimsynz, 2022-04-28:
Hey @Paciente8159 I checked out your branch and gave it a go. I have X2 on STEP3 and Y2 on STEP4.  X2 is working, although I need to figure out how to reverse X2's direction compared to X. Y2 doesn't seem to be working. I'm using TMC2208 drivers on all 5 steppers, and I enabled the TMC4 UART in the ramps board file.

rehanzfr, 2022-04-28:
@Paciente8159 yes it is good now with enabling the following line. 

`#define LIMIT_X2_DISABLE`

However, I cannot still home. On pressing Homing button on CNCJS or $H on terminal, it is saying about ALARM:11, is it on my side with machine?

Paciente8159, 2022-04-28:
Ok 2 notes on that.

Since you are using STEP3 to drive X2 you can reverse the stepper direction via setting $3. If you have trouble with this let me know and I'll try to detail it.

About the TMC drivers. You will need to have a serial Y cable for each driver. The pinout is similar to Marlin 1.1.9.
In Marlin these are the pins to UART with TMC
```
    #define X_SERIAL_TX_PIN  40 //pin G1
    #define X_SERIAL_RX_PIN  63 //pin K1
    #define Y_SERIAL_TX_PIN  59 //pin F5
    #define Y_SERIAL_RX_PIN  64 //pin K2
    #define Z_SERIAL_TX_PIN  42 //pin L7
    #define Z_SERIAL_RX_PIN  65 //pin K3
    #define E0_SERIAL_TX_PIN 44 //pin L5
    #define E0_SERIAL_RX_PIN 66 //pin K4
    #define E1_SERIAL_TX_PIN -1 //undefined
    #define E1_SERIAL_RX_PIN -1 //undefined
```

As you can see for Marlin E1 is not defined so you need to choose a set of pins.

For µCNC the configured pins are the same (in boardmap_ramps14.h):
```
// TMC0 UART
#define DOUT23_BIT 1
#define DOUT23_PORT G
#define DIN23_BIT 1
#define DIN23_PORT K
#define DIN23_PULLUP

// TMC1 UART
#define DOUT24_BIT 5
#define DOUT24_PORT F
#define DIN24_BIT 2
#define DIN24_PORT K
#define DIN24_PULLUP

// TMC2 UART
#define DOUT25_BIT 7
#define DOUT25_PORT L
#define DIN25_BIT 3
#define DIN25_PORT K
#define DIN25_PULLUP

// TMC3 UART
#define DOUT26_BIT 5
#define DOUT26_PORT L
#define DIN26_BIT 4
#define DIN26_PORT K
#define DIN26_PULLUP

// // TMC4 UART
// #define DOUT27_BIT 3
// #define DOUT27_PORT F
// #define DIN27_BIT 4
// #define DIN27_PORT F
// #define DIN27_PULLUP
```

If you have AUX 1 pins free on your MKS GEN L then you are in luck. Just uncomment those pins for TMC4 UART. Pin F3 (A3 on ramps) will be your TX and F4 (A4) will be your RX.

On cnc_hal_config.h don't forget to uncomment each TMC stepper you want to enable

This enables TMC0 for example
`#define STEPPER0_HAS_TMC`
Do the same for the others

Also on cnc_config.h on line 206 uncomment this to unlock all extended G and M codes (including the ones for the TMC drivers)

`#define ENABLE_PARSER_MODULES`

The codes almost similar to Marlin. 

- M350 will enable you to print or set the microstepping for each driver
- M906 will enable you to print or set the stepper current for each driver
- M920 will enable you to print or set any TMC register directly for each driver (be careful with this one)

Paciente8159, 2022-04-28:
> @Paciente8159 yes it is good now with enabling the following line.
> 
> `#define LIMIT_X2_DISABLE`
> 
> However, I cannot still home. On pressing Homing button on CNCJS or $H on terminal, it is saying about ALARM:11, is it on my side with machine?

@rehanzfr Alarm11 is Homing fail. Limit switch is already active. Try reposition the machine, or check wiring.

That means that you have and active/triggered switch. Send a `?` and check which in the Pn: portion of the status message.
Before homing no limit switch must be active. I'm guessing you may need to invert/mask some limit switches with setting [$5](https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#5----limit-pins-invert-mask).

Do you use NO or NC switches? without any switch pressed and all switches connected when you send a `?` what is the status report that is printed?

jimsynz, 2022-04-29:
Thanks for the replies @Paciente8159.  

I just realised that I went down a useless rabbit hole with the TMC stuff.  I'm just using them in "legacy mode", so basically just as A4988's. I've removed the TMC-related config.  There's definitely still something up though, because I have the following behaviour:

* G91 X1 steps X1 and X2 correctly
* G91 X-1 steps X1 and X2 correctly.
* G91 Y1 steps Y1 correctly, Y2 doesn't move.
* G91 Y-1 steps Y1 correctly, Y2 doesn't move.
* G91 Z1 steps X1 positive, Z doesn't move.
* G91 Z-1 steps X1 positive, Z doesn't move.

This is all deeply confusing, so I reflashed my previously-working Marlin config onto it, and the board appears to still be working fine. 

I've rebased my branch off yours and pushed my current configuration to https://github.com/jimsynz/uCNC/tree/add-mks-gen-l 
I'm building and flashing with platformio and with the `mks_gen_l` target that I've added.

Paciente8159, 2022-04-29:
2:00AM saw your message. Rolling in bed...AHHHHH lolol. I just can't keep it out of my mind.

Fixed the issue. A typo in some macros were the issue. I tested each motor individually in a RAMPS board and on my oscilloscope.
Please retry now.

@rehanzfr you should redownload the firmware too. Since this had a bug.

@jimsynz just a side note. On your mks gen if you put the `#define BOARD_NAME "MKS GEN L"` before `#include "boardmap_ramps14.h"` you don't need to do the `#undef BOARD_NAME`.
Also it's recommended to pass the `#include "boardmap_ramps14.h"` inside the extern "C" like this
You'll only need the `#include "mega_ide_pins.h"` if you want to configure pins using Arduino IDE pin identification.
I usually use the MCU PORT and PIN ID style

```
#ifndef BOARDMAP_MKS_GEN_L_H
#define BOARDMAP_MKS_GEN_L_H

#ifdef __cplusplus
extern "C"
{
#endif

//you can add extra customizations here
#define BOARD_NAME "MKS GEN L"
// MKS Gen L is based on Ramps 1.4.
#include "boardmap_ramps14.h"

#ifdef __cplusplus
}
#endif

#endif
```

In alternative if you only want to change the board name you can do it in the cnc_config.h

3:52AM...now I can sleep in peace ;-P

jimsynz, 2022-04-29:
Thanks @Paciente8159 - you really should try sleeping more :smiley: 

I'll give it a test tonight.

jimsynz, 2022-04-29:
I think you may have forgotten to push you changes to that branch. All good mate. I’ll check in again tomorrow.

Paciente8159, 2022-04-29:
Your right. My net connection failed for a brief momento and it got stucked. It's there now.

Thanks

jimsynz, 2022-04-29:
@Paciente8159 my machine (MPCNC Primo) doesn’t have end stops on the Z axis so I typically just do `G28 X Y` to home X & Y and then either eyeball or probe the Z. Homing X & Y is important for squaring but Z isn’t super important. On my old 3018 I added end stops and homed towards Z+ just to limit crashes (it has a small ~35mm Z range). Back to your question: I disabled the end Z end stop pins because I was planning on using the Zmin pin for my touch probe.

Paciente8159, 2022-04-29:
@jimsynz G28 is a Marlin command. On Grbl type firmwares (including µCNC) and in the RS274NGC standard it works in a different way.

Homing is done with $H. It performs on all axis not only specified ones.

I still have to move some code around and add additional pre-processor conditional checkings, (and I will take care of that issue) but for now just do this:

open file /src/hal/kinematics/kinematics_cartesian.c

on line 50 comment this whole section of code

```
#ifdef AXIS_Z
        result = mc_home_axis(AXIS_Z, LIMIT_Z_MASK);
        if (result != 0)
        {
                return result;
        }
#endif
```

this will skip Z homing and do the remaining.

jimsynz, 2022-04-29:
Thanks for that. I’ll give it a go. I’ve just been reading threads on adding a Zmax end stop. Just another rabbit hole!

Paciente8159, 2022-04-29:
Lol those rabbits are sneaky.
Before chasing them and if you need to do something that may not be clear at first glance you can ask me and I'll tell you if there is a work around or not.

If you also require any custom/special application case I can see and help if µCNC can be modified to fit the needs.

rehanzfr, 2022-04-29:
@Paciente8159 Homing is not working. No motors are moving. Rest is fine with dual axis. Please guide in this aspect.

Paciente8159, 2022-04-29:
@rehanzfr I will need to things to try to understand.

- When you send a `?` what does the board reports
- Also send me your current settings please (command `$$`)

Paciente8159, 2022-04-29:
By the way be aware that I've pushed yet another big code revision to this branch.
@jimsynz this revision will make Z axis homing be skipped if no Z axis endstops are present (no need to change the kinematics this time)

rehanzfr, 2022-04-29:
Sure, after light will come, I will try the same on machine again. I am using the same branch described above.  

? is giving <Idle|MPos:0.000,0.000,0.000|FS:0.000,0|Pn:P

$$ is giving:

$0=33.333 (Step pulse time, microseconds)
$1=0 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$7=0
$10=1 (Status report options, mask)
$11=0.200 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=10.000 (Homing locate feed rate, mm/min)
$25=50.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=1000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$80=1
$81=0.000
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)

rehanzfr, 2022-04-29:
@Paciente8159 Can we make a video series on the complete tutorial and advanced changes required to be done with information on pins and changes? I can help in that for uCNC.

Paciente8159, 2022-04-29:
The thought of doing a series of videos on configuring aspects of the firmware has come to my mind. I will do that for sure in a very near future. But if you want to do it sure all help is welcomed.

OK I see that no limit switches are triggered so that's good.
I'll put a series of steps here with questions for you to answer please.

Let's start with endstops.

- press the X endstop and send `?`. what did it print?
- press the Y endstop and send `?`. what did it print?
- press the Y2 endstop and send `?`. what did it print?
- press the Z endstop and send `?`. what did it print?
- Is your Z endstop on top or on bottom?

Next lets look at your configs.

- Are your steps for mm those?
- If not reconfigure $100, $101 and $102.

Next motion

- Send a G0X1. This the X axis moved? Did it go the positive direction?
- Send a G0Y1. This both Y axis moved? Did they go the positive direction?
- Send a G0Z1. This both Z axis moved? Did it go the positive (up) direction?

Please answer these and we will go from there.

Paciente8159, 2022-04-29:
> By the way be aware that I've pushed yet another big code revision to this branch.
> @jimsynz this revision will make Z axis homing be skipped if no Z axis endstops are present (no need to change the kinematics this time)

When I mentioned this is that there is a risk of something get broken in the way. I'm currently also running tests with similar config to yours to see steppers pins are working properly.

rehanzfr, 2022-04-29:
Ok I will report all asap.

Paciente8159, 2022-04-29:
@rehanzfr Your config works on my atmega simulator doing a homing.
@jimsynz found one small bug that cause the second dual axis to issue an error during home when reaching the switch. I'll test your config but I think it will work too.

Paciente8159, 2022-04-29:
@jimsynz after all found yet another bug. It's fixed and patched. Also removed some deprecated code.
Tested homing on simulator with your settings. Homing was successful.

jimsynz, 2022-04-29:
Just pulled your changes and uploading to my board. I will test this AM.

jimsynz, 2022-04-29:
@Paciente8159 thanks so much! This is so much better. It is now correctly stepping on all axis! I tried to home it, however the machine crashed.  It triggered the limit switches (sending `?` confirmed it) however it just kept on going. Any ideas?

Paciente8159, 2022-04-30:
@jimsynz please run the steps I asked above.
In your case:
What are your settings `$$`? 
Let's start with endstops.

press the X(keep it pressed) endstop and send ?. what did it print?
press the X2 (keep it pressed) endstop and send ?. what did it print?
press both X and X2 (keep it pressed) endstops and send ?. what did it print?
press the Y(keep it pressed) endstop and send ?. what did it print?
press the Y2 (keep it pressed) endstop and send ?. what did it print?
press both Y and Y2 (keep it pressed) endstops and send ?. what did it print?

Next motion

Send a G0X1. This the X axis moved? Did it go the positive direction?
Send a G0Y1. This both Y axis moved? Did they go the positive direction?

Please answer these and we will go from there.

Paciente8159, 2022-04-30:
@jimsynz also very important.
@rehanzfr this is also for you

Your endstop must make sure endstop X- is paired with stepper X and endstop X+ is paired with stepper E0 (paired as in the same side).
For Y the same. Y matches Y- and E1 matches Y+ or else the firmware Will stop the wrong motor while homing and hitting the first endstop

Paciente8159, 2022-04-30:
@jimsynz @rehanzfr I've even without your feedback I've repicked the homing issue and I realized there is a design flaw in my implementation for dual switch on an axis. It works fine with NC buttons like the ones I use on my CNC, for for NO if there is the need to invert the logic, not so much.

Again I'm sorry 🙏 for these setbacks. I'm working to modify the code to address this particular issue.

jimsynz, 2022-04-30:
Hey @Paciente8159 I will double check all the hoping setup tomorrow morning, but my end stops are NC.

Paciente8159, 2022-04-30:
@jimsynz Thanks. Never the less there is and issue that needs addressing. I don't think it will take me more then a day to fix it. Tomorrow will be just fine. I'll probably will have the code fixed by then.

Paciente8159, 2022-04-30:
@jimsynz @rehanzfr 
Uploaded new dual endstop check code. Tested as fas as I could. On a my ramps 1.4 board with X as in dual axis setup and 2 endstops plugged on X an X2. And all went OK.

Before doing a homing command I recommend doing the check up steps I said earlier.

Thanks for your patience and help.

Paciente8159, 2022-04-30:
Ok final modifications done. Testing all possible scenarios is very hard and a time consuming task. One apparently minor modification may have unforeseen consequences on other configs.

- Dual axis config now also detect correctly if the wrong axis switch is triggered.
- Also added check to make sure dual axis and Linear Delta machines are not enable at the same time (and Delta homing tested also due to changes in the whole endstop logic for homing).

Just a side note (later I have to update the wiki when merging the branch). All unused limits should be disabled or homing failures may occur.

rehanzfr, 2022-04-30:
Great.

Paciente8159, 2022-04-30:
@rehanzfr I've look into CNCjs not idle error and I know what is going on.
If you read [Grbl response messages](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages) error 8 states that $ commands are only allowed if Grbl is in idle state.

But apparently that is not true. It seems that at least $G command (report modal states) is allowed.
I will patch this issue once the current PR for dual axis is closed and merged.

jimsynz, 2022-05-01:
Hey @Paciente8159.  Here's the information you requested:

`$$` output:

```
$0=33.333 (Step pulse time, microseconds)
$1=0 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=10 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$7=2
$10=1 (Status report options, mask)
$11=0.200 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=90.000 (Homing locate feed rate, mm/min)
$25=180.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=1000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$80=1
$81=0.000
$100=100.000 (X-axis travel resolution, step/mm)
$101=100.000 (Y-axis travel resolution, step/mm)
$102=401.000 (Z-axis travel resolution, step/mm)
$110=1000.000 (X-axis maximum rate, mm/min)
$111=1000.000 (Y-axis maximum rate, mm/min)
$112=360.000 (Z-axis maximum rate, mm/min)
$120=180.000 (X-axis acceleration, mm/sec^2)
$121=180.000 (Y-axis acceleration, mm/sec^2)
$122=80.000 (Z-axis acceleration, mm/sec^2)
$130=590.000 (X-axis maximum travel, millimeters)
$131=390.000 (Y-axis maximum travel, millimeters)
$132=100.000 (Z-axis maximum travel, millimeters)
```

* press the X(keep it pressed) endstop and send ?. what did it print?  `Pn:X`
* press the X2 (keep it pressed) endstop and send ?. what did it print? `Pn:X`
* press both X and X2 (keep it pressed) endstops and send ?. what did it print? `Pn:X`
* press the Y(keep it pressed) endstop and send ?. what did it print? `Pn:Y`
* press the Y2 (keep it pressed) endstop and send ?. what did it print? `Pn:Y`
* press both Y and Y2 (keep it pressed) endstops and send ?. what did it print? `Pn:Y`
* Send a G0X1. This the X axis moved? Did it go the positive direction? **Yes.**
* Send a G0Y1. This both Y axis moved? Did they go the positive direction? **Yes.**

I haven't pulled your most recent branch, but I'll do that tomorrow morning and have it a go.

Thanks!

Paciente8159, 2022-05-01:
Ok @jimsynz.
Enable Hard limits $21=1. Then do a soft reset.
It Will go to alarm mode. If z is at proper height to not hit anything try send a home command $H

jimsynz, 2022-05-01:
Roger wilco.

jimsynz, 2022-05-01:
It works!  Thank you so much @Paciente8159. Now I just need to figure out how to increase the homing speed.

Paciente8159, 2022-05-01:
Settings $24 and $25 😉

Paciente8159, 2022-05-01:
Since @jimsynz confirmed on is config it's working I'll be merging this to the main branch.
@rehanzfr I'll keep this issue open while you test in your side and in case you have any doubts.

I'm not totally happy with the implementation code for this since limit pin invert setting $5 will reverse both pins of the same axis and for some reason (like using different types of switches or config) reversing X and not X2 might be required.

But I will do that on a different PR.

Thank you both.

Paciente8159, 2022-05-01:
@rehanzfr CNCjs issue fixed with #178 that is integrated in the master branch

jimsynz, 2022-05-01:
Thank you @Paciente8159. You’re awesome.

rehanzfr, 2022-05-01:
Hello @Paciente8159, I take time today and reporting these for my machine with latest master branch accessed 02/05/2022 at 12:38 PST. 

On first run, I just saw the inverted Z axis direction, when  Z- was pressed then it was going up. 

On following tests, the response is in front of these. 

**On pressing $$**

$0=33.333 (Step pulse time, microseconds)
$1=0 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$7=0
$10=1 (Status report options, mask)
$11=0.200 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=10.000 (Homing locate feed rate, mm/min)
$25=50.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=1000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$80=1
$81=0.000
$100=80.000 (X-axis travel resolution, step/mm)
$101=80.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$110=500.000 (X-axis maximum rate, mm/min)
$111=500.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$120=10.000 (X-axis acceleration, mm/sec^2)
$121=10.000 (Y-axis acceleration, mm/sec^2)
$122=10.000 (Z-axis acceleration, mm/sec^2)
$130=200.000 (X-axis maximum travel, millimeters)
$131=200.000 (Y-axis maximum travel, millimeters)
$132=200.000 (Z-axis maximum travel, millimeters)

**On pressing ?**  <Idle|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXYZ|WCO:0.000,0.000,0.000>

press the X(min) (keep it pressed) endstop and send ?. what did it print?**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PYZ|WCO:0.000,0.000,0.000>**

press the X(max) (keep it pressed) endstop and send ?. what did it print?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXYZ>**

press the Y(min) (keep it pressed) endstop and send ?. what did it print?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXYZ>**

press the Y(max) (keep it pressed) endstop and send ?. what did it print?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXYZ>**

press both Y(min) and Y2 (ymax) (keep it pressed) endstops and send ?. what did it print?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXZ>**

press the Z(min) (keep it pressed) endstop and send ?. what did it print?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:PXY>**

pressed the Z(max) (keep it pressed) endstop and sent ?
**<Alarm|MPos:-35.000,42.000,-20.000|FS:0.000,0|Pn:XYZ>**

Is your Z endstop on top or on bottom?
Both sides

Next lets look at your configs.

Are your steps for mm those?  **mm**
If not reconfigure $100, $101 and $102.  **Configured as above**
Next motion

Send a G0X1. This the X axis moved? Did it go the positive direction?  **Yes X axis moved to positive side**
Send a G0Y1. This both Y axis moved? Did they go the positive direction? "yes both Y motors moved positive if were on 0"
Send a G0Z1. This both Z axis moved? Did it go the positive (up) direction? "single Z-axis moved but opposite in direction"

Paciente8159, 2022-05-01:
Ok @rehanzfr I'll try to walk you through.

Ok. First thing. You have 2 endstops on Z but Z max is defined on your boardmap to work as your probe.
If you are using the Z+ as an actual Z limit also you need to change the boardmap.

Go to src/hal/boards/avr/boardmap_ramps14.h
on line 79 change from

```
// Setup probe pin
#define PROBE_BIT 2
#define PROBE_PORT D
#define PROBE_ISR -3
```

to

```
// Setup Z+
#define LIMIT_Z2_BIT 2
#define LIMIT_Z2_PORT D
#define LIMIT_Z2_ISR -3
```

In this case if you want to use a probe later you will have to configure a different pin for that function

Next is settings:

- your endstops inverted (logic) so you need to apply a bit mask to setting `$5`. In your case you need to invert X Y and Z. setting the first 3 bits to 1 will do the trick. Set `$5=7`
- next you need to invert the direction of your Z axis. Same thing add a bit mask to setting `$3` in this case you only need to set the 3rd bit on. Set `$3=4`

That's it.

Just one more note. By default µCNC will home towards the origin. If you want to make Z home on the top. you need to also invert the homing dir `$23`. Same thing as above. Bit mask to Z. `$23=4`

Repeat all the checks you've done before. The limits should be off without being pressed and on when pressed (in the case of Y and Z when any is pressed).

I'm thinking in your case and error while homing Z might occur. But please test and confirm this. I'll try to run your setup in my virtual HAL. I did not see that one coming :thinking: . lol. But if it turns out to give an error I'll fix.

Paciente8159, 2022-05-01:
Ok I just confirmed it. I'm working on it.

[17 more replies]

closed 2022-07-29 by rehanzfr