Paciente8159/uCNC #186: Issue with zeroing out work position.

jimsynz, 2022-05-04:
**Describe the bug**
So as you know my machine has end-stops on X & Y and I have a probe which I sometimes use on Z, but most of the time I just jog the Z axis into position at hit the "Zero out work offsets" for the Z axis in CNC.js.  This sends `G10 L20 P1 Z0` to which µCNC returns "Unsupported command".

**What is your hardware**
Describe your hardware:
https://github.com/jimsynz/uCNC/tree/master/

**What was the GCode running**
`G10 L20 P1 Z0`

**Additional context**
So I had a look at [the parser source](https://github.com/Paciente8159/uCNC/blob/master/uCNC/src/core/parser.c#L728-L747) and I see that `L20` is not supported and only `L2` is supported. A quick google says that they're synonyms and should both have the same behaviour.  I was about to open a PR to add it when I thought I'd better test it and sent `G10 L2 P1 Z0` and got back an `ok` however the WPos for the Z axis is still a non-zero value.

```
 $10=0
ok
> ?
<Idle|WPos:0.000,0.000,50.000|FS:0.000,0>
ok
> $10=1
ok
> ?
<Idle|MPos:0.000,0.000,50.000|FS:0.000,0>
ok
> $10=0
ok
> ?
<Idle|WPos:0.000,0.000,50.000|FS:0.000,0>
ok
> G10 L20 P1 Z0
error:20 (Unsupported command)
> G10 L2 P1 Z0
ok
> ?
<Idle|WPos:0.000,0.000,50.000|FS:0.000,0|WCO:0.000,0.000,0.000>
ok
>
```

I noticed the addition of `WCO:0,0,0` in the status, however if I'm understanding the `G10` command it should be setting the work position, not the offset?  Is that not right?

Thanks!

Paciente8159, 2022-05-04:
Sorry tripped on the closing button issue on my mobile phone 🙈 

The G10 L2 and not L20 is the way that it is defined in the [standard](https://github.com/Paciente8159/uCNC/blob/master/docs/rs274ngc.pdf).

I think you are mixing G54 G55, etcc. with G92.
G54 is an offset off your coordinate system relative to your machine origin. It's stored in memory and must be repeatable even after you turn the machine off and on.
That means that you send a command specifying the machine the point in machine coordinates to define that coordinate systems origin and not how you would like it to be at that position

A bit like this

G10 L2 P0 X(MX) Y(MY) Z(MZ)

Let's say you want to make your G54 origin in the machine `X0 Y0 Z50` your command should be `G10 L2 P1 Z50`. This makes it predictable. Image you set you G55 origin offset while with an offset on your current coordinate system. Where would your actual origin be? A sum of offsets?

G92 works like you said. It's relative to were you are and the coordinate system you are using. So G92 X0 Y0 Z0 adjusts the temporary offset to make like your origin is displaced to where you are standing. But that gets cleared after a power off/on. It's not stored permanently.

jimsynz, 2022-05-04:
Makes sense. If that is the case why does cncjs send `G10 L20 …`?

Paciente8159, 2022-05-04:
Grbl supports both G10L2, G10L20. I don't know why they adopted the L20.
They must have imported from LinuxCNC [G10](https://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L20)

With L20 it works like you stated. Has a relative offset.
I'll add this feature to µCNC too make it more compatible with existing softwares for Grbl.

Paciente8159, 2022-05-04:
Ok. It's done. Just need some proper testing. It's not yet merged in the main branch

Paciente8159, 2022-05-04:
Looking at Grbl confirms my suspicions.

[Grbl-gcode.c#550](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/gcode.c#L550)
// L20: Update coordinate system axis at current position (with modifiers) with programmed value
// WPos = MPos - WCS - G92 - TLO  ->  WCS = MPos - G92 - TLO - WPos

besides any active G92 also Tool Length Compensation is taken in account.

Paciente8159, 2022-05-04:
I did find one or two issues looking a bit more in the µCNC code 🤣.
Working on it.

jimsynz, 2022-05-04:
You’re a freaking machine my dude.

Paciente8159, 2022-05-04:
I'm merging #189 into master now. I've done a series of tests and seems to stick. If any thing goes off please let me know.
If you are able to replicate the commands prior to error it will be very useful.

Paciente8159, 2022-05-04:
By the way I'm evaluating the opportunity to implement the actual NIST G43. My current problem is that the UNO is already struggling for it's life as it is 🤣.
I still have the 16-bit bresenham card in the sleeve...but if I spend it...lol
Maybe I'll be able to optimize something later...although I'm running out of ideas

jimsynz, 2022-05-05:
Yeah. It's a tough call trying to fit it all in a ATMega328. I was thinking that there must be a breakout board for stepper drivers and stuff that works with esp32, since they're still pretty easy to get hold of.

Paciente8159, 2022-05-05:
There is. @bdring has an awesome firmware called Fluidnc. You can check it at https://github.com/bdring/Fluidnc. There are also some break out boards. His project is trully unique.

jimsynz, 2022-05-05:
Oh nice. Anyway, I see that #189 is merged so I’ll close this issue. Thanks again. You’re amazing.

closed 2022-05-05 by jimsynz

Paciente8159, 2022-05-05:
Just one side note.  NIST G43 was already implement 🤣🤣🤣 .
I forgot I already did that a while ago. That means you can add TLO via G43 H<tool number> command.
The TLO for each tool is available via $81 to $96 setting.