Paciente8159/uCNC #220: How to configure mega2560 and tmc 2209 driver

wrenchonline, 2022-06-27:
**Describe the bug**
amazing project.But I'm a newbie and the custom compilation fails.

**What is your hardware**
Describe your hardware:
1. arduino mega2560
    tmc2209 x 3
    
4. If this is a custom compilation tell us what configurations you changed
   platformio.ini 
      [platformio]
            include_dir=uCNC
            src_dir=uCNC
      [env:ramps14mirror]
      platform = atmelavr
      board = megaatmega2560
      framework = arduino
      build_src_filter = +<*>-<src/tinyusb>
      build_flags = -D BOARD=BOARD_RAMPS14_MIRROR -ffunction-sections -fdata-sections -mcall-prologues -mrelax -flto -fno-fat-lto-objects -fno-tree-scev-cprop -Wl,--gc-sections -Wl,--relax
      extra_scripts = avr_compiler.py
cnc_hal_config.h
      #define STEPPER0_HAS_TMC

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
uCNC\src\modules\../cnc_hal_config_helper.h:1209:17: error: invalid suffix "_BIT" on integer constant

Paciente8159, 2022-06-27:
Thanks. Although Tmc2209 was not tested it should work like the Tmc2208.
The pinout is tha same Marlin for the RAMPS or the the MKS Gen L.
Any problem let me know.

wrenchonline, 2022-06-29:
> Thanks. Although Tmc2209 was not tested it should work like the Tmc2208. The pinout is tha same Marlin for the RAMPS or the the MKS Gen L. Any problem let me know.
> 
> **EDIT**: here is an [instructable](https://www.google.com/amp/s/www.instructables.com/UART-This-Serial-Control-of-Stepper-Motors-With-th/%3famp_page=true). The connections and cables should match this setup.

After my day of research, I was finally able to drive the motor. But I found that running the command
`$4 = 0` 
the motor step_en_pin has low level，it didn't disabled.

Paciente8159, 2022-06-29:
Thanks. I'll look into that and report soon

Paciente8159, 2022-06-29:
Just one more note. Tmc initialization happens only on power up. Tmc drivers only accept commands when powered up. If the board is powered up via USB cable initial configuration will be sent but the tmc drivers will not get configure.
Power up must happen from the PSU. I will review this limitation soon.

Paciente8159, 2022-06-30:
I've tested $4 setting and all is working as expected.
As explained after changing the $4 value you need to either do a soft reset or send and unlock command ($X) and the step enable pins will update.

I'll build the M17/M18 module and add it to the modules repository. It's an easy module to build so I'll be able to build it and upload it today. I'll let you know when it's done.

Paciente8159, 2022-06-30:
I've created and uploaded the [M17-M18 module](https://github.com/Paciente8159/uCNC-modules/tree/master/M17-M18). You can include it in the project like explained in the instructions.

closed 2022-07-01 by wrenchonline