Paciente8159/uCNC #223: TMC2209 set stepper motor current failure

wrenchonline, 2022-07-06:
**Describe the bug**
A clear and concise description of what the bug is.

**What is your hardware**
Describe your hardware:
1. What's your board
    Arduino Mage2560 R3
    TMC2209
3. Paste your "Grbl/µCNC settings"
    Input  M906
`[STEPPER CURRENT:X30.596,Y30.596,Z-1.000,A-1.000,B-1.000,C-1.000,I-1.000,J-1.000]`
        Input   M906 X800 Y800   \
                   ok
                   M906
`[STEPPER CURRENT:X30.596,Y30.596,Z-1.000,A-1.000,B-1.000,C-1.000,I-1.000,J-1.000]`

5. If this is a custom compilation tell us what configurations you changed
    It seems that setting the current does nothing, I suspect it may be that I don't understand the tmc driver.

**Additional context**
Add any other context about the problem here.

Paciente8159, 2022-07-06:
Thanks for the report. I'll try to check it tonight.

Paciente8159, 2022-07-06:
I'm getting similar results on the TMC2208. I'll investigate. Have you tried sending motion commands. Do the motors stall or loose steps?

wrenchonline, 2022-07-07:
> 我在[tmc-drivers-fix](https://github.com/Paciente8159/uCNC/tree/tmc-drivers-fix)分支有一个提议的补丁。 我还没有测试过这个。我只能明天测试它。
> 
> 该分支尝试解决以下问题：
> 
> * 修复了未正确加载的 M914 命令和 TMC2130 的正确配置
> * 添加了 Marlin M913 命令
> * 通过 M 命令修改的 TMC 设置在软复位后不会丢失。这些仍然没有存储在 eeprom 中，因此硬重置会将它们恢复为默认值。
> * 现在所有 TMC M 命令都应该报告正确的值
> 
> > 实际设置电流也不行。我的可调电源用的是80mA。设置成功后，我把可调电压关掉再开机，电源是800MA，但是加速不大
> 
> 我不明白这个问题。你是不是电机转得太慢了？ 你能给我发一份你的设置输出`$$`和你的配置信息`$I`吗？

yeah，that right. this is my configuration:
$$
Sent 3 bytes
-> $0=33.333
$1=0
$2=0
$3=0
$4=0
$5=0
$6=0
$7=0
$10=1
$11=0.200
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=10.000
$25=50.000
$26=250
$27=2.000
$30=1000.000
$31=0.000
$32=0
$80=1
$81=0.000
$100=40.000
$101=40.000
$102=200.000
$110=500.000
$111=500.000
$112=500.000
$120=100.000
$121=100.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

$I
Sent 3 bytes
-> [VER: uCNC 1.4.6 - MKS GEN L V1]
[OPT:C3,DSS3_500,DC,XC,30,128]
ok

M350
Sent 5 bytes
-> [MICROSTEPS:X8.000,Y8.000,Z-1.000,A-1.000,B-1.000,C-1.000,I-1.000,J-1.000]

wrenchonline, 2022-07-07:
I think I solved the problem.run the command:
`$110=1000  $111=1000`

Paciente8159, 2022-07-07:
What kind of linear system are you using? I'm assuming based on your configs the X and Y axis are using gt2 belt with a 20teeth timing pulley and a 1.8º/step motor.
What GUI/PC app are you using? 
When you send the G0X50Y50 how does it move in the GUI? At normal speed or slow?
Does the board freezes? 

Tomorrow I Will test your settings on my board to see if I can replicate the problem. 

I did a quick test on the new branch and there are issues that I still have to solve with the new code for the Tmc driver. I'll repick that task tomorrow.

wrenchonline, 2022-07-08:
This shouldn't be an freezes issue, I'm just testing a single instruction.I use Candle GUI program And this is my gcode:
%
(Header)
(Generated by gcodetools from Inkscape.)
(Using default header. To add your own header create file "header" in the output dir.)
M3
(Header end.)
G21 (All units in mm)

(Start cutting path id: path7517)
(Change tool to Graffiti)
(Add G00 here to change sprayer if needed)
G00 Z5.000000
G00 X142.673750 Y36.041237
M42 P24 S170
G01 Z-1.000000 F100.0(Penetrate)
G01 X140.342110 Y36.041237 Z-1.000000 F400.000000
G01 X140.342110 Y43.929127 Z-1.000000
G03 X140.312358 Y44.882119 Z-1.000000 I-15.277773 J-0.000000
G03 X140.230490 Y45.715065 Z-1.000000 I-11.749558 J-0.734340
G03 X140.050282 Y46.533589 Z-1.000000 I-4.887294 J-0.646900
G03 X139.821210 Y47.029713 Z-1.000000 I-2.000975 J-0.622949
G03 X139.446249 Y47.476331 Z-1.000000 I-1.558264 J-0.927541
G03 X138.928240 Y47.798659 Z-1.000000 I-1.295230 J-1.504138
G03 X138.328686 Y47.981058 Z-1.000000 I-1.026344 J-2.297056
G03 X137.415150 Y48.059108 Z-1.000000 I-0.913536 J-5.307229
G03 X136.483559 Y47.950658 Z-1.000000 I-0.000000 J-4.055438
G03 X135.418380 Y47.587819 Z-1.000000 I1.341070 J-5.681884
G03 X134.411087 Y47.058874 Z-1.000000 I3.996172 J-8.833671
G03 X133.421600 Y46.384791 Z-1.000000 I5.540959 J-9.196856
G01 X133.421600 Y36.041237 Z-1.000000
G01 X131.089960 Y36.041237 Z-1.000000
G01 X131.089960 Y55.339283 Z-1.000000
G01 X133.421600 Y55.339283 Z-1.000000
G01 X133.421600 Y48.356764 Z-1.000000
G02 X134.562510 Y49.181725 Z-1.000000 I6.767962 J-8.158593
G02 X135.678830 Y49.770631 Z-1.000000 I4.522059 J-7.219439
G02 X136.870977 Y50.152914 Z-1.000000 I2.482407 J-5.691371
G02 X138.072480 Y50.279127 Z-1.000000 I1.201503 J-5.655849
G02 X140.138427 Y49.885952 Z-1.000000 I0.000000 J-5.624370
G02 X141.495520 Y48.927272 Z-1.000000 I-1.289614 J-3.265439
G02 X142.317314 Y47.441887 Z-1.000000 I-3.047404 J-2.656010
G02 X142.673750 Y45.032936 Z-1.000000 I-7.962167 J-2.408951
G01 X142.673750 Y36.041237 Z-1.000000
G00 Z5.000000
M42 P24 S0
 
(End cutting path id: path7517)

(Start cutting path id: path7515)
(Change tool to Graffiti)
(Add G00 here to change sprayer if needed)
G00 Z5.000000
G00 X128.101000 Y36.909401
M42 P24 S170
 
G01 Z-1.000000 F100.0(Penetrate)
G02 X126.918109 Y36.395800 Z-1.000000 I-7.556579 J15.784836 F400.000000
G02 X125.880980 Y36.041237 Z-1.000000 I-4.604307 J11.773873
G02 X124.827373 Y35.812238 Z-1.000000 I-2.045485 J6.872840
G02 X123.660960 Y35.731178 Z-1.000000 I-1.166413 J8.351511
G02 X122.179816 Y35.847218 Z-1.000000 I-0.000000 J9.510724
G02 X120.907640 Y36.165260 Z-1.000000 I1.153900 J7.319023
G02 X119.731999 Y36.733504 Z-1.000000 I2.036590 J5.713773
G02 X118.762030 Y37.504713 Z-1.000000 I2.893032 J4.634220
G02 X117.983687 Y38.479198 Z-1.000000 I3.709162 J3.760673
G02 X117.360570 Y39.761940 Z-1.000000 I5.867320 J3.642883
G02 X117.003613 Y41.150710 Z-1.000000 I6.944122 J2.525116
G02 X116.864480 Y42.949342 Z-1.000000 I11.556300 J1.798632
G02 X117.403411 Y46.130516 Z-1.000000 I9.658305 J-0.000000
[... 190 lines cut ...]
G01 X63.125120 Y46.719655 Z-1.000000
G01 X59.590452 Y36.041237 Z-1.000000
G01 X57.444847 Y36.041237 Z-1.000000
G01 X53.798558 Y49.894655 Z-1.000000
G01 X56.229417 Y49.894655 Z-1.000000
G01 X58.771898 Y39.166627 Z-1.000000
G01 X62.232152 Y49.894655 Z-1.000000
G01 X64.154515 Y49.894655 Z-1.000000
G01 X67.701585 Y39.166627 Z-1.000000
G01 X70.107640 Y49.894655 Z-1.000000
G01 X72.451683 Y49.894655 Z-1.000000
G00 Z5.000000
M42 P24 S0
 
(End cutting path id: path7507)

(Footer)
M5
G00 X0.0000 Y0.0000
M2
(Using default footer. To add your own footer create file "footer" in the output dir.)
(end)
%

However I am interested in F Value,Obviously I need more test info to report to you.

wrenchonline, 2022-07-08:
I misunderstood the meaning of F （feed rate）, what I am doing is a writing machine. It is controlled by the steering gear M42, which has little to do with F（feed rate）.

Paciente8159, 2022-07-08:
Ahh a custom machine...cool.
I see that you are using M42 to control the output of PWM0. You can in alternative use standart M3/M5 commands or eventually build a custom tool HAL module for your application. If you want to use the range os 0~255 instead of the range of 0~1000 that is configured by default you can change setting $30=255

µCNC also suports servo motors (PPM signal from 0.5ms~2.5ms) if you need to use that type of signal control.

Later tonight when I test your setup I'll also use your Gcode to see the behavior and give feedback.

About the board freeze, there are some strange issues with the atmega2560 MCU in terms of status query report speed.
Some programs like OpenCNCPilot default status pool rate is about 100ms. That means the program sends a `?` command every 100ms and some atmega2560 freezes at this rate. To cope with this and make the refresh rate responsive on the PC software it's possible to make this rate alot lower (for example send `?` command once every second) and enable an option on the config file:

```
// enables automatic status report sending
// this value sets the millisecond interval of the reports
// values bellow 100ms have no effect
#define STATUS_AUTOMATIC_REPORT_INTERVAL 0
```
 Setting `STATUS_AUTOMATIC_REPORT_INTERVAL` to 100 will make it send a status report automatically every 100ms

Paciente8159, 2022-07-08:
I've tested the patch and all seems to be working fine.
I've also tested your settings and your Gcode file and all when well. Try to bump speeds to about `$110=6000` and `$111=6000` and accelerations to `$120=100` and `$121=100`

I'll merge this one now. Later I'll probably add the M919 command althoug it's possible to do it using the generic M920 command

closed 2022-07-08 by Paciente8159

Paciente8159, 2022-07-08:
Just one more note on your gcode. You don't need that many decimal places in the coordinates.
With your config of 40steps/mm you Will get 0.025mm per step resolution. You should keep it to about 3 decimal places only (tipical for units in mm)