Paciente8159/uCNC #224: Compilation error

rombor73, 2022-07-08:
Hello, 
Tried to compile for Bluepill custom configuration, get an error when I disable //#define PROBE_BIT 9 in boardmap_bluepill.h.
It says "c:/users/dst/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/bluepill_f103c8/src/src/hal/mcus/stm32f1x/mcu_stm32f1x.c.o: in function `mcu_init':
mcu_stm32f1x.c:(.text.mcu_init+0x41e): undefined reference to `mcu_disable_probe_isr'".
Actually I am ok with not disable it, just reporting, Thanks for great project. Have not enough knowledge to fix it.

It is win10, vscode 1.69.0-insider, pio latest

Paciente8159, 2022-07-08:
Thanks for the report I look at it and give you feedback

Paciente8159, 2022-07-08:
Found the problem. I'll patch this in a couple of hours. There are several ways to fix this.
One of them is to change this piece of code in [line 4275 of file mcumap_stm32f1x.h](https://github.com/Paciente8159/uCNC/blob/a3b4316c3aaf6909605fc2014bee35fe35488da7/uCNC/src/hal/mcus/stm32f1x/mcumap_stm32f1x.h#L4275)

from this

```
#ifdef PROBE
#ifdef PROBE_ISR
#define mcu_enable_probe_isr() SETBIT(EXTI->IMR, PROBE_BIT)
#define mcu_disable_probe_isr() CLEARBIT(EXTI->IMR, PROBE_BIT)
#else
#define mcu_enable_probe_isr()
#define mcu_disable_probe_isr()
#endif
#endif
```

To this:

```
#if (defined(PROBE) && !(PROBE<0) && defined(PROBE_ISR))
#define mcu_enable_probe_isr() SETBIT(EXTI->IMR, PROBE_BIT)
#define mcu_disable_probe_isr() CLEARBIT(EXTI->IMR, PROBE_BIT)
#else
#define mcu_enable_probe_isr()
#define mcu_disable_probe_isr()
#endif
```

An alternative would be to simply add the missing function declarations anywhere inside mcu_stm32f1x.c file (for example [after function mcu_init()](https://github.com/Paciente8159/uCNC/blob/a3b4316c3aaf6909605fc2014bee35fe35488da7/uCNC/src/hal/mcus/stm32f1x/mcu_stm32f1x.c#L1224) add this:

```
#ifndef mcu_enable_probe_isr
	void mcu_enable_probe_isr(void){}
#endif

#ifndef mcu_disable_probe_isr
	void mcu_disable_probe_isr(void){}
#endif
```

closed 2022-07-08 by Paciente8159