Paciente8159/uCNC #229: Debugging failure with avr-debugger on mega 2560 R3

wrenchonline, 2022-07-11:
**Describe the bug**
A clear and concise description of what the bug is.

**What is your hardware**
Describe your hardware:
1. What's your board
   Mega2560 R3
   avr-debugger 
3. Paste your "Grbl/µCNC settings"
   none
5. If this is a custom compilation tell us what configurations you changed

[env:mks_gen_l_v1]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_deps = jdolinay/avr-debugger@^1.5
debug_tool = avr-stub
debug_port = \\.\COM3
debug_build_flags = -O0 -ggdb3 -g3 
build_flags =
    -D BOARD=BOARD_MKS_GEN_L_V1 -ffunction-sections -fdata-sections -mcall-prologues -mrelax -flto -fno-fat-lto-objects -fno-tree-scev-cprop -Wl,--gc-sections -Wl,--relax
    -D AVR8_UART_NUMBER=1
build_src_filter = +<*>-<src/tinyusb>
extra_scripts = avr_compiler.py

avr-debugger 1.5 supports other uart debugging arduino.so that I would set it like this.

on  cnc.c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <stdint.h>
#include "cnc.h"
#include "avr8-stub.h"
......
void cnc_init(void)
{
	debug_init();
       ......
}

I have a breakpoint at cnc_run() on the vscode platformio platform, and I cannot debug.
this  error information:
        Ignoring packet error, continuing...
        warning: unrecognized item "timeout" in "qSupported" response
        2
        Ignoring packet error, continuing...
        .pioinit:13: Error in sourced command file:
        Bogus trace status reply from target: timeout

wrenchonline, 2022-07-11:
In another small project of mine was able to drop off

Paciente8159, 2022-07-11:
I did not try it but I believe [jdolinay](https://github.com/jdolinay)/[avr_debug](https://github.com/jdolinay/avr_debug) uses WDT.

µCNC disables WDT at [init](https://github.com/Paciente8159/uCNC/blob/7f69dafa1ec8e7125720a1ede154d5ee42be1f94/uCNC/src/hal/mcus/avr/mcu_avr.c#L419) for most architectures.

I'll try to modify this by adding a DEBUG option for AVR. I'll give you feedback on that in a day or two.

Paciente8159, 2022-07-11:
Also take note that you should probably disable Z limits pins in the board. They colide with Serial port 1 TX/RX pins.

wrenchonline, 2022-07-12:
> I did not try it but I believe [jdolinay](https://github.com/jdolinay)/[avr_debug](https://github.com/jdolinay/avr_debug) uses WDT.
> 
> µCNC disables WDT at [init](https://github.com/Paciente8159/uCNC/blob/7f69dafa1ec8e7125720a1ede154d5ee42be1f94/uCNC/src/hal/mcus/avr/mcu_avr.c#L419) for most architectures.
> 
> I'll try to modify this by adding a DEBUG option for AVR. I'll give you feedback on that in a day or two.
> 
> You can try to modify the [RTC timer](https://github.com/Paciente8159/uCNC/blob/7f69dafa1ec8e7125720a1ede154d5ee42be1f94/uCNC/src/hal/boards/avr/boardmap_ramps14.h#L124) (uses Timer0 by default) used by µCNC in the boardmap file and use the AVR8_USE_TIMER0_INSTEAD_OF_WDT = 1 on the stub.

thanks. I will be try.

Paciente8159, 2022-07-12:
Seems to work...[avr-debug branch](https://github.com/Paciente8159/uCNC/tree/avr-debug)

wrenchonline, 2022-07-12:
> Seems to work...[avr-debug branch](https://github.com/Paciente8159/uCNC/tree/avr-debug)

It looks like the situation is quite complicated.It worked, but not very well, maybe somewhere else shield the ISR.

Paciente8159, 2022-07-12:
I've went through the docs in the avr-debugger.
I've disabled input ISR interrupts to avoid conflicts with the stub.
I've changed the default INT used from 0 to 6 (unsused pin of Mega as stated in the docs of the avr-debug)
I've fixed a typo in the include.

Please re-test the new changes.

wrenchonline, 2022-07-13:
The test still fails.I re-checked the possibility of mega2560 serial communication and avr-debug. I uploaded my test code and found that using breakpoint() can break to the position I specified with serial communication.It is indeed an imperfect thing that arduino can both serial communicate and debug

Paciente8159, 2022-07-13:
I am also waiting on an ESP32 in the mail to start the core development for that MCU.

In the mean time I will try to update the virtual/Windows HAL. Although a bit different from the microcontrollers it's good enough to simulate all the functionalities and interactions of the core code.

Paciente8159, 2022-07-13:
> I only have esp32 on hand (not esp32 8266), I just bought bluepill f103c8 and ST-Link, it takes time to wait, I want to debug with esp32 before that.

I also had good debugging experience with the SAMD21 MCU and again a CMSIS-DAP. I use a second Bluepill as my CMSIS-DAP and it has worked pretty well with OpenOCD

wrenchonline, 2022-07-13:
> > 我手头只有esp32（不是esp32 8266），我刚买了bluepill f103c8和ST-Link，需要时间等，我想在那之前用esp32调试。
> 
> 我对 SAMD21 MCU 和 CMSIS-DAP 也有很好的调试经验。我使用第二个 Bluepill 作为我的 CMSIS-DAP，它与 OpenOCD 配合得很好

thanks.I learned about the debugging process of stm32, I canceled ST-Link and used the more popular CMSIS-DAP.

Paciente8159, 2022-07-13:
I'll also review the windows emulator code and integrate it in PlatformIO. I think this will be fairly quick.

wrenchonline, 2022-07-13:
> I'll also review the windows emulator code and integrate it in PlatformIO. I think this will be fairly quick.

this sounds cool.I'm reading about CMSIS-DAP as platformio debugging, can you run it there?

Paciente8159, 2022-07-13:
When I ran CMSIS-DAP and used it to debug SAMD21 I did it a bit manually (few automations) (you can check  some example config files inside the makefiles directory).

But yes it should be possible. I'll try it out tonight and let you know. If that is the case I will also update the platformIO.ini file with all the changes needed.

Paciente8159, 2022-07-13:
I've uploaded an example on how to enable debug with CMSIS-DAP via PIO.
Just uncomment the lines in platformio.ini file that enable that.

I did not got PIO to make the windows emulator debug session running. It compiles fine but does not start the debug.

Paciente8159, 2022-07-13:
Just one more note. Debugging requires you to use the UART interface option (with an external USB<->serial converter) and not the internal STM32 USB. This is because prolonged breaks causes the usb driver on PC side to think the device is malfunctioning and stops communication. Serial communication does not have this problem

wrenchonline, 2022-07-14:
thanks.we will try and report the results to you.

Paciente8159, 2022-07-14:
I've updated the virtual HAL for Windows. Unfortunately I can't seem to be able get the debug session working on VSCode.
I still need the [DevCpp Portable](https://sourceforge.net/projects/orwelldevcpp/files/Portable%20Releases/Dev-Cpp%205.11%20TDM-GCC%20x64%204.9.2%20Portable.7z/download)
I open the project file with DevCpp (the uCNC.dev file inside makefiles/virtual).
- to change settings go to Project->Project options
- then navigate to the Parameters tab and change the setting in the left column
- compile and run in debug mode
- WIN_INTERFACE is set to 1 (use a telnet connection with port 34000 to communicate with the app)
- WIN_INTERFACE set to 0 allows you to use a virtual null modem (like com0com). WIN_COM_NAME allow you to change the port to connect
- you can connect with Candle or other software you choose

To interact with the virtual HAL download the [C# GUI](https://github.com/Paciente8159/uCNC-virtual-GUI/releases/download/1.0.2/uCNC.zip) I made for it. You will be able to get a visual feedback of the outputs and simulate inputs.

[image]

wrenchonline, 2022-07-17:
> I've uploaded an example on how to [enable debug with CMSIS-DAP via PIO](https://github.com/Paciente8159/uCNC/tree/f2dd62edf2d48461b16701d544a31066883697b3). Just uncomment the lines in [...]
With openocd, a parameter should be set:
`
debug_port = localhost:3333
`
I got two boards, stm32f103c8 and stm32f407zg, but the stm32f103c8 board is broken, and I am currently debugging with the stm32f407zg board.The above is the solution to the problem I encountered when debugging. It seems that I don't know how the serial port of f407zg is mapped. this is my option:
[common_stm32]
platform = ststm32
; debug with st-link
; platform_packages = platformio/tool-openocd
upload_protocol = cmsis-dap
debug_port = localhost:3333
debug_tool = cmsis-dap
debug_build_flags = -Og -g3 -ggdb3 -gdwarf-2
debug_server =
  E:\\xpack-openocd-0.11.0-4\\bin\\openocd.exe
  -f
  E:\\xpack-openocd-0.11.0-4\\scripts\\interface\\cmsis-dap.cfg
  -f
  E:\\xpack-openocd-0.11.0-4\\scripts\\board\\stm32f4x.cfg
debug_init_cmds =
  target extended-remote $DEBUG_PORT
  load
  monitor reset init
build_flags = ${common.build_flags} -D HAL_TIM_MODULE_DISABLED -D HAL_EXTI_MODULE_DISABLED -D HAL_UART_MODULE_ONLY

; [env:bluepill_f103c8]
; extends = common_stm32
; platform = ststm32
; board = bluepill_f103c8
; build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLUEPILL -D INTERFACE=1

; [env:bluepill_f103c8_clone]
; extends = common_stm32
; platform = ststm32
; board = bluepill_f103c8
; build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLUEPILL -D INTERFACE=1
; ; uncomment if it's a bluepill clone
; upload_flags = -c set CPUTAPID 0x2ba01477

[env:blackpill_f401cc]
extends = common_stm32
platform = ststm32
board = blackpill_f401cc
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL -D INTERFACE=1

Paciente8159, 2022-07-17:
To use the UART communication change the INTERFACE option inside build_flags in the Stm32F4 ini from 1 to 0. INTERFACE=1 uses USB and INTERFACE=0 uses UART.
By default Uart0 is used (TX-pin A9 and RX-pinA10). This is set in the boardmap

wrenchonline, 2022-07-17:
> To use the UART communication change the INTERFACE option inside build_flags in the Stm32F4 ini from 1 to 0. INTERFACE=1 uses USB and INTERFACE=0 uses UART. By default Uart0 is used (TX-pin A9 and RX-pinA10). This is set in the boardmap

In the boardmap_mks_robin_nano_v1_2.h file：

#if (INTERFACE == INTERFACE_USART)
#define UART_PORT 3
#define TX_BIT 10
#define TX_PORT B
#define RX_BIT 11
#define RX_PORT B
#else
#error "This board uses an external USB<->Serial converter"
#endif

B10-B11 OR A9-A10 In fact in debug mode I can't get serial messages

wrenchonline, 2022-07-17:
> To use the UART communication change the INTERFACE option inside build_flags in the Stm32F4 ini from 1 to 0. INTERFACE=1 uses USB and INTERFACE=0 uses UART. By default Uart0 is used (TX-pin A9 and RX-pinA10). This is set in the boardmap

In fact I don't see the serial reply message.Afrer Serial link to TX-pin A9 and RX-pinA10. Display garbled characters

Paciente8159, 2022-07-17:
Now I noticed you are running an STM32F**407** not a an STM32F**401**. I have not tested that yet but you might have to make some modifications to take that in account.

In the stm32.ini file change the board to match yours (including cpu speed). Also in the boardmap for the black pill you might need to adjust these lines:

```
#ifndef STM32F401xC
#define STM32F401xC
#endif

#ifndef FLASH_SIZE
#define FLASH_SIZE 0x60000
#endif

#ifndef F_CPU
#define F_CPU 84000000UL
#endif
```

Again I have not tested this (I don't have a stm32f407 to test so I can't test this by myself).
My stm32f401 runs fine.

wrenchonline, 2022-07-19:
> Now I noticed you are running an STM32F**407** not a an STM32F**401**. I have not tested that yet but you might have to make some modifications to take that in account.
> 
> In the stm32.ini file change the board to match yours (including cpu speed). Also in the boardmap for the black pill you might need to adjust these lines:
> 
> ```
> #ifndef STM32F401xC
> #define STM32F401xC
> #endif
> 
> #ifndef FLASH_SIZE
> #define FLASH_SIZE 0x60000
> #endif
> 
> #ifndef F_CPU
> #define F_CPU 84000000UL
> #endif
> ```
> 
> Again I have not tested this (I don't have a stm32f407 to test so I can't test this by myself). My stm32f401 runs fine.
> 
> **EDIT**: Note that the F_CPU is taken in account to calculate the values for for the MCU registers given the desired baudrate. That means that an incorrect value will make the UART in your board actually run at a different speed from the desired.

The answer is gradually off topic, this question is closed, stm32f407zgt6 I will continue to test and report back to you.

closed 2022-07-19 by wrenchonline