Paciente8159/uCNC #235: Auxiliary function of screw compensation,

wrenchonline, 2022-08-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the feature is:
HAL kinematics

**Describe the solution you'd like**
There will inevitably be errors in accuracy in some purchased CNC accessories，This will inevitably cause the software to perform precision compensation. I think there is a good solution to facilitate the implementation of the compensation interface.This sounds absurd and an unnecessary feature, but it would be a crazy move to have the CNC let the ball screw do the optimization of error compensation or use the grating scale for closed loop detection and dynamic adjustment compensation.

Paciente8159, 2022-08-08:
µCNC already implements a backlash compensation mechanism that can be enabled via config file.
It works by performing a configurable amount of additional steps (via settings)[https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#140-to-147---xyzabc-backlash-compensation], every time there is a direction inversion on a linear actuator/motor.

Do you you need a different (non linear) mechanism compensation?
Can yes, can you give a better description on what you need?

I have build in modules that may be combined to perform closed loop feedback compensation although I don't have a specific module for a real case application developed yet. That might also be a possible solution.

wrenchonline, 2022-08-10:
> µCNC already implements a backlash compensation mechanism that can be enabled via config file. It works by performing a configurable amount of additional steps (via settings)[https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#140-to-147---xyzabc-backlash-compensation], every time there is a direction inversion on a linear actuator/motor.
> 
> Do you you need a different (non linear) mechanism compensation? If yes, can you give a better description on what you need?
> 
> I have build in modules that may be combined to perform closed loop feedback compensation although I don't have a specific module for a real case application developed yet. That might also be a possible solution.

solve nonlinearity? I don't have any ideas at the moment. It seems that relying on software is not a good solution. The bottle top of the machine tool is also a problem. I currently DIY an aluminum profile cnc engraving machine, and I am thinking of ways to improve the accuracy. There are so many hidden causes and solutions to this problem

Paciente8159, 2022-08-10:
Distortions caused by machine deformation/bending are nearly impossible to compensate. The only solution is to lower your engraving cutting speeds to minimize the problem.

The main issue you referred was precision compensation due to known (linear) linear or mechanical weaknesses. µCNC features several precision compensation mechanisms.

- Skew compensation to fix structural issues like lack of perpendicularity between axis that can't be or are hard to do mechanically
- Backlash compensation to reduce errors caused by direction inversions in linear motion systems that are prone to this problem (leadscrew and nut mainly)
- Linear actuator based acceleration profile. This was mainly designed for machines with very heavy moving gantry parts (like cement epoxy based machines). With this calculations are done based on the change of direction/speed of the linear actuators and not the based on the tool travel path.

THC and Feed/Spindle synchronization are other also in my TODO list to had but I don't have a setup to develop these features yet. These are examples of (non linear/on the fly compensations) . The building blocks for those exist inside µCNC. It's just a question of linking them the right way.

If you have a special/custom need that you can define let me know and I'll try to help you.

wrenchonline, 2022-08-19:
> Distortions caused by machine deformation/bending are nearly impossible to compensate. The only solution is to lower your engraving cutting speeds to minimize the problem.
>
> The main issue you [...]

Sorry for the late answer, I may have to wait until the CNC is done before making a design request

Paciente8159, 2023-01-12:
Due to inactivity this issue will be closed.
Please feel free to comment when you have your design request or open a new issue.

Thanks.

closed 2023-01-12 by Paciente8159