Paciente8159/uCNC #269: Bed Mesh autoleveling support for pcb engraving

pwilkowski, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the feature is:
- Extension module - GCode and transformation

**Describe the solution you'd like**
If you want to engrave PCBs properly, you need to have a bed mesh as copper layer is super thin, cannot do that without having "bed mesh" and transformation it applies to gcode

**Describe your target machine hardware**
Every machine as it can be done with clamps to pcb + engraving tool

**Additional context**
Klipper, Marlin, Reprap firmwares all have this implemented

Paciente8159, 2022-09-20:
Thanks for your feedback.
I will eventually implement this (to be usable for example with files that are executed from an SD card) but for now I have a few issues in the queue so it might take a while.

This can be done by several PC software like OpenCNCPilot or bCNC. They map the board with the probing Gcode and readjust the file.
[OpenCNCPilot](https://github.com/martin2250/OpenCNCPilot) is specially target at this for PCB milling, and it's fully integrated with µCNC and Grbl.

Paciente8159, 2022-12-14:
Implemented an embedded surface leveling that works in the following way:

- Enables G39, G39.1 and G39.2 gcodes (G39 is in the motion modal group. It will remain active until another motion group command is issued)
- G39 makes a 9 point rectangular probing and stores the results has an internal Hmap
- Any motion inside the the probed rectangle area has the Z coordinate modified via a bilinear interpolation of the Hmap matrix
- G39.1 disables the Hmap
- G39.2 re-enables the previous Hmap
- Reset or homing clears the Hmap
- Since this is a per job feature the Hmap is not stored in non volatile memory

G39 command has the following syntax:
G39 X<initial probing point> Y<initial probing point> Z<max probing point depth> I<rectangle X length> J<rectangle J length> R<retraction distance>

- X is the initial probing point. This is optional. If undefined the current position is used. Depending if G90 or G91 is active, absolute or relative coordinates will be used.
- Y is the initial probing point. This  is optional. If undefined the current position is used. Depending if G90 or G91 is active, absolute or relative coordinates will be used.
- Z is the probing point depth limit. Probing will halt and machine will enter alarm state if object not found before the Z limit. It must be set explicitly. Depending if G90 or G91 is active, absolute or relative coordinates will be used.
- I is the X offset that defines the limit region to probe. It must be set explicitly and must be a positive value. If undefined the current position is used. It's always a relative distance to X.
- J is the Y offset that defines the limit region to probe. It must be set explicitly and must be a positive value. If undefined the current position is used. It's always a relative distance to Y.
- R is the retract height to move the probe between probing actions. It must be set explicitly. This distance is relative to the position defined by Z (point depth limit).

Motion will be done like this:

- If Z is lower then retract do a fast retract motion
- Do a fast transverse motion to the 1st point (defined by X, Y)
- Probe at the programmed feedrate. If fail enter alarm mode. If OK redo retract and transverse motion prior to the next probe.
- After all points probed successfully do a retract, followed by a transverse motion to the 1st point (defined by X, Y) and do a linear motion to the 1st probed Z at the programmed feedrate.

closed 2022-12-18 by Paciente8159

zenggood, 2025-04-17:
I would like Hmap to be able to store in non-volatile memory

Paciente8159, 2025-04-19:
Thanks for the feedback. That should be easy to implement. I'll give some feedback in a few days.

Paciente8159, 2025-04-25:
> I would like Hmap to be able to store in non-volatile memory

Check #865