Paciente8159/uCNC #347: [BUG] Homing issue

etet100, 2023-01-05:
**Describe the bug**

I have noticed this issues during homing procedure:

Expected behaviour: 

[image]

Current behaviour:

[image]

I does not happen every time. It's random.

**What is your hardware**

Arduino Uno + CNC Shield v3
uCNC 1.5.6

Settings:

```
$0=33.333
$3=2
$4=0
$10=1
$11=0.200
$13=0
$20=1
$23=0
$26=250
$27=4.000
$32=0
$80=1
$100=80.000
$101=144.000
$110=6000.000
$111=6000.000
$120=1000.000
$121=1000.000
```

**What was the GCode running**

$H

**Additional context**

My uCNC is slightly modified. I have to add a couple of lines of code. However, i don't think my changes could be related to this issues. I will try to add some datails if necessary.

Paciente8159, 2023-01-05:
And just a note about the homing motion.

The actual motion is like this:
[image]

Depending on the type of switch and repeatability the retract distance can be unnoticeable.
Bad switch repeatability, can act like you described.

Just another question. Did you modified the boardmap setting to match shield V3?
In that version the pinout is similar to grbl 0.9 and not 1.1.

etet100, 2023-01-05:
- Are you using self-squaring? NO? I'm not sure
- Are you using backlash compensation? NO
- Are you using skew compensation? NO
- Is it a COREXY or a Cartesian machine? Cartesian
- What type of endstops are you using? Like this:
[image]

This is a 2 axis machine with servo based pen lifting mechanizm. And this is what i had to modify. I need another limit switch, not related to any stepper motor. Basically it's something like:

```
void mcu_set_servo_min(uint8_t servo)
{
	#define SERVO_MIN 100
	uint8_t val = mcu_get_servo(servo);
	if (val < SERVO_MIN) {
		mcu_set_servo(servo, val + 2);
	}
}
```

and 
```

if (!mcu_get_input(LIMIT_Z)) {
    mcu_set_servo_min(SERVO0);
}
```

Pushed Z limit means that servo is locked and position has to be decreased until limit switch is released. 

I have modified boardmap a bit. But i guess i had to do it right becouse usually it's working fine.

Check this, please:

https://www.youtube.com/shorts/AK2CYPMRM7c

Paciente8159, 2023-01-05:
Thanks for the video. I will try to reproduce the behavior and I will give some feedback on it.

About the the pen holder. The best way to implement it would be a custom tool I believe. It should be fairly trivial.
M3/M4 S0 or M5 to lift the pen, and M3/M4 with any value to lower the pen should work.
I will create a pen holder tool based on your setting and add it to the project.

I don't recommend using LIMIT_Z. Instead you could use any of the generic DIN pins. I did not test but triggering LIMIT_Z with hardware limits enabled might set the machine in ALARM mode.

etet100, 2023-01-05:
No worries. It won't trigger alarm because:
```
#if !(LIMIT_Z < 0)
	value |= LIMIT_Z_MASK;
	//value |= ((mcu_get_input(LIMIT_Z)) ? LIMIT_Z_MASK : 0);
#endif
```
I'm going to try to reverse most of my changes. Or maybe all of them.

Paciente8159, 2023-01-05:
I've added a servo controlled pen holder tool example to the branch I linked in this issue. It will be later added to the main branch.

It does not use a switch. Instead you can find adjust it with M3 S<value>.
M5, M3 S0, or M4 with any S value will retract it.

If you want to use the switch and automatically adjust the pen via switch you can modify the `set_speed` function inside the tool code file. Again I recommend you use any other generic input pin by creating it in the boardmap file and checking it's state while adjusting the servo output value.

I have not tested the homing issue because I don't have my UNO+Shield v3 with me right now and I'll post give you feed back on it later.

Paciente8159, 2023-01-05:
I found an issue that was introduced with version 1.5.6 that is explained in this issue #349 and is patched by pull request #350.

Can you apply the modifications in #350 to your copy and see if it fixes the issue?

I was not able to reproduce the same issue as you but I had another type of issue that caused random homing (ultra low speed homing) speed problems.
Maybe the both problems are related.

etet100, 2023-01-05:
Much better after aplying https://github.com/Paciente8159/uCNC/pull/350! I have already tried 40 times.

Paciente8159, 2023-01-05:
Thanks for the input.
I'll merge the changes. A new release will be ready in the next days.
Feel free to submit a new issue if the issue reappears.

etet100, 2023-01-07:
It happened twice since yesterday. Damn.

Paciente8159, 2023-01-07:
I'll reopen and try to reproduce it.

Paciente8159, 2023-01-07:
I can't seem to reproduce the behavior on current version 1.5.7.
On my UNO+Shield v3 either with the default 3 AXIS configuration nor with only 2 AXIS all seems to work fine.
But I did find some issues regarding alarm states messages output caused by the changes in version 1.5.6 and I'm working on that.

Can you send the full print of the setting command ($$) and the info command ($I)?
Did you add any custom code?
Just another question what are you using to build the code? Arduino IDE or PlatformIO?

You can also try using previous version 1.5.5 and see if the issues remain.

etet100, 2023-01-08:
PlatformIO

```
14:38:36: [VER: uCNC 1.5.6 - Arduino UNO]
14:38:36: [OPT:C2,DSS3_500,IE,DP,13,128]
```
```
14:34:35: $0=33.333 (Step pulse time, microseconds)
14:34:35: $1=0 (Step idle delay, milliseconds)
14:34:35: $2=2 (Step pulse invert, mask)
14:34:35: $3=2 (Step direction invert, mask)
14:34:35: $4=0 (Invert step enable pin, boolean)
14:34:35: $5=255 (Invert limit pins, boolean)
14:34:35: $6=0 (Invert probe pin, boolean)
14:34:35: $7=0 (-)
14:34:35: $10=1 (Status report options, mask)
14:34:35: $11=0.200 (Junction deviation, millimeters)
14:34:35: $12=0.002 (Arc tolerance, millimeters)
14:34:35: $13=0 (Report in inches, boolean)
14:34:35: $20=1 (Soft limits enable, boolean)
14:34:35: $21=1 (Hard limits enable, boolean)
14:34:35: $22=1 (Homing cycle enable, boolean)
14:34:35: $23=0 (Homing direction invert, mask)
14:34:35: $24=1500.000 (Homing locate feed rate, mm/min)
14:34:35: $25=2000.000 (Homing search seek rate, mm/min)
14:34:35: $26=250 (Homing switch debounce delay, milliseconds)
14:34:35: $27=4.000 (Homing switch pull-off distance, millimeters)
14:34:35: $30=1000 (Maximum spindle speed, RPM)
14:34:35: $31=0 (Minimum spindle speed, RPM)
14:34:35: $32=0 (Laser-mode enable, boolean)
14:34:35: $80=1 (-)
14:34:35: $81=0.000 (-)
14:34:35: $100=80.000 (X-axis travel resolution, step/mm)
14:34:35: $101=144.000 (Y-axis travel resolution, step/mm)
14:34:35: $110=6000.000 (X-axis maximum rate, mm/min)
14:34:35: $111=6000.000 (Y-axis maximum rate, mm/min)
14:34:35: $120=1000.000 (X-axis acceleration, mm/sec^2)
14:34:35: $121=1000.000 (Y-axis acceleration, mm/sec^2)
14:34:35: $130=420.000 (X-axis maximum travel, millimeters)
14:34:35: $131=750.000 (Y-axis maximum travel, millimeters)
```

I have isolated smaller example of non-working code. Sometimes it happens almost every time, but sometimes every 20-30-40-50 times!

Here are my changes to v1.5.6:

[homing_issues_v1.5.6.patch](https://github.com/Paciente8159/uCNC/files/10368747/homing_issues_v1.5.6.patch)

I haven't tried v1.5.5 yet.

Paciente8159, 2023-01-09:
> I have isolated smaller example of non-working code. Sometimes it happens almost every time, but sometimes every 20-30-40-50 times!

Can you detail a bit better. I thought you were having problems with homing only. Does it happen in other situations too?

I'm reworking the internal machine states and some issues I found that were preventing some alarm messages to be printed.
I should have (I hope) a working branch today.

Paciente8159, 2023-01-10:
After a good night sleep I was going over the issue you are getting and something I changed yesterday in #353 that might be related.

There are a couple of changes on version 1.5.6 that you can try.
The first one is adding this line of code in io_control.c

```
void io_invert_limits(uint8_t limitmask)
{
	io_invert_limits_mask = limitmask;
	mcu_limits_changed_cb(); // add this line
}
```

if this does not solve your issues you can also try to enable on the config file this option
#define FORCE_SOFT_POLLING

etet100, 2023-01-10:
I tried about 50 times before adding this line and 50 times after. Before adding it, the problem occurred 6 times, after adding it, 0 times.

Version 352 does not work for me at all. It always shows ALARM:1 or ALARM:11. I didn't test it too long.

Paciente8159, 2023-01-10:
Ok. As I supected after inverting the limits state on the first contact, the switch release was not being detected.

I was still testing version #353 today and applied several patches, so probably you tried a bugged version.
I think I'm close to having it working.

On this new version any alarm bellow 4 will require a hard or soft reset before allow unlocking or rehoming.

I'll give you feedback as soon as it's ready.

Paciente8159, 2023-01-10:
Just another question. On your endstops are you using the NC or the NO contacts?

Paciente8159, 2023-01-10:
Just did about 40ish test (and simulated correct and incorrect triggers by hand) and all results were correct with current branch #353.

I'll just do a couple more test tomorrow and if all checks out it will be integrated.

Again the behavior of µCNC was slightly adjusted, so some expect the following:

- a new alarm code was added (13) that is triggered if a limit switch is activated without any motion. this requires only an unlock code or rehoming to continue
- all alarm codes bellow 4 (hard limits with motion, soft limits, and abort command with motion) will cause require a soft or emergency stop press to release. If the homing cycle is enable it the machine will remain in Alarm state until an unlock or homing command is issued
- it is now possible to have homing cycle enabled and hard limits disabled. The homing command will check the hard limits like usual but after that they will be disabled and can be crossed (this might be useful with some type of optical sensors that can enable motion beyond that point).

Paciente8159, 2023-01-11:
#353 was integrated. I hope that this patch adds all the necessary fixes and adds improves user experience.
I hope I covered everything. Tested on UNO and on STM32 bluepill.

Paciente8159, 2023-01-12:
Version 1.5.7 released.
This should address this issue. A huge thank you for your feedback and help.

Feel free to report any other issue, or if you get any problems again.

closed 2023-01-12 by Paciente8159