Paciente8159/uCNC #365: [BUG]

Juank23, 2023-04-20:
good day, friend download the image for a stm32f401 blackpill but only the pin 13 led is flashing, the computer recognizes the usb port but the candle does not connect, this firmware how does it work ??? graccias.

Paciente8159, 2023-04-21:
If the led is flashing the CNC is running.
If you connect to the board using a terminal (like Termite or other) and you send a `?` command it will probably send an Alarm message indicating that the emergency stop is active.

If you used the precompiled version you need to connect ESTOP (emergency stop) and FHOLD (feed hold) pins to ground.
You can then invert the logic state of those pins in setting $7 and remove the shorts to ground.
You can read about it in the [Wiki FAQ](https://github.com/Paciente8159/uCNC/wiki/FAQ#ok-ive-installed-%C2%B5cnc-on-my-board-but-nothing-is-happening-what-went-wrong) and in the [basic user guide](https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#%C2%B5cnc-wiring)

Let me know if it solved the issue.
Thanks

Juank23, 2023-04-21:
good morning friend, I use Candle, Gcode Sender, code is not compatible with these sender? I stay tuned thanks

Juank23, 2023-04-21:
friend to put the pin only b5 to ground stops flashing the led STM32F401 and disconnects from the usb port, without the pins to ground on the console gives me this message [MSG:Reset to continue] from the source code where I can change the status of these pins?

Paciente8159, 2023-04-21:
I've tested the [blackpill-f401.bin](https://github.com/Paciente8159/uCNC/releases/download/v1.6.0/blackpill-f401.bin) and it's working as expected with Candle.

I think the issue that you are having is that you are setting $7 to invert the logic state of ESTOP and FHOLD but are not removing the shorts to ground and sending a $X to unlock, or you are setting $7 and unplugging and plugging in USB cable and resetting the board and with that resetting $7.

There is and important [note on the STM32F4 wiki](https://github.com/Paciente8159/uCNC/wiki/STM32Fx:-Customizing#important-note-on-stm32f4) that you might have missed. It does not have EEPROM emulation support. If you unpower the board, all settings will be lost.

The easiest way for you to use the board is one of two ways:
short of B5 and B6 to ground. Plug it to the USB port and don't modify $7. LEAVE IT AS IT IS
OR
Recompile the firmware with [option DISABLE_ALL_CONTROLS](https://github.com/Paciente8159/uCNC/blob/ac652fc2e7cf7bf66868af8c899e6a74c2fe3829/uCNC/cnc_config.h#L456) enabled. This will deactivate those pins

Juank23, 2023-04-21:
thanks for your prompt reply, I tried the firmware.bin you sent me and it does the same thing, I uncomment in the code // #define DISABLE_ALL_LIMITS and it disables the usb port, I uncomment //#define INVERT_EMERGENCY_STOP
and also the usb port is disabled..... I have 4 blackpill and with all of them I get the same thing,
I also tried with a pair of 10k resistors on pins B5 and B6 and put them to ground and disconnects from the usb port ....
I remain attentive

Paciente8159, 2023-04-21:
That is very weird. What are you using to upload the firmware? How are you uploading?

Juank23, 2023-04-22:
I upload the firmware.bin with Stlink utility, and the source code with VScode, the led flashes but as soon as I ground pin B5 it turns off and disconnects the usb, with the bluepill it does the same thing 
[image]

Juank23, 2023-04-22:
is the same but STM32F401
[image]

Paciente8159, 2023-04-22:
Your blackpill has a [STM32F411](https://docs.platformio.org/en/stable/boards/ststm32/blackpill_f411ce.html#board-ststm32-blackpill-f411ce) and not a [STM32F401](https://docs.platformio.org/en/stable/boards/ststm32/blackpill_f401cc.html#board-ststm32-blackpill-f401cc).
It also runs at a different clock from mine.
Try this to see if it works

modify these [lines on the boardmap_blackpill.h](https://github.com/Paciente8159/uCNC/blob/ac652fc2e7cf7bf66868af8c899e6a74c2fe3829/uCNC/src/hal/boards/stm32/boardmap_blackpill.h#L31) from:

```
#ifndef STM32F401xC
#define STM32F401xC
#endif

#ifndef FLASH_SIZE
#define FLASH_SIZE 0x60000
#endif

#ifndef F_CPU
#define F_CPU 84000000UL
#endif
```
 to

```
#ifndef STM32F411xE
#define STM32F411xE
#endif

#ifndef FLASH_SIZE
#define FLASH_SIZE 0x80000
#endif

#ifndef F_CPU
#define F_CPU 100000000UL
#endif
```

also on the [stm32.ini](https://github.com/Paciente8159/uCNC/blob/ac652fc2e7cf7bf66868af8c899e6a74c2fe3829/uCNC/src/hal/boards/stm32/stm32.ini#L31)

change this:

```
[env:blackpill_f401cc]
extends = common_stm32
board = blackpill_f401cc
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL
```

to this

```
[env:blackpill_f411ce]
extends = common_stm32
board = blackpill_f411ce
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL
```

Let me know if it worked.

Paciente8159, 2023-04-22:
Setting MCU frequency to 100Mhz might result in some compilation errors.
If that's the case keep #define F_CPU 84000000UL unchanged and try again.

I will be investigating this issue and the best solution for board variants. I'll try to get my hands on some different blackpill versions but it might take me a few weeks.

I'll let you know

Paciente8159, 2023-04-22:
I just noticed you said that it's the same with your other SMT32F401 board.

It works just fine in mine. It should work on yours too

Paciente8159, 2023-04-22:
I've started a [branch](https://github.com/Paciente8159/uCNC/tree/365-better-stm32-variants-support) with some modifications to allow easier STM32 variants setup. I've also added BlackpillF411 build variant to PlatformIO. You can give it a try.

If USB is having issues and you have a TTL serial to USB cable you can also use hardware UART1 port @ pins A9-TX and A10-RX as they are enabled in the default build and provide an alternative serial interface.

I've ordered myself a Blackpill F411 and expect to get it in the next 2 or 3 weeks. I'll see if I can come up with a solution for EEPROM emulation in the meantime.

There is an alternative to EEPROM, but that implies setting up the hardware SPI on that board and add an SD card reader and use the sd_card module addon. With an external SD card µCNC stores the settings in the SD card and it's portable from one machine to the other.

Paciente8159, 2023-04-23:
I've implemented EEPROM for STM32F4 [here](https://github.com/Paciente8159/uCNC/tree/stm32f4-eeprom-emulation).

I have to run some more tests before integrate since this may affect all architectures (I modified some core code).
Since STM32F4 memory layout is kind of odd eeprom emulation is done using the last 128K of flash memory which is a waste to emulate 1k of EEPROM. The good side is that this allows the eeprom to be recorded in 1k pages only requiring one flash erase every 128 modifications (or more depending on the bits changed).

I strongly recommend using all devices with emulated EEPROMS with option [ENABLE_EXTRA_SYSTEM_CMDS](https://github.com/Paciente8159/uCNC/blob/6824c50649502197e5f1a72c19d34971838b2ca5/uCNC/cnc_config.h#L501) enabled. This will keep all settings changes in RAM until a [$SS](https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#additional-settings-commands) command is issued.

Juank23, 2023-04-24:
good day, my friend I have not yet managed to make the code work, I inverted the pin B5 by B6 and the result is the same, change this value #define DEFAULT_CONTROL_INV_MASK 1 and it turns out that the controller does not turn on until I put to ground the pin B5 and if I put 
#define DEFAULT_CONTROL_INV_MASK 0, it turns on the controller but when I put the B5 pin to ground it turns off, note that when I touch the pin it tries to connect to the program because it is giving me this error

Paciente8159, 2023-04-24:
Ok.
I think I was able to reproduce the problems you are having. I'm investigating.

Thanks for reporting this.

Paciente8159, 2023-04-24:
I believe I've isolated the issue.
Both USB and UART are enabled. I forgot to enable the pullup in the RX pin of UART.
Left unconnected in a noisy environment it can pickup an insane amount of ghost chars and overflow the RX buffer.

I was not having that issue at home but I pick it up in another location and also by rubbing my finger tips in the pins.
I've opened issue #366 for this and will fix that really quick.

In the mean time please try this new firmware for STM32F401
[firmware.zip](https://github.com/Paciente8159/uCNC/files/11308793/firmware.zip)

 Thanks for your report

Paciente8159, 2023-04-24:
There is something wrong with the version with eeprom implementation of the version I sent you. It was still untested and the modifications introduced some problem I have to investigate.

Some software (like Candle) have some kind of problem query µCNC state. But there is a work around by forcing the status message to be automatically printed via STATUS_AUTOMATIC_REPORT_INTERVAL.

Try this compilation.
This is from the master branch.
I disabled the 4th axis on the blackpill just to avoid show the A limit in the status message.
I also enabled these options:

STATUS_AUTOMATIC_REPORT_INTERVAL 100
DISABLE_ALL_CONTROLS
ENABLE_EXTRA_SYSTEM_CMDS

[firmware.zip](https://github.com/Paciente8159/uCNC/files/11314008/firmware.zip)

I tested this on Candle and was showing the tool updating.
Remember this version (master branch) does not support EEPROM emulation.

Juank23, 2023-04-24:
Amigo buen día, está el firmware.elf , el stlink no lo Lee que pena mi ignorancia quedo atento

Paciente8159, 2023-04-24:
Sorry. Here you go
[firmware.zip](https://github.com/Paciente8159/uCNC/files/11315111/firmware.zip)

Juank23, 2023-04-24:
El compu no reconoce el puerto USB y el LED queda encendido sin parpadear

Paciente8159, 2023-04-24:
I just tested the bin and it's working.
Please try this. Download the master branch
Enable these in cnc_config.h

DISABLE_ALL_CONTROLS
ENABLE_EXTRA_SYSTEM_CMDS
STATUS_AUTOMATIC_REPORT_INTERVAL 100

Compile and upload

Paciente8159, 2023-04-24:
And the new version with EEPROM is now also working.
I've improved clock configuration to fit the MCU variants

Juank23, 2023-04-24:
Bueno pues te cuento que acabo de hacer lo que me dijiste pero no funcionó, no lo reconoce el compu, si invierto emergency stop también se queda sin puerto USB, pero le pongo el pin a tierra y ahí si prende el LED esto está muy raro quedo atento

Paciente8159, 2023-04-24:
I'm finishing modifying the STM32 cores with (I believe) better clocks configuration.
It's basically done for STM32F4. If you want you can try it out downloading the code from this [branch](https://github.com/Paciente8159/uCNC/tree/365-better-stm32-variants-support)

I want to do the same for the STM32F1 too.
I should be able to get it running tomorrow.

I'll feedback you as soon as it's done.
Thanks for you tests and help.

Juank23, 2023-04-24:
Ese código si funcionó con el pines b5 y b6 a tierra ya funciona el controlador aunque ahora a veces se pausa ligeramente al leer los gcode, le hice ENABLE_EXTRA_SYSTEM_CMDS
STATUS_AUTOMATIC_REPORT_INTERVAL 100 y funcionó mejor

Paciente8159, 2023-04-24:
I had the gcode hickup issue before these last clock changes. Can you provide your gcode file so I can run tests with it tomorrow please?
This way could try to replicate the behaviour in my blackpill

Juank23, 2023-04-25:
Ready friend, next I send you some gcodes and an image of what happens when the gcode ends I remain attentive .... 
[Gcodes.zip](https://github.com/Paciente8159/uCNC/files/11315437/Gcodes.zip)

Paciente8159, 2023-04-25:
Thanks. As far as I can tell the GCode is running smooth on my blackpill. I just did both 3d1 and 3d2 files and it went all good.
No stuttering.
Please do a $$ command to print the settings you are using and send me so I can reproduce your conditions too.

Paciente8159, 2023-04-25:
I have finished the code changes and they are ready to be integrated.
Be sure to check the [compilation notes in the STM32 boards wiki](https://github.com/Paciente8159/uCNC/wiki/STM32Fx:-Compiling#important-note) regarding flash size definition.

If you find any issue, please be free to report them and I'll try to address them ASAP.

closed 2023-04-25 by Paciente8159

Juank23, 2023-04-25:
ok my friend I will try and let you know thanks for your help.

Paciente8159, 2023-05-03:
Hi there!
I received my stm32F411 today.
I was able to load µCNC and run go code on it.
I did not do any extensive testing on it but it seems to be working ok so the variant optimization seems to had the desired effect.

I'm also developing a new feature and during the development I found a code optimization that should be able to improve µCNC status responses to command `?`
I've not tested but maybe this will fix detection issues by some programs.

Thanks