Paciente8159/uCNC #389: [BUG] expected unqualified-id before '-' token

timint, 2023-05-08:
Hi, I'm trying to compile this thing on Arduino Uno but receive the following error:

> In file included from R:\arduino_modified_sketch_389122\sketch_may08a.ino:2:0:
> C:\Users\JohnDoe\Documents\Arduino\libraries\uCNC/build_opt.h:1:1: error: expected unqualified-id before '-' token
>  -DHAL_TIM_MODULE_DISABLED -DHAL_EXTI_MODULE_DISABLED
>  ^
> exit status 1
> Error compiling for board Arduino Uno.

Is the Arduino compiler no longer supported for this project?

timint, 2023-05-08:
So, I tried compiling with PlatformIO using Visual Studio Code and I got this fatal error:

>  *  Executing task: C:\Users\JohnDoe\.platformio\penv\Scripts\platformio.exe run --environment uno_shield_v3 
> 
> Can not remove temporary directory `C:\Users\JohnDoe\Documents\Arduino\libraries\uCNC\.pio\build`. Please remove it manually to avoid build issues
> Processing uno_shield_v3 (board: uno; platform: atmelavr; framework: arduino)
> ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> FileNotFoundError: [WinError 2] Cannot find the file: 'C:\\Users\\JohnDoe\\Documents\\Arduino\\libraries\\uCNC\\.pio':
>   File "C:\Users\JohnDoe\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
>     os.makedirs(env.subst("$BUILD_DIR"))
>   File "C:\Users\JohnDoe\.platformio\python3\lib\os.py", line 215:
>     makedirs(head, exist_ok=exist_ok)
>   File "C:\Users\JohnDoe\.platformio\python3\lib\os.py", line 215:
>     makedirs(head, exist_ok=exist_ok)
>   File "C:\Users\JohnDoe\.platformio\python3\lib\os.py", line 225:
>     mkdir(name, mode)
> ======================================================================================================== [FAILED] Took 0.54 seconds ========================================================================================================
> 
> Environment    Status    Duration
> -------------  --------  ------------
> uno_shield_v3  FAILED    00:00:00.543
> =================================================================================================== 1 failed, 0 succeeded in 00:00:00.543 ===================================================================================================
> 
>  *  The terminal process "C:\Users\JohnDoe\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'uno_shield_v3'" terminated with exit code: 1. 

Note that there is no directory C:\Users\JohnDoe\Documents\Arduino\libraries\uCNC\.pio\build

Paciente8159, 2023-05-08:
I there.
I am able to compile the project just fine in both Arduino IDE and PlatformIO.

What version are you using?
Did you made any changes?

timint, 2023-05-09:
>This does not look like a correct path

Github didn't render the backslash properly when used inside a quote. This is the path with forwards slashes instead `C:/Users/JohnDoe/.platformio/...`

At first I used Arduino IDE 1.8.x. That didn't work, so I thought I found version 2 on their website. Same errors there.

I put uCNC in the library tree and included it in a blank new project.
[image]

If I remove the line build_opt line I get the following error:

```
/ccqIATMb.ltrans0.ltrans.o: In function `main':
C:/Users/JohnDoe/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/main.cpp:43: undefined reference to `setup'
C:/Users/JohnDoe/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

Paciente8159, 2023-05-09:
I think you are not using correctly.
µCNC is not an Arduino library... although it could be used as one with some limitations.
Open the uCNC.ino sketch inside the uCNC directory and compile it.

timint, 2023-05-09:
@Paciente8159 Thank you that was the problem. I came from grbl looking for something more maintained and up to date and this project looked to be it. This is all a learning curve for me and I didn't reflect how things were made differently. I will see if I can provide some wiki instructions for the next newcomer. Appreciate you pointing me in the right direction.

closed 2023-05-09 by timint

Paciente8159, 2023-05-09:
Glad I could help.
You can go this project wiki. There is lots of info there and almost all up to date.

You can find compilation info for [AVR here](https://github.com/Paciente8159/uCNC/wiki/AVR:-Compiling)