Paciente8159/uCNC #412: Doesn't compile on Arduino-ide

honeybadger2023, 2023-06-04:
Looks like the code doesn't compile on Arduino-Ide anymore. So many things have changed.

```
/arduino/current/Arduino/uCNC-master/uCNC/src/hal/mcus/avr/mcumap_avr.h:42:10: fatal error: io.h: No such file or directory
 #include <avr/io.h>
```

Tried with Arduino 1.8.19 for an ESP32.

honeybadger2023, 2023-06-04:
Compiles with platform.io in visual studio.

Paciente8159, 2023-06-05:
Hi there. What board are you trying to compile for?

honeybadger2023, 2023-06-05:
ESP32 WROOM-32 Developer Board.

I'm not sure but think that with recent updates in Arduino-Ide so much things have changed that the includes to /avr/xyz and maybe others fail. I don't have an avr folder. Some libraries are now integrated or moved to somewhere else.

But platform.io works. So this problem is closed for me.

Other question: What is the maximum pulse rate for the library?

Paciente8159, 2023-06-05:
Ok. Just 2 notes regarding Arduino IDE. This is not a library. You should open the uCNC.ino sketch that is inside the uCNC directory and compile that sketch. Don't forget to edit cnc_config.h  BOARD to the board you want to use. Compiled it under Arduino IDE 1.8.19 and worked fine (some warnings but they are OK)

About pulse limit on that board. I've only performed full benchmark tests on Arduino UNO.

On ESP32 and ARM I only did some basic tests and did not record the results but those boards are firmware limited to [100K pulses per second](https://github.com/Paciente8159/uCNC/blob/a65778a5267f9c1c5a515e5d1b3ce691bf8be1eb/uCNC/src/hal/mcus/esp32/mcumap_esp32.h#L52) in the formware. You can override that setting by defining your own value for `F_STEP_MAX` (on cnc_config.h for example) like this.

```
// set max step rate
#define F_STEP_MAX 200000
```

honeybadger2023, 2023-06-05:
Nice. The library looks very goo and clean. Well done.