Paciente8159/uCNC #415: [BUG]

andy0983, 2023-06-09:
Problem:
The display shows the µCNC logo and v 1.7.1
I tried a few settings but nothing helped
Please help

My equipment is:
Arduino MEGA/RAMPS 1.4
FULL_GRAPHIC_SMART_CONTROLLER

My "Grbl/µCNC" settings:
cnc_hal_config.h
// assign the tools from 1 to 16
#define TOOL1 spindle_pwm
// #define TOOL2 laser
#define TOOL3 laser_ppi
// #define TOOL4 spindle_besc
// #define TOOL5 spindle_relay

#define SPINDLE_PWM_HAS_RPM_ENCODER
#define ENCODERS 1
#define ENC0_PULSE DIN18
#define ENC0_DIR DIN17
#define ENABLE_ENCODER_RPM
#define RPM_PPR 24

#define U8X8_MSG_GPIO_SPI_CLOCK_PIN DOUT4
#define U8X8_MSG_GPIO_SPI_DATA_PIN DOUT5
#define U8X8_MSG_GPIO_CS_PIN DOUT6
#define GRAPHIC_DISPLAY_ENCODER_BTN DIN16
#define GRAPHIC_DISPLAY_ENCODER_ENC1 DIN17
#define GRAPHIC_DISPLAY_ENCODER_ENC2 DIN18
#define GRAPHIC_DISPLAY_BEEP DOUT7
#define GRAPHIC_DISPLAY_MAX_LINES 5

The configuration is standard up to RAMPS 1.4:

boardmap_ramps14.h
//////lcd///////////
//clk
#define DOUT4_BIT 1
#define DOUT4_PORT A
//date
#define DOUT5_BIT 0
#define DOUT5_PORT H
// cs
#define DOUT6_BIT 1
#define DOUT6_PORT H
//beep
#define DOUT7_BIT 0
#define DOUT7_PORT C
//enc btn
#define DIN16_BIT 2
#define DIN16_PORT C
#define DIN16_PULLUP
//enc 1
#define DIN17_BIT 4
#define DIN17_PORT C
#define DIN17_PULLUP
//enc 2
#define DIN18_BIT 6
#define DIN18_PORT C
#define DIN18_PULLUP

Paciente8159, 2023-06-09:
Thanks for your report. I will investigate.

Paciente8159, 2023-06-09:
Did you enabled `ENABLE_MAIN_LOOP_MODULES` ?

Paciente8159, 2023-06-09:
There are also several definitions that are not correct

Only DIN0 to DIN7 are ISR capable
You need to define one of those DIN pins in your board and enable ISR for that pin
Then you assign it to ENC0_PULSE

Please read this part in the [wiki](https://github.com/Paciente8159/uCNC/wiki/Customizing-the-HAL-file#%C2%B5cnc-encoder-module)

Also this is wrong

```
#define U8X8_MSG_GPIO_SPI_CLOCK_PIN DOUT4
#define U8X8_MSG_GPIO_SPI_DATA_PIN DOUT5
#define U8X8_MSG_GPIO_CS_PIN DOUT6
```

You need to replace them by these ones
```
#define GRAPHIC_DISPLAY_SPI_CLOCK DOUT4
#define GRAPHIC_DISPLAY_SPI_DATA DOUT5
#define GRAPHIC_DISPLAY_SPI_CS DOUT6
```

Can you do a fork with your config with all the modules you want to install so I can check it?

andy0983, 2023-06-09:
Thanks for the tips
ENABLE_MAIN_LOOP_MODULES is enabled.  
For now, I have this one module for the LCD

andy0983, 2023-06-09:
[image]

andy0983, 2023-06-09:
This screen is on all the time

I made changes:
#define GRAPHIC_DISPLAY_SPI_CLOCK DOUT4
#define GRAPHIC_DISPLAY_SPI_DATA DOUT5
#define GRAPHIC_DISPLAY_SPI_CS DOUT6

#define GRAPHIC_DISPLAY_ENCODER_BTN DIN8
#define GRAPHIC_DISPLAY_ENCODER_ENC1 DIN0
#define GRAPHIC_DISPLAY_ENCODER_ENC2 DIN9
#define GRAPHIC_DISPLAY_MAX_LINES 5

//////lcd///////////
//pins for smart adapter
//clk
#define DOUT4_BIT 1
#define DOUT4_PORT A
//data
#define DOUT5_BIT 0
#define DOUT5_PORT H
//cs
#define DOUT6_BIT 1
#define DOUT6_PORT H
//beep
#define DOUT7_BIT 0
#define DOUT7_PORT C
//enc btn
#define DIN8_BIT 2
#define DIN8_PORT C
#define DIN8_PULLUP
//enc 1
#define DIN0_BIT 5
#define DIN0_PORT E
#define DIN0_ISR -6
#define DIN0_PULLUP
//enc 2
#define DIN9_BIT 6
#define DIN9_PORT C
#define DIN9_PULLUP

After changing it still doesn't work

andy0983, 2023-06-09:
#define ENC0_PULSE DIN0
 #define ENC0_DIR DIN8

Paciente8159, 2023-06-09:
I tested and the display seems to work.
Try compiling this code
[uCNC-1.7.1.zip](https://github.com/Paciente8159/uCNC/files/11710594/uCNC-1.7.1.zip)

You can check the changes I've made there.
Just another note. Tools must be sequential.

You can define TOOL1 and TOOL2 but TOOL1 and TOOL 3 is not possible. I'll probably change this in the future.

andy0983, 2023-06-11:
Thanks for your help, I had the wrong setting:
// assign the tools from 1 to 16
#define TOOL1 spindle_pwm
// #define TOOL2 laser
#define TOOL3 laser_ppi
// #define TOOL4 spindle_besc
// #define TOOL5 spindle_relay

Correctly it should be like this:
// assign the tools from 1 to 16
#define TOOL1 spindle_pwm
// #define TOOL2 laser
#define TOOL2 laser_ppi
// #define TOOL4 spindle_besc
// #define TOOL5 spindle_relay

After changing these settings it started working properly

closed 2023-06-11 by andy0983

Paciente8159, 2023-06-11:
Any problem please report. Thanks