Paciente8159/uCNC #429: [BUG] Overloading the command buffer blocks communication completely.

etet100, 2023-06-24:
**Describe the bug**
Sending too many commands blocks communication. This refers to the situation when the system stops sending OK immediately and we send even more commands. Some of the commands are executed, the system returns to idle state but no further command is accepted.

**What is your hardware**
Arduino Nano (ATMEGA328P)

**What was the GCode running**
```
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
```

Paciente8159, 2023-06-24:
I think this might be related to #428 and was probably introduced by #425.
I'm investigating this.
Thanks

Paciente8159, 2023-06-25:
Protocol needs some sort of revision and a clearer call path.

When starting to print a response the protocol output is not locked allowing for reentrant main loop calls to print other stuff as well. The protocol functions need a lock flag to prevent these situations.
I'll review the protocol compilation unit in the next couple of days to address this issue.

Paciente8159, 2023-06-25:
was that all the code you were running?
I tested it on lasergrbl with the proposed fix and all went ok on uno board.

Paciente8159, 2023-06-25:
I'm reopening this until you can confirm if the issue is solved or not

etet100, 2023-06-25:
I don't see much difference. Last log:
```
$X
[MSG:Caution: Unlocked]
ok
$H
>:ok
>:ok
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300
ok
G90 X100 Y120 F300
ok
G90 X100 Y150 F300 
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
G90 X100 Y150 F300
G90 X100 Y120 F300
ok
? 
<Idle|MPos:100.000,150.000,0.000|FS:0.000,0|WCO:0.000,0.000,0.000>
?
<Idle|MPos:100.000,150.000,0.000|FS:0.000,0|Ov:100,100,100>
?
<Idle|MPos:100.000,150.000,0.000|FS:0.000,0>
$H << command ignored
?
<Idle|MPos:100.000,150.000,0.000|FS:0.000,0>
$X << command ignored
aa << ignored
fsafdas << ignored
sads << ignored
```

Paciente8159, 2023-06-25:
What program are you using?
If you are sending all command in a console at once that is normal.
The buffer has a limit of 128 bytes.
Usually programs like lasergrbl and others try to keep the buffer as full as possible but manage that to not overflow the buffer.

etet100, 2023-06-25:
I know. But after system returns to idle i should be able to send more commands.

Paciente8159, 2023-06-25:
I'll run some more tests.
Error 11 is not being issued.
Thanks

Paciente8159, 2023-06-26:
The new patch prevent buffer locking in case of overflow

Paciente8159, 2023-06-26:
I did some tests, and although it unlocks the buffer it takes a while, beacuse it keeps processing the data in the buffer until it reaches the point were the the overflow occurred (I guess that is the expected), but in that time the buffer is locked depending on the complexity of the commands still waiting to be dispached.

I'll see if I can find a better way that can cause the main loop to discard the buffer and relase the buffer earlier.

Paciente8159, 2023-06-26:
It's confirmed. While the planner is full and the last gcode is not fully processed the buffer overflow marker is not detected. Only after trying to read a new command from the buffer is the overflow marker detected and the whole buffer is flushed since all the data at that point cannot be trusted.

After this the buffer gets cleared and the overflow error is sent. At this point it's possible to send new commands.

I'm still going to do some more test, but this should be merged in the master branch.

closed 2023-06-26 by Paciente8159