Paciente8159/uCNC #465: [FR] TMC2209/TMC2226 read and write data by UART address

bwenquan, 2023-08-06:
TMC2209/TMC2226 only the packaging is different. All other parameters are exactly the same.

The uart of tmc2209/tmc2226 supports addressing, and the address is set by ms1 and ms2:
[image]

It can be connect like this:
[image]

According to the manual, just modify the communication data to read and write by different addresses: 
[image]
[image]

This is very useful for mcu with few pins. I have a 7 drivers machine that is made of 4z, 1x, and 2y.

Thanks
:)

Paciente8159, 2023-08-07:
Ok thanks.
Yes currently µCNC only uses slave address 0 for all TMC modules but that should be easy and quick to change with an option for each IC address.

Which version of µCNC are you using? I'll try to provide you a modified version with that option.

Paciente8159, 2023-08-07:
I've added a proposed patch for version 1.7.3. I will later add it to the upcoming version (1.8).
If you are using a different version you can also modify the addresses your self by changing the value of address for each stepper in the tmcdriver.c file:
For example, here is the [stepper 0 slave address line](https://github.com/Paciente8159/uCNC/blob/1858a94ed63a3483e0eeab8c65d15cf47ae5374c/uCNC/src/modules/tmcdriver.c#L1278). If you navigate a little bellow that you will find the remaining slave addresses for the other steppers.

bwenquan, 2023-08-07:
Thanks, I'm trying releases of `µCNC v1.8.0-beta` and will report if there are any issues. 

Another question, does each axis support only 2 drives? How should I configure my 4z machine like voron 2.4? [(Photo by Voron Design)](https://www.icloud.com/sharedalbum/zh-cn/#B0H53qWtHnblTC;A99F9CE3-7FA1-4BB6-9248-4E22889715D2)

bwenquan, 2023-08-07:
Thanks, but I think it's unnecessary to spend too many your time to do this.

Yes, I'm DIY a CO2 laser machine, and I have an 8 axis motherboard (I designed it long time ago for a 3d printer, and the mcu is stm32f103v8t6), so each motor can move independently.  

The machine has a  laser-head(tool), 1 driver for the x-axis, and 2 drivers for the y-axis to automatically align. The y-axis needs 2 motors to drive it Because it is too long.

I have a micro-switch on the laser-head, move the laser-head near the motor and get the z-axis height, after getting different 4 values  it can calculate the height difference and move z motor independently. 

Then I can adjust the Z height based on my material thickness to make the machined surface always in the focus of the laser.

But now, there are many problems whem I compile 1.8.0-beta firmware. I'll make new issue later.

Paciente8159, 2023-08-07:
Thanks.

It's not a waste of time. Actually I may have spotted an opportunity make several improvements with this modification:
- simplify ganged motors motion and selfsquaring
- allow selfsquaring on all 3 axis simultaneously
- allow multi-axis homing
- unify limit detection while homing for all kinematics

But even that might not work for you. This assumes that you have independent limits for each motor (in your case 4 limits in z - one for each motor). For the Z plane do you have a limit switch for each bed corner, or use the switch in the tool-head?

An alternative would be to use µCNC embedded height mapping. Instead of making the plane horizontal it would move the tool up and down to compensate for the plane tilting.

Again web build is kind of buggy. If you need help configuring the firmware please let me know and I'll try to assist you getting it done.

closed 2023-08-09 by Paciente8159

Paciente8159, 2023-08-11:
As mentioned a new multi-axis solution is now for testing in PR #477.

This new PR at least expands from the possibility of configuring from mono and bi-motor setups to poly-motor setups (2, 4, etc motors on a single axis).

It's not a direct solution for something like the Voron because the Voron uses the probe as a Z limit and bed leveling probe but it's a start.

This opens the possibility to perform the bed leveling via probe with a custom extension module (that might be added in the future).