Paciente8159/uCNC #479: [Help]

bwenquan, 2023-08-22:
I tried 1.8.0 but it did not work. It did not respond anything to the serial port. And the crystal oscillator is not working

I tried 1.7.4 but my motor won't move, I've done everything I can to try and fix it, but it not work until now. So I need some help.

Schematic:
[image]
[image]

My config:
```c++
#ifndef BOADMAP_OVERRIDES_H
#define BOADMAP_OVERRIDES_H
#ifdef __cplusplus
extern "C"
{
#endif
#ifdef MCU
#undef MCU
#endif
#define MCU MCU_STM32F1X
#ifdef KINEMATIC
#undef KINEMATIC
#endif
#define KINEMATIC KINEMATIC_CARTESIAN
#ifdef AXIS_COUNT
#undef AXIS_COUNT
#endif
#define AXIS_COUNT 2
#ifdef TOOL_COUNT
#undef TOOL_COUNT
#endif
#define TOOL_COUNT 1
#ifdef BAUDRATE
#undef BAUDRATE
#endif
#define BAUDRATE 115200
#ifdef S_CURVE_ACCELERATION_LEVEL
#undef S_CURVE_ACCELERATION_LEVEL
#endif
#define S_CURVE_ACCELERATION_LEVEL 0
#ifdef BOARD
#undef BOARD
#endif
#define BOARD BOARD_CUSTOM
#ifdef BOARD_NAME
#undef BOARD_NAME
#endif
#define BOARD_NAME "Kirizu-laser"
#ifdef ITP_TIMER
#undef ITP_TIMER
#endif
#define ITP_TIMER 2
#ifdef SERVO_TIMER
#undef SERVO_TIMER
#endif
#define SERVO_TIMER 3
#ifdef ONESHOT_TIMER
#undef ONESHOT_TIMER
#endif
#define ONESHOT_TIMER 1
#ifdef STEP0_BIT
#undef STEP0_BIT
#endif
#define STEP0_BIT 4
#ifdef STEP0_PORT
#undef STEP0_PORT
#endif
#define STEP0_PORT B
#ifdef STEP0_IO_OFFSET
#undef STEP0_IO_OFFSET
[... 2345 lines cut ...]
[O:48:0]
[O:49:0]
[O:50:0]
[O:51:0]
[O:52:0]
[O:53:0]
[O:54:0]
[O:55:0]
[O:57:1]
[O:58:1]
[SI:100:0]
[SI:101:0]
[SI:104:0]
[A:114:3]
[A:115:255]
[A:116:255]
[A:117:255]
[I:130:1]
[I:140:1]
[I:141:1]
[STEPS:0,0]
[RUNTIME:154398]
ok
ok
```

bwenquan, 2023-08-22:
The board and TMC2209 I have tried on the klipper firmware, it works fine.
This is klipper config:
```
[include mainsail.cfg]
[virtual_sdcard]
path: /home/kirizu/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 60
max_accel: 2000
max_z_accel: 30
max_z_velocity: 20

[safe_z_home]
home_xy_position: 0,0
z_hop: 5

[probe]
pin: PC13
z_offset: 6

[output_pin TOOL]
pin: !PB1       # use your fan's pin number
pwm: True
hardware_pwm: True
cycle_time: 0.00001
shutdown_value: 0
maximum_mcu_duration: 5
# Default: 0 (disabled)
# Amount of time in which the host has to acknowledge
# a non-shutdown output value.
# Suggested value is around 5 seconds.
# Use a value that does not burn up your stock.
# Please note that during homing, your tool
# needs to be in default speed.

[gcode_macro M3]
gcode:
    {% set S = params.S|default(0.0)|float %}
    SET_PIN PIN=TOOL VALUE={S / 255.0}

[gcode_macro M4]
gcode:
    {% set S = params.S|default(0.0)|float %}
    SET_PIN PIN=TOOL VALUE={S / 255.0}

[gcode_macro M5]
gcode:
    SET_PIN PIN=TOOL VALUE=0

[stepper_x]
step_pin: PB4
dir_pin: !PB5
enable_pin: !PB7
rotation_distance: 40
homing_speed: 20
microsteps: 16
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_x:virtual_endstop
homing_retract_dist: 0
position_min: 0
position_endstop: 0
position_max: 310

[tmc2209 stepper_x]
uart_pin: PA3
#tx_pin: PA2
interpolate: True
run_current: 0.6
uart_address: 2
diag_pin: ^PB6
sense_resistor: 0.110
stealthchop_threshold: 0
driver_SGTHRS: 23

[stepper_y]
step_pin: PD5
dir_pin: !PD6
enable_pin: !PB3
rotation_distance: 40	
microsteps: 16
full_steps_per_rotation: 200 
endstop_pin: PE5
position_min: 0	
position_endstop: 0	
position_max: 312	
homing_speed: 40
homing_retract_dist: 5.0
homing_retract_speed: 20
second_homing_speed: 20	

[tmc2209 stepper_y]
uart_pin: PA3
#tx_pin: PA2
interpolate: True
run_current: 0.4
hold_current: 0.4
uart_address: 0
diag_pin: ^PD7
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_y1]
step_pin: PD1
dir_pin: !PD2
enable_pin: !PD4
microsteps: 16
rotation_distance: 40
endstop_pin: PE6
full_steps_per_rotation: 200

[tmc2209 stepper_y1]
uart_pin: PA3
#tx_pin: PA2
interpolate: True
run_current: 0.4
hold_current: 0.4
uart_address: 1
diag_pin: ^PD3
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_z]
step_pin: PC11
dir_pin: PC10		
enable_pin: !PD0		
rotation_distance: 8	
microsteps: 16			
full_steps_per_rotation: 200 
endstop_pin: probe:z_virtual_endstop	
position_max: 200		
homing_speed: 5		

[tmc2209 stepper_z]
uart_pin: PA3
#tx_pin: PA2
interpolate: True
run_current: 1
hold_current: 1
uart_address: 3
diag_pin: ^PC12
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_z1]
step_pin: PA11		
dir_pin: PA8			
enable_pin: !PA15			
rotation_distance: 8	
microsteps: 16		
full_steps_per_rotation: 200
endstop_pin: probe:z_virtual_endstop

[tmc2209 stepper_z1]
uart_pin: PB10
#tx_pin: PB11
interpolate: True
run_current: 1
hold_current: 1
uart_address: 0
diag_pin: ^PA12
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_z2]
step_pin: PC7
dir_pin: !PC6		
enable_pin: !PC9			
rotation_distance: 8		
microsteps: 16				
full_steps_per_rotation: 200 
endstop_pin: probe:z_virtual_endstop			

[tmc2209 stepper_z2]
uart_pin: PB10
#tx_pin: PB11
interpolate: True
run_current: 1
hold_current: 1
uart_address: 2
diag_pin: ^PC8
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_z3]
step_pin: PD10				
dir_pin: !PD9				
enable_pin: !PD12			
rotation_distance: 8		
microsteps: 16				
full_steps_per_rotation: 200 
endstop_pin: probe:z_virtual_endstop			

[tmc2209 stepper_z3]
uart_pin: PB10
#tx_pin: PB11
interpolate: True
run_current: 1
hold_current: 1
uart_address: 1
diag_pin: ^PD11
sense_resistor: 0.110
stealthchop_threshold: 0

[force_move]
enable_force_move: True

[temperature_sensor mcu_temp]
sensor_type: temperature_mcu

[temperature_sensor host_temp]
sensor_type: temperature_host

```

Paciente8159, 2023-08-22:
Thanks for the report. I will check it.

Paciente8159, 2023-08-22:
Could you submit the Web Builder Config Setting (json) file?

bwenquan, 2023-08-23:
```json

{"VERSION":10780,"MCU":"MCU_STM32F1X","KINEMATIC":"KINEMATIC_CARTESIAN","AXIS_COUNT":3,"TOOL_COUNT":1,"BAUDRATE":115200,"ENABLE_COOLANT":true,"ENABLE_WIFI":null,"ENABLE_BLUETOOTH":null,"ENABLE_SKEW_COMPENSATION":null,"SKEW_COMPENSATION_XY_ONLY":null,"ENABLE_LINACT_PLANNER":null,"ENABLE_LINACT_COLD_START":null,"ENABLE_BACKLASH_COMPENSATION":null,"S_CURVE_ACCELERATION_LEVEL":0,"BRESENHAM_16BIT":true,"ENABLE_EXTRA_SYSTEM_CMDS":true,"RAM_ONLY_SETTINGS":null,"STATUS_AUTOMATIC_REPORT_INTERVAL_ENABLE":null,"BOARD":"BOARD_CUSTOM","BOARD_NAME":"\"Kirizu-laser\"","ITP_TIMER":2,"SERVO_TIMER":3,"ONESHOT_TIMER":1,"STEP0_BIT":4,"STEP0_PORT":"B","STEP0_IO_OFFSET":null,"STEP1_BIT":5,"STEP1_PORT":"D","STEP1_IO_OFFSET":null,"STEP2_BIT":1,"STEP2_PORT":"D","STEP2_IO_OFFSET":null,"STEP3_BIT":11,"STEP3_PORT":"C","STEP3_IO_OFFSET":null,"STEP4_BIT":11,"STEP4_PORT":"A","STEP4_IO_OFFSET":null,"STEP5_BIT":7,"STEP5_PORT":"C","STEP5_IO_OFFSET":null,"STEP6_BIT":10,"STEP6_PORT":"D","STEP6_IO_OFFSET":null,"STEP7_BIT":14,"STEP7_PORT":"D","STEP7_IO_OFFSET":null,"DIR0_BIT":5,"DIR0_PORT":"B","DIR0_IO_OFFSET":null,"DIR1_BIT":6,"DIR1_PORT":"D","DIR1_IO_OFFSET":null,"DIR2_BIT":2,"DIR2_PORT":"D","DIR2_IO_OFFSET":null,"DIR3_BIT":10,"DIR3_PORT":"C","DIR3_IO_OFFSET":null,"DIR4_BIT":8,"DIR4_PORT":"A","DIR4_IO_OFFSET":null,"DIR5_BIT":6,"DIR5_PORT":"C","DIR5_IO_OFFSET":null,"DIR6_BIT":9,"DIR6_PORT":"D","DIR6_IO_OFFSET":null,"DIR7_BIT":13,"DIR7_PORT":"D","DIR7_IO_OFFSET":null,"STEP0_EN_BIT":7,"STEP0_EN_PORT":"B","STEP0_EN_IO_OFFSET":null,"STEP1_EN_BIT":3,"STEP1_EN_PORT":"B","STEP1_EN_IO_OFFSET":null,"STEP2_EN_BIT":4,"STEP2_EN_PORT":"D","STEP2_EN_IO_OFFSET":null,"STEP3_EN_BIT":0,"STEP3_EN_PORT":"D","STEP3_EN_IO_OFFSET":null,"STEP4_EN_BIT":15,"STEP4_EN_PORT":"A","STEP4_EN_IO_OFFSET":null,"STEP5_EN_BIT":9,"STEP5_EN_PORT":"C","STEP5_EN_IO_OFFSET":null,"STEP6_EN_BIT":12,"STEP6_EN_PORT":"D","STEP6_EN_IO_OFFSET":null,"STEP7_EN_BIT":15,"STEP7_EN_PORT":"D","STEP7_EN_IO_OFFSET":null,"PWM0_BIT":1,"PWM0_PORT":"B","PWM0_CHANNEL":4,"PWM0_TIMER":3,"PWM0_MUX":null,"PWM0_IO_OFFSET":null,"PWM1_BIT":0,"PWM1_PORT":"B","PWM1_CHANNEL":3,"PWM1_TIMER":3,"PWM1_MUX":null,"PWM1_IO_OFFSET":null,"PWM2_BIT":null,"PWM2_PORT":null,"PWM2_CHANNEL":null,"PWM2_TIMER":null,"PWM2_MUX":null,"PWM2_IO_OFFSET":null,"PWM3_BIT":null,"PWM3_PORT":null,"PWM3_CHANNEL":null,"PWM3_TIMER":null,"PWM3_MUX":null,"PWM3_IO_OFFSET":null,"PWM4_BIT":null,"PWM4_PORT":null,"PWM4_CHANNEL":null,"PWM4_TIMER":null,"PWM4_MUX":null,"PWM4_IO_OFFSET":null,"PWM5_BIT":null,"PWM5_PORT":null,"PWM5_CHANNEL":null,"PWM5_TIMER":null,"PWM5_MUX":null,"PWM5_IO_OFFSET":null,"PWM6_BIT":null,"PWM6_PORT":null,"PWM6_CHANNEL":null,"PWM6_TIMER":null,"PWM6_MUX":null,"PWM6_IO_OFFSET":null,"PWM7_BIT":null,"PWM7_PORT":null,"PWM7_CHANNEL":null,"PWM7_TIMER":null,"PWM7_MUX":null,"PWM7_IO_OFFSET":null,"PWM8_BIT":null,"PWM8_PORT":null,"PWM8_CHANNEL":null,"PWM8_TIMER":null,"PWM8_MUX":null,"PWM8_IO_OFFSET":null,"PWM9_BIT":null,"PWM9_PORT":null,"PWM9_CHANNEL":null,"PWM9_TIMER":null,"PWM9_MUX":null,"PWM9_IO_OFFSET":null,"PWM10_BIT":null,"PWM10_PORT":null,"PWM10_CHANNEL":null,"PWM10_TIMER":null,"PWM10_MUX":null,"PWM10_IO_OFFSET":null,"PWM11_BIT":null,"PWM11_PORT":null,"PWM11_CHANNEL":null,"PWM11_TIMER":null,"PWM11_MUX":null,"PWM11_IO_OFFSET":null,"PWM12_BIT":null,"PWM12_PORT":null,"PWM12_CHANNEL":null,"PWM12_TIMER":null,"PWM12_MUX":null,"PWM12_IO_OFFSET":null,"PWM13_BIT":null,"PWM13_PORT":null,"PWM13_CHANNEL":null,"PWM13_TIMER":null,"PWM13_MUX":null,"PWM13_IO_OFFSET":null,"PWM14_BIT":null,"PWM14_PORT":null,"PWM14_CHANNEL":null,"PWM14_TIMER":null,"PWM14_MUX":null,"PWM14_IO_OFFSET":null,"PWM15_BIT":null,"PWM15_PORT":null,"PWM15_CHANNEL":null,"PWM15_TIMER":null,"PWM15_MUX":null,"PWM15_IO_OFFSET":null,"SERVO0_BIT":null,"SERVO0_PORT":null,"SERVO0_IO_OFFSET":null,"SERVO1_BIT":null,"SERVO1_PORT":null,"SERVO1_IO_OFFSET":null,"SERVO2_BIT":null,"SERVO2_PORT":null,"SERVO2_IO_OFFSET":null,"SERVO3_BIT":null,"SERVO3_PORT":null,"SERVO3_IO_OFFSET":null,"SERVO4_BIT":null,"SERVO4_PORT":null,"SERVO4_IO_OFFSET":null,"SERVO5_BIT":null,"SERVO5_PORT":null,"SERVO5_IO_OFFSET":null,"DOUT0_BIT":5,"DOUT0_PORT":"C","DOUT0_IO_OFFSET":null,"DOUT1_BIT":4,"DOUT1_PORT":"C","DOUT1_IO_OFFSET":null,"DOUT2_BIT":9,"DOUT2_PORT":"E","DOUT2_IO_OFFSET":null,"DOUT3_BIT":8,"DOUT3_PORT":"E","DOUT3_IO_OFFSET":null,"DOUT4_BIT":7,"DOUT4_PORT":"E","DOUT4_IO_OFFSET":null,"DOUT5_BIT":13,"DOUT5_PORT":"E","DOUT5_IO_OFFSET":null,"DOUT6_BIT":12,"DOUT6_PORT":"E","DOUT6_IO_OFFSET":null,"DOUT7_BIT":11,"DOUT7_PORT":"E","DOUT7_IO_OFFSET":null,"DOUT8_BIT":10,"DOUT8_PORT":"E","DOUT8_IO_OFFSET":null,"DOUT9_BIT":null,"DOUT9_PORT":null,"DOUT9_IO_OFFSET":null,"DOUT10_BIT":2,"DOUT10_PORT":"A","DOUT10_IO_OFFSET":null,"DOUT11_BIT":11,"DOUT11_PORT":"B","DOUT11_IO_OFFSET":null,"DOUT12_BIT":null,"DOUT12_PORT":null,"DOUT12_IO_OFFSET":null,"DOUT13_BIT":null,"DOUT13_PORT":null,"DOUT13_IO_OFFSET":null,"DOUT14_BIT":null,"DOUT14_PORT":null,"DOUT14_IO_OFFSET":null,"DOUT15_BIT":null,"DOUT15_PORT":null,"DOUT15_IO_OFFSET":null,"DOUT16_BIT":null,"DOUT16_PORT":null,"DOUT16_IO_OFFSET":null,"DOUT17_BIT":null,"DOUT17_PORT":null,"DOUT17_IO_OFFSET":null,"DOUT18_BIT":null,"DOUT18_PORT":null,"DOUT18_IO_OFFSET":null,"DOUT19_BIT":null,"DOUT19_PORT":null,"DOUT19_IO_OFFSET":null,"DOUT20_BIT":null,"DOUT20_PORT":null,"DOUT20_IO_OFFSET":null,"DOUT21_BIT":null,"DOUT21_PORT":null,"DOUT21_IO_OFFSET":null,"DOUT22_BIT":null,"DOUT22_PORT":null,"DOUT22_IO_OFFSET":null,"DOUT23_BIT":null,"DOUT23_PORT":null,"DOUT23_IO_OFFSET":null,"DOUT24_B [... cut ...] ull,"STEPPER7_STALL_SENSITIVITY":10,"ENABLE_MAIN_LOOP_MODULES":true,"ENABLE_IO_MODULES":true,"ENABLE_PARSER_MODULES":true,"ENABLE_MOTION_CONTROL_MODULES":true,"ENABLE_SETTINGS_MODULES":true,"ENABLE_RT_SYNC_MOTIONS":null,"g5":null,"g7_g8":null,"g33":null,"m17_m18":true,"m42":true,"m62_m65":null,"m67_m68":null,"m80_m81":null,"i2c_lcd":null,"smoothie_clustering":null,"graphic_display":null,"sd_card":null,"sd_card_pf":null,"bltouch":null}

```

I made some changes, so it may be different from the .h

bwenquan, 2023-08-23:
[image]

bwenquan, 2023-08-23:
Thanks.

My TMC drivers are on the write/read (each as one address)

I know laserPPI used step3, so I tried to move X (STEP0) but it not worked.

Paciente8159, 2023-08-24:
Just one question.
If you are using read/write mode and enabled parser modules options what does the TMC MCodes (M350, M906, M913, etc...) commands return?

bwenquan, 2023-08-24:
```
Grbl 1.7 [uCNC v1.7.4 '$' for help]
>:ok
>:ok
» $
» $
« $0=10.000
$1=0
$2=0
$3=0
$4=0
$5=0
$6=0
$7=0
$10=1
$11=0.200
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=10.000
$25=50.000
$26=250
$27=2.000
$30=1000
$31=0
$32=0
$33=254
$34=1500
$35=0.250
$36=0.750
$80=1
$81=0.000
$100=200.000
$101=200.000
$110=500.000
$111=500.000
$120=10.000
$121=10.000
$130=200.000
$131=200.000
ok
ok
» $
« [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $C $X $H ~ ! ? ctrl-x]
ok
ok
» $I
« [VER: uCNC 1.7.4 - Kirizu-laser]
[OPT:C2,16B,DSS3_500,PPI,DC,DP,XC,30,128]
ok
ok
» M350
« [MICROSTEPS:X16.000,Y16.000,Z16.000,A16.000,B-1.000,C-1.000,I0.000,J0.000]
» M906
« ok
ok
» M906
« error:2
ok
» M906
« [STEPPER CURRENT:X795.497,Y795.497,Z795.497,A795.497,B-1.000,C-1.000,I-1.000,J-1.000]
» M913
« ok
ok
» M914
« error:2
ok
» M914
« [STEPPER STALL SENSITIVITY:X0,Y0,Z0,A0,B0,C0,I-255,J-255]
» M920
« ok
ok
error:51
ok
» M920
« error:51
ok
» M920
« error:51
ok
```

Paciente8159, 2023-08-24:
So apparently TMC communication is working.
Then I will redirect my focus to the web build configuration files.
Thanks

bwenquan, 2023-08-24:
Hi. I just checked the pin status using the logic analyzer. 
I found that the STEP pin was always at a high level without change (even though I used the G0 command to move the motor).

Paciente8159, 2023-08-24:
You are correct. It's not. An even worse it's making a very odd behaviour. When I connect the bluepill using the config file you sent it stays in alarm mode for a while and it takes quite sometime to get unstuck.

I'm not sure if this is caused by the configuration files generated by the web builder or not.

I will try to debug the code with the debug DAP to see if can understand what is going on.
I'll also try to build a custom boardmap manually that fits your config and test it. Just one question regarding that. Are you using a commercial board or a custom board?

Paciente8159, 2023-08-25:
Ok. I only have a bluepill. Some of IO pins you have defined I don't have at my disposal.
I'll use the same pins you are using and that I have available to do the tests.

Since you are using a MCU with more ram/flash and IO some changes are required.
The platformio board might need to be changed and also an option (FLASH_SIZE) like explained [here](https://github.com/Paciente8159/uCNC/wiki/STM32Fx:-Compiling).

But first let me get this running on my bluepill and I'll give you feedback before going there.

Paciente8159, 2023-08-25:
It seems that you chose a particular pin that is not used as a GPIO on reset.
I believe this is the cause. Since I do not use the STM32 HAL and do all bare-metal these kind of issues pop.
Let me study this for a bit.

[image]
[image]

Paciente8159, 2023-08-26:
I'm pushing #486 to both master and v1.8.x.
This will unlock pins PB3, PB4 and PA15 as GPIO, so the firmware is able to use them.
I tested using my scope and STEP0 was working ok on PB4 with this modification.

bwenquan, 2023-08-26:
Thank you very much!
Now my motor is working well, but I find that using the `G0 X100 F100` and `G0 X100 F200` the speed is the same. how should I change the speed and acceleration, is there something like a `rotation_distance` (distance that the axis travels with one full rotation of the stepper motor) configuration here?

Paciente8159, 2023-08-26:
That is odd. I should not be the same. I will check that.

bwenquan, 2023-08-26:
For another question, I defined:
```
#define DEFAULT_HOMING_ENABLED 1
#define ALLOW_SOFTWARE_HOMING
```
but when I use `$H`, I received `error:5`. (Homing cycle is not enabled via settings.). How do I need to enable the setting?

Paciente8159, 2023-08-26:
It should be enabled via `DEFAULT_HOMING_ENABLED ` but you can check it's value via `$$` and modify it without the need to reflash. See setting $22. For software homing you should also enable soft limits (setting $20)

Paciente8159, 2023-08-26:
> Thank you very much! Now my motor is working well, but I find that using the `G0 X100 F100` and `G0 X100 F200` the speed is the same. how should I change the speed and acceleration, is there something like a `rotation_distance` (distance that the axis travels with one full rotation of the stepper motor) configuration here?
> 
> Edit: I tried:
> 
> ```
> » $100=80
> $110=300
> $120=2000
> G91
> G0 X40 F200
> « ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> » $100=80
> $110=300
> $120=2000
> G91
> G0 X40 F100
> « ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> ok
> ```
> 
> The motor speed is the same.

About this. I tested and on my side is working quite well.
G1X40F100 took about 24 seconds (60*40/100=24 so it's spot on)
G1X40F200 took about 12 seconds (again 60*40/200=12 so it's spot on too)

bwenquan, 2023-08-26:
Sorry, I tested it agagin, it works well.
I have tested most of the functions and found no problems.
I will test it on my machine a few days later.

Here are some functional suggestions:
1. Air Evacuation motor relay output, enables Menu item and G-code M10/M11 control.
2. Ammeter support: This is a low side configuration (Ground) current IC that can be used for both CO2 and Diode based lasers. It’s ideal with High Voltage CO2 laser tubes, it connects between chassis ground and the tube ground. 
3. CO2 Coolant flow meter support: Provides tube protection by detecting flow failures and controlling laser enable signals.
4. Laser cooling control, provides chiller control with temperature monitoring and safety.

You can get more information at [Marlin Laser Document](https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html#other-features).

Thank you very much again!

Paciente8159, 2023-08-26:
> Sorry, I tested it again, it works well. I have tested most of the functions and found no problems. I will test it on my machine a few days later.
> 
> Here are some functional suggestions:
> 
> 1. Air Evacuation motor relay output, enables Menu item and G-code M10/M11 control.
> 2. Ammeter support: This is a low side configuration (Ground) current IC that can be used for both CO2 and Diode based lasers. It’s ideal with High Voltage CO2 laser tubes, it connects between chassis ground and the tube ground.
> 3. CO2 Coolant flow meter support: Provides tube protection by detecting flow failures and controlling laser enable signals.
> 4. Laser cooling control, provides chiller control with temperature monitoring and safety.
> 
> I implemented these functions on another board, so it's not necessary to build them into uCNC right now for me. You can get more information at [Marlin Laser Document](https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html#other-features).
> 
> Thank you very much again!

Thanks for the sugestions.

Just a note about that:
  - M10 is already part of the scope and is used to control servo motors (similar to Marlin).
  
I will look at the provided Marlin documentation link and try to add these as modules.
Any problem you ran into, please report.