Paciente8159/uCNC #481: compilation failed for graphic_display module [BUG]

mvackel, 2023-08-23:
Compilation failed when graphic_display motule is added to the project.

Error:
C:\uCNC-1.7.3\uCNC\src\modules\graphic_display\graphic_display.c:20:10: fatal error: clib/u8g2.h: No such file or directory
 #include <clib/u8g2.h>
          ^~~~~~~~~~~~~
compilation terminated.

Compilation error: clib/u8g2.h: No such file or directory

PC running Windows 11.  Arduino Mega 2560 with RAMPS 1.4 and offline display (RepRap Discount Full Graphic Smart Controller).

Used config files generated by the web config interface.

Added (on web config): Modules: Support for RepRap Full Graphic Display
( The instruction in the uCNC-modules-1.7.0.zip  file "graphic_display/README.me" seems to be wrong, as there is no need to change the file "src/modules.c" (the config changes "cnd_hal_overrides.h" defining "LOAD_MODULES_OVERRRIDE()".)

I downloaded the u8g2 lib from the arduino rep and installed it on the default arduino lib, but the compiler (arduino IDE 2.1) can not find the needed include file. 
Also tried the VSCode+Platformio but could not find any file "platformio.ini" in the project and could't write one that could work.

Paciente8159, 2023-08-23:
Hi there.
On vscode you need to uncomment this [Line](https://github.com/Paciente8159/uCNC/blob/8a8054b9912ce9bd6438fd811e864cb31639d1d4/platformio.ini#L36)
On arduino ide I think you also add the include in the ino file before the cnc.h include that is there.

mvackel, 2023-08-24:
Hello, 
thanks for answering.

I've made the changes you suggested and the error has changed:

Compiling .pio\build\megaatmega2560\src\src\modules\graphic_display\clib\u8x8_d_ssd1306_48x64.c.ouCNC\src\modules\graphic_display\clib\u8x8_d_sdl_128x64.c:7:10: fatal error: SDL2/SDL.h: No such file or directory #include "SDL2/SDL.h"          ^~~~~~~~~~~~compilation terminated.

I'm using platformio.

On arduino IDE 2.1, after installing the correct library (olikraus/U8g2), I got the same error.

Paciente8159, 2023-08-24:
Ahh that must have been forgotten there. It's the librarybto run u8lib on windows virtual environment.
I'll check what's going on.
What is your board?

Paciente8159, 2023-08-24:
I compiled the code in VSCode for RAMPS board and I did not had that error. But I had other issue.
U8g2 requires both Arduino Wire and SPI libraries (although they are not used on most boards in µCNC).

For AVR you will also need to comment or remove this [line](https://github.com/Paciente8159/uCNC/blob/8a8054b9912ce9bd6438fd811e864cb31639d1d4/uCNC/src/hal/boards/avr/avr.ini#L14C1-L14C1). After this compilation on VSCode worked for me.

mvackel, 2023-08-25:
Hello, 
I've downloaded the full zip file from github (named uCNC-master.zip), created a folder and unzip the file. copied the modified "boardmap_overrides.h" and "cnc_hal_overrides.h" to the uCNC folder.

Removed the comment on line: 36 from  platformio.ini:
; lib_deps = olikraus/U8g2

Commented line 14 of: uCNC/src/hal/avr/avr.ini:
lib_ignore = EEPROM, SPI, Wire

I had to change the platformio.ini to the following because the build was installing a huge amount of libs that I do not intend to use:
----- new platformio.ini for ramps 1.4 & ReprapDiscount Full Graphic Smart Controller (to be validated) ----
[platformio]
include_dir=uCNC
src_dir=uCNC
default_envs = ramps14
;default_envs = uno, ramps14, rambo, arduinoM0, bluepill_f103c8, blackpill_f401cc, d1, re_arm, d1_r32, rpi_pico_w
extra_configs = 
  uCNC/src/hal/boards/avr/avr.ini
;  uCNC/src/hal/boards/samd21/samd21.ini
;  uCNC/src/hal/boards/stm32/stm32.ini
;  uCNC/src/hal/boards/esp8266/esp8266.ini
;  uCNC/src/hal/boards/lpc176x/lpc176x.ini
;  uCNC/src/hal/boards/esp32/esp32.ini
;  uCNC/src/hal/boards/rp2040/rp2040.ini
;   uCNC/src/hal/mcus/virtual/virtual.ini

[common]
build_flags = -std=gnu99 -Wall -fdata-sections -ffunction-sections -fno-exceptions -Wl,--gc-sections

[env]
framework = arduino
build_flags = ${common.build_flags}
monitor_speed = 115200
monitor_eol = LF
monitor_echo = yes
monitor_filters = colorize, time
; uncomment to add graphic display library
lib_deps = olikraus/U8g2

--------------------------------------------------------

--- Build using platformio task:  ramps14/build

Now I receive the following error/warning (the warning seams to be irrelevant):

uCNC\src\interface\settings.c: In function 'settings_register_external_setting':
uCNC\src\interface\settings.c:666:3: warning: #warning "External/extension settings storing is disabled" [-Wcpp]
  #warning "External/extension settings storing is disabled"
   ^~~~~~~
.....
Linking .pio\build\ramps14\firmware.elf
C:\Users\Mack\AppData\Local\Temp\cc9FrQJ2.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x398): undefined reference to `graphic_display_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ramps14\firmware.elf] Error 1

Paciente8159, 2023-08-25:
> I had to change the platformio.ini to the following because the build was installing a huge amount of libs that I do not intend to use:
> ----- new platformio.ini for ramps 1.4 & ReprapDiscount Full [...]

You don't need to.
Just select the environment.
[image]

> Now I receive the following error/warning (the warning seams to be irrelevant):
>
> uCNC\src\interface\settings.c: In function 'settings_register_external_setting': uCNC\src\interface\settings.c:666:3: [...]

You might have forgot to add the graphic_display module directory and files to the `uCNC/src/modules` directory

mvackel, 2023-08-26:
Oops... my mistake... sorry. 
After copying the graphic_display module to the uCNC/src/modules, the error returned:

uCNC\src\modules\graphic_display\clib\u8x8_d_sdl_128x64.c:8:10: fatal error: SDL2/SDL.h: No such file or directory
 #include "SDL2/SDL.h"
          ^~~~~~~~~~~~
compilation terminated.

I've tried to declare  "#declare NO_SDL" in the files that are issuing this compilation error ("u8x8_d_sdl_128x64.c" and "u8x8_sdl_key.c"). The project compiles and uploads OK, grbl is working (on the serial monitor) but the display is blank, not working.

Paciente8159, 2023-08-26:
> Oops... my mistake... sorry. After copying the graphic_display module to the uCNC/src/modules, the error returned:
>
> uCNC\src\modules\graphic_display\clib\u8x8_d_sdl_128x64.c:8:10: fatal error: [...]

Again that error??

That should be only used if compiling the code to run the windows emulator. Not for RAMPS. And I compiled the code with the graphic display module and it compiled ok.

Are you compiling using Arduino IDE or VSCode?
Are you using web config build? If yes send me the config json.
If possible please do a fork and add all your changes. This will make it easier for me to check the issue

Paciente8159, 2023-08-29:
Here is a [fork](https://github.com/Paciente8159/uCNC/tree/graphic-test) with the graphic_display module for RAMPS 1.4.

You can also try the hex from this build here. This is the default settings. I just added the graphic_display module.
[firmware.zip](https://github.com/Paciente8159/uCNC/files/12467760/firmware.zip)

mvackel, 2023-09-01:
Hello,
I finally got a time to test the fork and it's working perfectly with the graphic display. I'll test it and post a comment. 
Thank you very much for your great work, your interest and good will on helping.

Paciente8159, 2023-09-13:
Since this question was solved I'm closing the issue

closed 2023-09-13 by Paciente8159