Paciente8159/uCNC #496: 5-Axis Compilation problem

mvackel, 2023-09-16:
When compiling for 5 axis the compiler issues the following error:

uCNC\src\core\interpolator.c:1625:18: error: 'STEP4_ITP_MASK' undeclared (first use in this function);

Tracking the issue, the following patch worked and fixed the problem:
- File: cnc_hal_config_helper.h:  Insert the following lines just after line 781:
```
#if (STEP4_MASK == STEP_DUAL0) 
    #define STEP4_ITP_MASK (STEP4_MASK | STEP_DUAL0_MASK) 
#elif (STEP4_MASK == STEP_DUAL1) 
    #define STEP4_ITP_MASK (STEP4_MASK | STEP_DUAL1_MASK)  
#elif (STEP4_MASK == STEP_DUAL2)
    #define STEP4_ITP_MASK (STEP4_MASK | STEP_DUAL2_MASK)
#elif (STEP4_MASK == STEP_DUAL3)
    #define STEP4_ITP_MASK (STEP4_MASK | STEP_DUAL3_MASK)
#else
    #define STEP4_ITP_MASK STEP4_MASK
#endif
```

Paciente8159, 2023-09-17:
Thanks. I'll check it.

Paciente8159, 2023-09-17:
You are correct. Patch applied.

closed 2023-09-17 by Paciente8159