Paciente8159/uCNC #502: Board Compabilty

Avatarsia, 2023-09-17:
Hi,

is this fork compatabile with the azteeg x5 gt it have a lpc 1769 mcu

tia

Paciente8159, 2023-09-18:
Hi there,

There are several configurations for LPC1768 board.
In theory, LPC1769 should also be supported. The only thing that would be needed would be to create a board pinout configuration (boardmap file) and add an entry on platformio lpc ini file to compile the code to the LPC1769.

I can't find the board schematics (since Panucatt documentation link is dead).
With it I would be able to provide a bin file for you to load in the SD card and test if at least the communication works and if it does then a full boardfile can be added.

Avatarsia, 2023-09-18:
i will try to get my hands on the pin map

Avatarsia, 2023-09-18:
will this from marlin work ? 
[pins_AZTEEG_X5_GT.txt](https://github.com/Paciente8159/uCNC/files/12646799/pins_AZTEEG_X5_GT.txt)

Paciente8159, 2023-09-18:
It misses some important information like the communication ports pins (UART, USB, etc..). Without that info I cannot configure the bare minimal for communication.

Avatarsia, 2023-09-18:
Its possible to port from a smoothie config?
https://github.com/eclsnowman/HercuLien/blob/master/Azteeg%20X5%20GT%20Smoothieware/config.txt

Paciente8159, 2023-09-18:
Found it [here](https://reprap.org/wiki/Azteeg_X5_mini) ..
I'll try to get something for you to test in a few days.

Avatarsia, 2023-09-18:
Okay, seen that also before, just not sure if the mini have the same pinmap as the x5 gt

Paciente8159, 2023-09-18:
Ups my mistake then. I was interested in the schematic files. I'll search a bit more

Avatarsia, 2023-09-19:
https://16827713144580466377.googlegroups.com/attach/715600944ca3b/Capture_LI%20(2).jpg?part=0.1&view=1&vt=ANaJVrFC_OzNZgqw-aDNQ1mo_nI7Bc-87KNAtbP47EDfNxx1SSwV7eW1WcW50mNMMZYgcW0oEIfZTLz3exLyN12rBz0dU6TUm-EvNggIhynGUgMYUzQo7ZI

yeah usb and sd card

Avatarsia, 2023-09-29:
seems the board died, so closed for now and have a look for a better documented board

Paciente8159, 2023-09-29:
> seems the board died, so closed for now and have a look for a better documented board

Sorry to ear about that.
I will close the issue. I might try to get my hands on a board with that mcu in the future.

closed 2023-09-29 by Paciente8159