Paciente8159/uCNC #544: [FR] 74HC595 support for RP2040

Paciente8159, 2023-10-20:
### Discussed in https://github.com/Paciente8159/uCNC/discussions/543

<div type='discussions-op-text'>

<sup>Originally posted by **Hyakutake12** October 20, 2023</sup>
Hi!
I see shift register outputs are not supported for rp2040 mcu. Any plans to implement this feature?</div>

Paciente8159, 2023-10-30:
RP2040 already support 74HC595 via bitbang or SPI.
I2S is not yet supported.

Hyakutake12, 2023-10-30:
Hi. I could configure 74hc595 after the web builder update. Even after chaining 4 shift registers, the firmware is able to run 6 steppers and other outputs at high enough speed. I feel bit banging fulfills my requirement and no need for i2s. You can close the FR.. thanks

Paciente8159, 2023-10-30:
Ok thanks for the feedback.
You might get even better throughput using hardware SPI @ 20MHz (theoretical top speed for the 74HC595).
Never the less I will leave this open for future development when I find the time.

andrewmarles, 2023-11-01:
Always keep an eye on this project, and I just wanted to mention that GRBLHAL has support for shift registers on RP2040 using the PIO.  This creates the entire waveform for step/dir on 4 axes (direction, step pulse rise, step pulse fall).  Because the whole waveform is handled in the PIO, it really cuts the IRQ overhead in half and is just as fast as hardware SPI.

https://github.com/grblHAL/RP2040/blob/6ecdf608fbfe1d9d8091a73c2e46a12cd9d638b9/driverPIO.pio#L111

Paciente8159, 2023-11-01:
Thanks for the info. Very interesting stuff indeed. I'll will dive deeper into this for sure.

Paciente8159, 2023-11-03:
> Always keep an eye on this project, and I just wanted to mention that GRBLHAL has support for shift registers on RP2040 using the PIO. This creates the entire waveform for step/dir on 4 axes (direction, step pulse rise, step pulse fall). Because the whole waveform is handled in the PIO, it cuts the IRQ overhead in half and is just as fast as hardware SPI.
> 
> https://github.com/grblHAL/RP2040/blob/6ecdf608fbfe1d9d8091a73c2e46a12cd9d638b9/driverPIO.pio#L111
> 
> I just thought I'd post here because it might give someone ideas for uCNC and it is the only place I have really seen where a project has leveraged the PIO for motion control.

I've just implemented a custom 74HC595 control IO via PIO in this [branch](https://github.com/Paciente8159/uCNC/tree/74hc595-pio). Thanks for the tip.

Paciente8159, 2024-07-14:
Although I would like o implement I2S support for RP2040, it's currently not a priority and the PIO integration should allow a usable solution (#568). I'm closing this for now.

closed 2024-07-14 by Paciente8159