Paciente8159/uCNC #550: [BUG]

jimfong1, 2023-10-21:
**Describe the bug**
Testing out uCNC and output G0 step pulse rate does not match the grbl settings 

**What is your hardware**
Describe your hardware:
1. ESP8266 Wemos D1
2. Paste your "Grbl/µCNC settings
>>> $i
[VER: uCNC 1.8.0 - WEMOS D1]
[OPT:C3,DSS3_500,DC,DL,DP,20,128]
ok
4. >>> $$
$0 = 15.625    (Step pulse time, microseconds)
$1 = 0    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$7 = 0   
$10 = 1    (Status report options, mask)
$11 = 0.200    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 10.000    (Homing locate feed rate, mm/min)
$25 = 50.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$80 = 1   
$81 = 0.000   
$100 = 500.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$110 = 2000.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok

5. If this is a custom compilation tell us what configurations you changed
no using the supplied firmware at 
https://github.com/Paciente8159/uCNC/releases/download/v1.8.0/ESP8266-Wemos-D1-R2.zip

**What was the GCode running**
A clear and concise description of what you expected to happen. If needed paste a sample of your Gcode file.
With the following settings
$100 = 500.000    (X-axis travel resolution, step/mm)
$110 = 2000.000    (X-axis maximum rate, mm/min)

With the above settings,  step output rate should be 16.6Khz with a G0 move
500x2000/60=16666 steps/sec
However I get 21300Hz when viewing output pulse on my Rigol Scope. 

When I set $110=500 the output pulse is very close 
500x500/60=4166 step/sec
My scope shows 4257Hz

when set to $110=1500 the output is also correct
500x1500/60=12500 step/sec
My scope shows 12700Hz

when set to $110=3000 the step pulse output is also too high
500x3000/60=25000 steps/sec
My scope shows 31500Hz

Seems like having the $110 to high causes mismatched output pulse rate for G0 moves. 
My external Fluke frequency counter also shows the same reading as the Rigol Scope does. 
Is this just a problem with the esp8266 processor? Does this happen with other firmware versions?

Thanks
Jim

**Additional context**
Add any other context about the problem here.

Paciente8159, 2023-10-21:
I will investigate.
That is odd because the stepping frequency is set the same way independent of the max speed $110.

This seems to point to issues that an inherit to MCU running FreeRTOS. It's even more odd when the frequency rises instead of dropping unless the FreeRTOS is trying to compensate for some delay in response.

I will test and give feedback.

Paciente8159, 2023-10-21:
These were my results

Same as you 
$100=500

$110=500 - 4.13Khz
$110=1500 - 12.8Khz
$110=2000 - 16.0Khz
$110=3000 - 21.3Khz

The results seem to be closer correct (since the single available timer also as to cope with generating software PWM) or else PWM would not be available.

There is also another implementation detail I just remembered. To be able to generate software PWM the ISR is running at a fixed 128KHz rate and firing the step gen at this rate.
At higher step rates this introduces imprecisions in the step rate that are to be expected.
So the resolution of the step rate reduces as you get closer to the limit.
The possible step rates are 64Khz/N

so
64KHz
32KHz
21.3KHz
16KHz

If you want I can try to provide you a version with a better resolutions but that might take an effect in processing power because of the high interrupt rate.

jimfong1, 2023-10-21:
Hmm wonder why my ESP8266 giving such higher values. I even compiled my own firmware and same results. I have to look at the esp8266 model used and maybe try a different one.   This latest batch I received from Aliexpress a couple weeks ago.

I test all different cnc firmware on a bunch of different processors.  uCNC is new for me so just playing around when I noticed the discrepancy.  I think its the only cnc firmware available that runs on the esp8266 and what initially peaked my interest in uCNC.   I have every processor that uCNC supports here so will choose another one to play.  Thanks

Paciente8159, 2023-10-21:
Maybe some rounding factor caused that or the distance in the GCode command you chose that lead to accelerate a bit further.

You see you are not that far. As I said the next jump from 21.3Khz is 32KHz which is what you got. It's not a linear jump.

I don't know of any other CNC firmware for ESP8266 and after implementing it (and I almos t gave up on the idea) I understood why. The total lack of both IO and capabilities (timers, port drivers, etc) for these type of tasks.

Even so it should be able to run a machine with modest speeds and/or low resolution (steps/mm).
Let me know if you want to try to run a modification with a bit higher ISR resolution.

jimfong1, 2023-10-21:
No need for a special firmware version. Just testing uCNC/esp8266 and not for any machine project that I had in mind.  I'll be testing S curve acceleration with a STM32 next. I have alot of linear motion hardware here, both stepper and servo systems that I can use to test.  Plus THK linear stages.  You can close this  issue and thanks for your time.

Paciente8159, 2024-01-10:
I may have found a solution for this problem using step aliasing. I need to do some tests and if all goes ok I will release an update for ESP8266.

closed 2024-02-03 by Paciente8159