Paciente8159/uCNC #597: [BUG] TMC2209 screeching driver problems

c---, 2024-01-22:
**Describe the bug**
I'm using a BTT SKR Pico board with TMC2209 drivers. When the board is powered on or when reset the steppers make a high pitched screeching sound. It goes away after a second or so. Sounds bad. Might be because of lack of stealthchop, I'm not sure. Movement sounds way louder and more "crunchy" than Klipper even with stealthchop OFF in Klipper.

For some reason microstepping 64 is not available as a config option. Setting it causes the screeching to get worse. I think 64 is a valid microstep setting without using interpolation. I will have to check the datasheet again.

If I get time I will compare the driver code versus something that I know works like Klipper. grblHAL doesn't seem to work at all with the 2209 and Marlin support is only half implemented.

**What is your hardware**
1. BTT SKR Pico 

`boardmap_overrides.h`
```
#include "boardmap_reset.h"
#define MCU MCU_RP2040
#define KINEMATIC KINEMATIC_CARTESIAN
#define AXIS_COUNT 3
#define TOOL_COUNT 1
#define BAUDRATE 2000000
#define S_CURVE_ACCELERATION_LEVEL -1
#define BOARD BOARD_RPI_PICO
#define BOARD_NAME "RPi Pico"
#define UART_PORT 0
#define UART2_PORT 1
#define SPI_PORT 0
#define I2C_PORT 1
#define ITP_TIMER 1
#define RTC_TIMER 2
#define STEP0_BIT 11
#define STEP1_BIT 6
#define STEP2_BIT 19
#define DIR0_BIT 10
#define DIR1_BIT 5
#define DIR2_BIT 28
#define STEP0_EN_BIT 12
#define STEP1_EN_BIT 7
#define STEP2_EN_BIT 2
#define PWM0_BIT 21
#define DOUT8_BIT 8
#define LIMIT_X_BIT 4
#define LIMIT_X_PULLUP
#define LIMIT_Y_BIT 3
#define LIMIT_Y_PULLUP
#define LIMIT_Z_BIT 25
#define LIMIT_Z_PULLUP
#define PROBE_BIT 22
#define ESTOP_BIT 16
#define DIN9_BIT 9
#define TX_BIT 0
#define RX_BIT 1
```
`cnc_hal_overrides.h`
```
#include "cnc_hal_reset.h"
#define BRESENHAM_16BIT
#define ENABLE_EXTRA_SYSTEM_CMDS
#define ESTOP_PULLUP_ENABLE
#define SAFETY_DOOR_PULLUP_ENABLE
#define FHOLD_PULLUP_ENABLE
#define CS_RES_PULLUP_ENABLE
#define LIMIT_X_PULLUP_ENABLE
#define LIMIT_Y_PULLUP_ENABLE
#define LIMIT_Z_PULLUP_ENABLE
#define LIMIT_X2_PULLUP_ENABLE
#define LIMIT_Y2_PULLUP_ENABLE
#define LIMIT_Z2_PULLUP_ENABLE
#define PROBE_PULLUP_ENABLE
#define TOOL1 laser_pwm
#define LASER_PWM PWM0
#define LASER_FREQ 8000
#define SOFT_SPI_CLK DOUT30
#define SOFT_SPI_SDO DOUT29
#define SOFT_SPI_SDI DIN29
#define STEPPER0_HAS_TMC
#define STEPPER0_DRIVER_TYPE 2209
#define STEPPER0_TMC_INTERFACE TMC_UART
#define STEPPER0_UART_TX DOUT8
#define STEPPER0_UART_RX DIN9
#define STEPPER0_UART_ADDRESS 0
#define STEPPER0_CURRENT_MA 800
#define STEPPER0_MICROSTEP 32
#define STEPPER0_RSENSE 0.11
#define STEPPER0_HOLD_MULT 0.7
#define STEPPER0_STEALTHCHOP_THERSHOLD 999363
#define STEPPER0_ENABLE_INTERPLATION false
#define STEPPER1_HAS_TMC
#define STEPPER1_DRIVER_TYPE 2209
#define STEPPER1_TMC_INTERFACE TMC_UART
#define STEPPER1_UART_TX DOUT8
#define STEPPER1_UART_RX DIN9
#define STEPPER1_UART_ADDRESS 1
#define STEPPER1_CURRENT_MA 800
#define STEPPER1_MICROSTEP 32
#define STEPPER1_RSENSE 0.11
#define STEPPER1_HOLD_MULT 0.7
#define STEPPER1_STEALTHCHOP_THERSHOLD 0
#define STEPPER1_ENABLE_INTERPLATION false
#define STEPPER2_HAS_TMC
#define STEPPER2_DRIVER_TYPE 2209
#define STEPPER2_TMC_INTERFACE TMC_UART
#define STEPPER2_UART_TX DOUT8
#define STEPPER2_UART_RX DIN9
#define STEPPER2_UART_ADDRESS 2
#define STEPPER2_CURRENT_MA 800
#define STEPPER2_MICROSTEP 32
#define STEPPER2_RSENSE 0.11
#define STEPPER2_HOLD_MULT 0.7
#define STEPPER2_STEALTHCHOP_THERSHOLD 0
#define STEPPER2_ENABLE_INTERPLATION false
#define IC74HC595_COUNT 0
#define ENCODERS 0
#define ENABLE_PARSER_MODULES
```

Paciente8159, 2024-01-23:
Thanks for the report.
One question. Have you tried sending any of the TMC M codes? (like M350, M906, etc...)
What does the driver responds?

Paciente8159, 2024-01-23:
One other question. This setting:

`#define STEPPER0_STEALTHCHOP_THERSHOLD 999363`

Is this on purpose?
Will probably force StealthChop to be active all the time. To disable it set this to 0. (register TPWMTHRS - page 28 of the datasheet)

Paciente8159, 2024-01-23:
I've taken a look at [Marlin as suggested](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/module/stepper/trinamic.cpp).
There are some parameters I am not configuring and (maybe wrongly) assumed to be default and others I had no knowledge it can influence the driver functionality.

I will try to dig a bit more on the matter.

c---, 2024-01-23:
I have not tried the M-codes yet. That stealthchop setting was on purpose because I was going to try to determine if it made the sound better or worse. I haven't tested it yet.

I may be offline for a few days but I hope to also experiment and see if I can find the issue.

Paciente8159, 2024-01-24:
The MCodes can tell if the communication is good or not.
For example when you send a `M350` the controllers reads the microstepping values for all stepper drivers.
If the stepper is not a TMC driver or the communication fails it will reply with a value of -1. I the value read from the driver is not a valid value it will return a 0.

On an 3 axis system if you send a `M350` the response should be something like:

`[MICROSTEPS:X32,Y32,Z32]`

I will probably take the opportunity to implement #258

Paciente8159, 2024-01-24:
TMC communication does not seem to be working quite well on v1.8. I suspect the fault is on softuart module (under investigation).
You can try the v1.7.x-bugfix branch to see if the stepper driver works better with that version.

I tested with a TMC2208 on a RAMPS board.

Paciente8159, 2024-01-25:
As I said the problem was on the soft uart module. PR #600 addresses this issue and I tested on the RAMPS board with a TMC2208 and now it reads the TMC correctly.

This is a temporary fix in the master branch. Please test it and let me know if you find any difference or the stepper still "screeches".

I am moving the TMC module to the [µCNC modules git](https://github.com/Paciente8159/uCNC-modules) and modifying the setup based on what is done on Marlin.

Paciente8159, 2024-02-03:
Merged #599. TMC drivers have moved to modules. If you find any problem please report.

closed 2024-02-03 by Paciente8159