Paciente8159/uCNC #605: [BUG] If I start the spindle, ALARM:13 is displayed

andy0983, 2024-01-31:
**Describe the bug**
A clear and concise description of what the bug is.
If I start the spindle, ALARM is displayed, but not always
Examples 1:
M3 S12000
ok
[MSG:Reset to continue]
[MSG:Reset to continue]
ok
[MSG:Reset to continue]
Examples 2:
M3 S12000
ok
ALARM:13
[MSG:'$H'|'$X' to unlock]

Describe your hardware:
1. What's your board
uCNC 1.8.6 -master
Arduino Mega  -  Ramps 1.4

2. Paste your "Grbl/µCNC settings"

#ifndef BOADMAP_OVERRIDES_H
#define BOADMAP_OVERRIDES_H
#ifdef __cplusplus
extern "C"
{
#endif

#include "boardmap_reset.h"
#define MCU MCU_AVR
#define KINEMATIC KINEMATIC_CARTESIAN
#define AXIS_COUNT 3
#define TOOL_COUNT 1
#define BAUDRATE 115200
#define S_CURVE_ACCELERATION_LEVEL 0
#define BOARD BOARD_RAMPS14
#define BOARD_NAME "RAMPS 1.4"
#define UART_PORT 0
#define UART2_PORT 2
#define ITP_TIMER 1
#define RTC_TIMER 0
#define ONESHOT_TIMER 4
#define STEP0_BIT 0
#define STEP0_PORT F
#define STEP1_BIT 6
#define STEP1_PORT F
#define STEP2_BIT 3
#define STEP2_PORT L
/*
#define STEP3_BIT 4
#define STEP3_PORT A
#define STEP4_BIT 1
#define STEP4_PORT C
*/
#define STEP3_BIT 1
#define STEP3_PORT C
#define STEP4_BIT 4
#define STEP4_PORT A
#define DIR0_BIT 1
#define DIR0_PORT F
#define DIR1_BIT 7
#define DIR1_PORT F
#define DIR2_BIT 1
#define DIR2_PORT L
/*
#define DIR3_BIT 6
#define DIR3_PORT A
#define DIR4_BIT 3
[... 242 lines cut ...]
#define ENABLE_IO_MODULES
#define ENABLE_PARSER_MODULES 
#define ENABLE_MOTION_CONTROL_MODULES
#define ENABLE_EXTRA_SYSTEM_CMDS

//cnc_hal_config.h
#define ESTOP_PULLUP_ENABLE
#define SAFETY_DOOR_PULLUP_ENABLE
#define FHOLD_PULLUP_ENABLE
#define CS_RES_PULLUP_ENABLE
#define LIMIT_X2_DISABLE
#define LIMIT_Z2_DISABLE
#define LIMIT_A_DISABLE
#define LIMIT_B_DISABLE
#define LIMIT_C_DISABLE
#define DISABLE_A_HOMING
#define DISABLE_B_HOMING
#define DISABLE_C_HOMING
#define SYSTEM_MENU_MAX_STR_LEN 32	

#ifdef __cplusplus
}
#endif

#endif

Paciente8159, 2024-01-31:
If this only happens when you run the spindle...most probably it's not a bug but a noise issue.
If you can run GCODE ok and nothing goes wrong and when you turn the spindle on it trips an alarm now and then you have  a noise issue.

You can confirm this by running the same GCODE with the spindle unpowered (but still using M3, M4 and M5 commands) and nothing trips.

You can solve this with hardware. I recommend using some optocouplers (like the inexpensive PC817) and work with 2 voltage levels.  5V on the Arduino side and 12V or 24V (or whatever power supply you are using on the other side.

Here is an example:
https://planet-cnc.com/wp-content/uploads/2016/02/switch.png

andy0983, 2024-02-12:
You were right, there are interferences with the Spindle+VFD and the relay that turns on the spindle cooling
Thank you for your help