Paciente8159/uCNC #607: [FR] Support SCARA machines with  2 fixed motors

ademenev, 2024-02-01:
SCARA kinematics implies that the arm motor is located on the forearm, and rotating only the forearm motor does not change the angle between the arm and the forearm. 

There is another way to implement a SCARA machine, pictured below. 

[image]

In this type of machine, keeping constant angle between the  arm and the forearm requires rotating both motors. Forward and inverse kinematics is not much more complex, but is different. 

I intend to implement this time of kinematics, the only question is: what would be the best name for it?

Paciente8159, 2024-02-01:
Awesome.
If it's the first of it's kind you can give any name you want. If not it would be preferable to follow existing naming conventions (like in GCode implementations) (using naming conventions used in other firmware like Marlin or LinuxCNC).
Let me know if any question arises.

ademenev, 2024-02-01:
It is not new, the idea is pretty simple and I guess it has been implemented independently by multiple people. In Marlin, it is known as MP SCARA (mostly printed SCARA). 

Probably, it would be easier to make that not a separate kinematics, but just an additional config option

Paciente8159, 2024-02-02:
> Probably, it would be easier to make that not a separate kinematics, but just an additional config option

If that can be done over the existing SCARA kinematic then ok.
I'm not against having a completely separate kinematic if it improves readability or for customization/maintenance reasons.

Paciente8159, 2024-03-03:
Implemented with #645

closed 2024-03-03 by Paciente8159