Paciente8159/uCNC #611: Sketch to big to fit into Arduino UNO memory

mikh3x4, 2024-02-09:
I've downloaded the newest master branch and loaded with a simple config generated by the web tool (I attached the config file). My Arduino IDE says it uses too much of program memory "Sketch uses 33024 bytes (102%) of program storage space. Maximum is 32256 bytes."

I tried compiling it with the -Os and -Oz flags to reduce the program size but with no luck

 
[ucnc_build.json](https://github.com/Paciente8159/uCNC/files/14220664/ucnc_build.json)

Paciente8159, 2024-02-09:
This is a intrinsic problem with Arduino IDE. The compiled binary is bigger than the one generated via VSCode because you cannot customize the building flags (not without messing with the Arduino IDE platform.txt file).
The default configuration can be build...just not with Arduino IDE.

If you know how to tweak the build flags on Arduino IDE use the ones that are used inside the avr.ini file in the `uCNC/src/hal/board/avr/` dir to be able to build a better/optimized version of the binary.

marcinmaruszewski, 2024-07-26:
Can you please explain how to compile it in VSCode? I'm having the following problem. I'd appreciate any help.

[image]

With flags from mikh3x4 _Sketch uses 33418 bytes (103%)_.

Paciente8159, 2024-07-26:
Can you share more details about your config files/options?

marcinmaruszewski, 2024-07-27:
TBH I just cloned the repository and ino file into VSCode, selected the Arduino UNO platform from the list and that's it. Is there any manual witch I should have checked before?

Paciente8159, 2024-07-27:
That is odd. Using the default settings should have compiled anyway. You should get a hex file similar to the one that is available in the release.

hmmueller, 2024-08-03:
FWIW: I had no problem installing VSCode + PlatformIO (on Win10) and then compiling and uploading the master branch, without any warnings or errors just a few days ago. If it helps, I could post screenshots of my build; or other files if instructed.

H.M.

MikeWent, 2025-06-06:
I confirm that issue still persists. 

```
Linking .pio/build/AVR-CNC-Shield-V3/firmware.elf
Checking size .pio/build/AVR-CNC-Shield-V3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (32368 bytes) is greater than maximum allowed (32256 bytes)
RAM:   [====*** [checkprogsize] Explicit exit, status 1
===   ]  67.7% (used 1387 bytes from 2048 bytes)
Flash: [==========]  100.3% (used 32368 bytes from 32256 bytes)
======================================================== [FAILED] Took 3.45 seconds ========================================================

Environment        Status    Duration
-----------------  --------  ------------
AVR-CNC-Shield-V3  FAILED    00:00:03.450
```

using `AVR-arduino-uno-shield-v3.json` config zip from web builder. 

PlatformIO Core, version 6.1.18, Linux.

MikeWent, 2025-06-07:
it's `master` branch with replaced files from downloaded .zip from uCNC web config builder.

Paciente8159, 2025-06-07:
There is. You ca read about it here. https://github.com/Paciente8159/uCNC/wiki/Basic-user-guide#7----control-pins-invert-mask
The main problem is that with the emergency stop pin active the system will loco completly for safety.
You can short the estop pin to ground then set $7 and $5 and then remove the shunt. Also make sure to read the FAQ so you don't bump into the corrupted settings/first run issue

deltalejo, 2025-12-23:
Indeed it was a toolchain version problem, I tried with avr-gcc 5.4.0 and The build succeed.
Thanks!