Paciente8159/uCNC #613: Running ucnc on NUCLEO-F411RE (STM32F411RET6)??? [question]

andy0983, 2024-02-11:
Running ucnc on NUCLEO-F411RE (STM32F411RET6)???

I added board support to the project, such as STM32F4-Blackpill-F411CE.
That is, changes in files:
///////// boardmap_nucleo.h - gpio setting  ///////// 
///////// stm32.ini - adding /////////
[env:NUCLEO-F411RE]
extends = common_stm32
board = nucleo_f411re
board_build.mcu = stm32f411ret6
board_build.f_cpu = 100000000L
;upload_protocol = dfu
;upload_protocol = stlink
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_NUCLEO_F411RE
///////// mcudefs.h - additions /////////
#if (MCU == MCU_STM32F4X)
#include "stm32f4x/mcumap_stm32f4x.h"
#endif
///////// boardmap_overrides.h /////////
#ifndef BOADMAP_OVERRIDES_H
#define BOADMAP_OVERRIDES_H
#ifdef __cplusplus
extern "C"
{
#endif
#include "boardmap_reset.h"
#define MCU MCU_STM32F4X
#define KINEMATIC KINEMATIC_CARTESIAN
#define AXIS_COUNT 3
#define TOOL_COUNT 1
#define BAUDRATE 115200
#define S_CURVE_ACCELERATION_LEVEL 0
#define BOARD BOARD_NUCLEO_F411RE
#define BOARD_NAME F411RE
#define UART_PORT 1
#define SPI_PORT 3
#define I2C_PORT 1
#define ITP_TIMER 2
#define SERVO_TIMER 3
#define ONESHOT_TIMER 1
#define STEP0_BIT 2
#define STEP0_PORT B
#define STEP1_BIT 1
#define STEP1_PORT B
#define STEP2_BIT 15
#define STEP2_PORT B
#define STEP3_BIT 14
#define STEP3_PORT B
#define DIR0_BIT 13
#define DIR0_PORT B
#define DIR1_BIT 4
#define DIR1_PORT C
#define DIR2_BIT 5
#define DIR2_PORT C
#define DIR3_BIT 12
#define DIR3_PORT B
#define STEP0_EN_BIT 6
#define STEP0_EN_PORT B
#define STEP1_EN_BIT 7
#define STEP1_EN_PORT C
#define STEP2_EN_BIT 8
#define STEP2_EN_PORT A
#define STEP3_EN_BIT 10
#define STEP3_EN_PORT B
#define PWM0_BIT 9
#define PWM0_PORT C
#define PWM0_CHANNEL 4
#define PWM0_TIMER 3
#define PWM1_BIT 8
#define PWM1_PORT C
#define PWM1_CHANNEL 3
#define PWM1_TIMER 3
#define PWM2_BIT 6
#define PWM2_PORT C
#define PWM2_CHANNEL 1
#define PWM2_TIMER 3
#define PWM3_BIT 6
#define PWM3_PORT A
#define PWM3_CHANNEL 1
#define PWM3_TIMER 4
#define DOUT0_BIT 11
#define DOUT0_PORT C
#define DOUT1_BIT 2
#define DOUT1_PORT D
#define DOUT2_BIT 10
#define DOUT2_PORT C
#define DOUT3_BIT 12
#define DOUT3_PORT C
#define DOUT31_BIT 5
#define DOUT31_PORT A
#define LIMIT_X_BIT 0
#define LIMIT_X_PORT C
#define LIMIT_X_PULLUP
#define LIMIT_Y_BIT 1
#define LIMIT_Y_PORT C
#define LIMIT_Y_PULLUP
#define LIMIT_Z_BIT 2
#define LIMIT_Z_PORT C
#define LIMIT_Z_PULLUP
#define LIMIT_Y2_BIT 3
#define LIMIT_Y2_PORT C
#define LIMIT_Y2_PULLUP
#define PROBE_BIT 14
#define PROBE_PORT A
#define PROBE_PULLUP
#define ESTOP_BIT 0
#define ESTOP_PORT A
#define ESTOP_PULLUP
#define SAFETY_DOOR_BIT 1
#define SAFETY_DOOR_PORT A
#define SAFETY_DOOR_PULLUP
#define FHOLD_BIT 4
#define FHOLD_PORT A
#define FHOLD_PULLUP
#define ANALOG0_CHANNEL -1
#define ANALOG1_CHANNEL -1
#define ANALOG2_CHANNEL -1
#define ANALOG3_CHANNEL -1
#define ANALOG4_CHANNEL -1
#define ANALOG5_CHANNEL -1
#define ANALOG6_CHANNEL -1
#define ANALOG7_CHANNEL -1
#define ANALOG8_CHANNEL -1
#define ANALOG9_CHANNEL -1
#define ANALOG10_CHANNEL -1
#define ANALOG11_CHANNEL -1
#define ANALOG12_CHANNEL -1
#define ANALOG13_CHANNEL -1
#define ANALOG14_CHANNEL -1
#define ANALOG15_CHANNEL -1
#define DIN0_BIT 13
#define DIN0_PORT A
#define DIN1_BIT 9
#define DIN1_PORT A
#define DIN2_BIT 10
#define DIN2_PORT A
#define TX_BIT 15
#define TX_PORT A
#define RX_BIT 7
#define RX_PORT B
#define USB_DM_BIT 11
#define USB_DM_PORT A
#define USB_DP_BIT 12
#define USB_DP_PORT A
#define SPI_CLK_BIT 3
#define SPI_CLK_PORT B
#define SPI_SDI_BIT 5
#define SPI_SDI_PORT B
#define SPI_SDO_BIT 4
#define SPI_SDO_PORT B
#define SPI_CS_BIT 0
#define SPI_CS_PORT B
#define I2C_CLK_BIT 8
#define I2C_CLK_PORT B
#define I2C_DATA_BIT 9
#define I2C_DATA_PORT B

#ifdef __cplusplus
}
#endif
#endif

I compiled the project and uploaded it to NUCLEO-F411RE the LED is blinking (ACTIVITY_LED) on Windows 10 "Unknown USB device" appears and error code 43

What should I do to make UCNC work? Please help and advice

Paciente8159, 2024-02-11:
Thanks for the report.
I don't have a Nucleo board but I will investigate. I'll give you feedback

Paciente8159, 2024-02-12:
I can't seem to find my F411 blackpill.
I spot a potential problem in your config.
this line
```
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_NUCLEO_F411RE
```

replace it by
```
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL
```

The overrides and the stm32.ini changes will enforce your configurations. Just maintain the µCNC boardmap definition.
Please retry. I will try to find my BlackpillF411 it in the meantime and do further testing.
By the way did you test the UART port? Was it communicating or muted also?

andy0983, 2024-02-12:
I changed it to "build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL"

USB still has error code 43 (USB\VID_0000&PID_0002\5&165ECDDA&0&2)
USB communication connection:
Nucleo board - USB cable
PA12(USB_DP) - USB Data
PA11(USB_DM) - USB Data
GND - GND
U5V - 5V

UART - after connecting and pressing (reset), this pops up
Grbl 1.8 [uCNC v1.8.7 '$' for help]
client>$$
>:ok
>:ok
$0=10,000 (Step pulse time, microseconds)
.
............
and accepts Gcode commands. so it works

I am attaching the log from the software upload:
Checking size .pio\build\NUCLEO-F411RE\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM: [ ] 3.5% (used 4532 bytes from 131072 bytes)
Flash: [= ] 10.5% (used 55104 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\NUCLEO-F411RE\firmware.elf
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
         http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x1fff4252 msp: 0x20002e58
** Programming Started **
**Programming Finished**
**Verify Started**
**Verified OK**
**Resetting Target**
shutdown command invoked

"Another question is where are you placing this code?"
I do not understand the question.
Which code are you talking about??

Paciente8159, 2024-02-12:
I found my blackpill f411.
I will try to test your config tomorrow despite being a different version from the one used in your board.

> "Another question is where are you placing this code?"
> I do not understand the question.
> Which code are you talking about??

I mean..in which file did you places the changes? In the files indicated in the comments?

andy0983, 2024-02-13:
yes, I added new code in these files 
//////// stm32.ini - adding ///// ////

Paciente8159, 2024-02-13:
Ok I tried your configuration and got a few errors.

Has explained keep the blackpill definition on the the stm32.ini like stated above

```
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL
```

Also remove this line from the boardmap_override.h
```
#define BOARD BOARD_NUCLEO_F411RE
```
And correct this line
```
#define BOARD_NAME F411RE
```
to this
```
#define BOARD_NAME F411RE
```

Also for now comment the whole part of the SPI pins for now. I think I need to fix the mcumap core file
```
//#define SPI_CLK_BIT 3
//#define SPI_CLK_PORT B
//#define SPI_SDI_BIT 5
//#define SPI_SDI_PORT B
//#define SPI_SDO_BIT 4
//#define SPI_SDO_PORT B
//#define SPI_CS_BIT 0
//#define SPI_CS_PORT B
```

And retry to upload like that

Paciente8159, 2024-02-13:
I've patched the SPI problem with the incorrect alternate function settings of the pins here #622 .
By the way, and according to the STM32 datasheet you have SPI_SDO and SPI_SDI pins swapped. Also the SPI_CS pin is not valid. But you will only need the CS pin if you plan on using the SPI in slave mode.

andy0983, 2024-02-13:
I took a clean design.
I added it to the stm32.ini file
[env:NUCLEO-F411RE]
extends = common_stm32
board = nucleo_f411re
upload_protocol = stlink
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL

I copied the files with corrections written in the post above
boardmap_override.h
boardmap_reset.h
cnc_hal_overrides.h
cnc_hal_reset.h

Next, USB, error code 43.

I did tests with such settings
[env:STM32F4-Blackpill-F411CE]
extends = common_stm32
board = nucleo_f411re
upload_protocol = stlink
;board = blackpill_f411ce
;upload_protocol = dfu
build_flags = ${common_stm32.build_flags} -D BOARD=BOARD_BLACKPILL

Next, USB, error code 43.
Tomorrow I will buy a new USB cable and test it again.
I can send you such a nucleo f411re board

Paciente8159, 2024-02-13:
> Tomorrow I will buy a new USB cable and test it again.
Please let me know if it worked. I don't think beside the pin count and extra drivers there is not much difference between mine 411CE and your 411RE.

> I can send you such a nucleo f411re board
That would be great and I deeply appreciate the offer. But lets see how it works with the cable.
If it fails I can try to draft a test using Arduino USB library to see if it solves.
Thanks

andy0983, 2024-02-14:
Works :)
The error was the wrong connection of the USB 5v power supply, it was connected to U5V and should have been to E5V

I am sending the correct connection.

USB D+ (green wire) to PA12
USB D- (white wire) to PA11
USB 5V to E5V
USB gnd to GND
Jumper on E5V
Jumper on boot0 and VCC (pins shorted only when uploading the software using the dfu / stlink protocol)

[image]

andy0983, 2024-02-14:
Paciente 8159, please contact me by e-mail (andy9@wp.pl) to send the Nukleo F411re board

Paciente8159, 2024-02-14:
Again. Thank you for your offer. As I said the nucleo chip is very similar to the blackpill with f411 with an attached st-link.
I have several bluepills, blackpills, st-links and cmsis-daps...so currently I am not in the need of such board.

But again thank you for your kind offer.

Paciente8159, 2024-02-15:
Merging PR #628.
Let me know if you find any issue in the future

closed 2024-02-15 by Paciente8159