Paciente8159/uCNC #665: [BUG]

andy0983, 2024-03-19:
with #define F_STEP_MAX 100000 defined there is a compilation error but when we reduce it to F_STEP_MAX 90000, the compilation goes without error

#define MCU MCU_STM32F4X
#define KINEMATIC KINEMATIC_CARTESIAN
#define AXIS_COUNT 4
#define TOOL_COUNT 2
#define BAUDRATE 115200
#define BOARD BOARD_NUCLEO_F411RE_SHIELD_V3
#define BOARD_NAME "STM32 Nucleo F411RE"

In file included from uCNC\src\core\../cnc.h:130,
                 from uCNC\src\core\interpolator.c:20:
uCNC\src\core\../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\core\../cnc.h:130,
                 from uCNC\src\core\io_control.c:21:
uCNC\src\core\../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\cnc.h:130,
                 from uCNC\src\cnc.c:23:
uCNC\src\cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\core\../cnc.h:130,
                 from uCNC\src\core\motion_control.c:18:
uCNC\src\core\../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\core\../cnc.h:130,
                 from uCNC\src\core\parser.c:24:
uCNC\src\core\../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\core\../cnc.h:130,
                 from uCNC\src\core\planner.c:20:
uCNC\src\core\../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
In file included from uCNC\src\hal\kinematics\../../cnc.h:130,
                 from uCNC\src\hal\kinematics\kinematic.c:19:
uCNC\src\hal\kinematics\../../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
      |  ^~~~~
*** [.pio\build\STM32F4-Nucleo-F411RE-Shield-V3\src\src\core\interpolator.c.o] Error 1
In file included from uCNC\src\hal\kinematics\../../cnc.h:130,
                 from uCNC\src\hal\kinematics\kinematic_cartesian.c:20:
uCNC\src\hal\kinematics\../../cnc_hal_config_helper.h:1834:2: error: #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"
 1834 | #error "DSS_CUTOFF_FREQ should not be set above 1/8th of the max step rate"

Paciente8159, 2024-03-19:
Well the error indicates that you should also modify DSS_CUTOFF_FREQ (cnc_config.h)
Currently it must be set manually.

But I guess it can be auto adjusted in the sanity check phase. I'll try to review that in the future.

closed 2024-03-22 by andy0983

Paciente8159, 2024-03-26:
reviewed by #670