Paciente8159/uCNC #667: [FR] Default servo position

ademenev, 2024-03-24:
Is it possible to set the default position for a servo (ideally, separately for each configured servo)? 

I have a pen plotter where zero servo position corresponds to lowered pen, so the pen is dropped each time the machine is powered on. In my particular case, I can redesign and rebuild the pen drive unit so that it turns in opposite direction, but that is not always possible, and I feel like the start position should be configurable.

Paciente8159, 2024-03-24:
Hum... I see.

No currently there is not a way to configure that. That is one to add to the bucket list for sure.
Currently the best place for you to place that is to place it after [cnc_init() call in the ino file](https://github.com/Paciente8159/uCNC/blob/970d920bd0c73695f8ffbc07490a1121a39f987b/uCNC/uCNC.ino#L6).

But if you need to re-run that code at every main loop reset then adding a listener to the cnc_reset event is the best bet. There are some examples on how to do that in the modules git.

Paciente8159, 2024-03-26:
I was starting to implement something for this but I just realized that there maybe a better solution already at hand.

If this is to control a servo that is part of a tool you could add this to the tool startup code. Each tool as a startup and a shutdown function that runs on every tool loading/changing.

This might be the best place for you to place the code.

Never the less I'm adding a custom code entry point to set IO state on reset.
I'm thinking of implementing tools loading and unloading macros soon.. so stay tuned

Paciente8159, 2024-03-26:
#669 should allow you to customize the IO values at power up easily. just need to declare the mcu_io_reset anywhere function with your custom pin states.
```
void mcu_io_reset(){
  // set to SERVO0 to the opposite position
  mcu_set_pwm(SERVO0, 255);
}
```

closed 2024-03-26 by Paciente8159

ademenev, 2024-03-27:
Thank you!

Ironically, my local store is selling defective servos, and after 3 of them burnt, literally, with red glow and smoke, I redesigned my machine to use a stepper instead of servo :)