Paciente8159/uCNC #672: Ask if it has the function of PCB cutting proximity switch to test the flatness of the mesh bed

wrench1997, 2024-03-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the feature is:
custom harware

**Describe the solution you'd like**
Ask if it has the function of PCB cutting proximity switch to test the flatness of the mesh bed.

**Describe your target machine hardware**
A clear and concise description of your target machine hardware (mechanics, electronics, etc..).

**Additional context**
Add any other context or screenshots about the feature request here.

Paciente8159, 2024-03-28:
I'm not sure if I understood correctly. You need a height map feature?
The height map (bed leveling similar to 3d printing) feature is implemented.
Just need to enable it in [the configuration](https://github.com/Paciente8159/uCNC/blob/17a08e90815bf1ee58a6e25c4af573314591c0d2/uCNC/cnc_config.h#L407)

wrench1997, 2024-03-29:
> I'm not sure if I understood correctly. You need a height map feature? The height map (bed leveling similar to 3d printing) feature is implemented. Just need to enable it in [the [...]
Yes, that's right! There are instructions in the code, but it’s not clear how to select the xy axis  boundary length of the detection range.

closed 2024-03-30 by wrench1997