Paciente8159/uCNC #679: [BUG] zero after hmap results in non-zero work position

tomjnixon, 2024-04-07:
Hi,

I'm using uCNC on a custom machine, with a new board, with the code/configuration as in this branch https://github.com/tomjnixon/uCNC/tree/my_config (based on the current master version, nothing exciting.

The runtime configuration is:

<details>
<summary>runtime configuration</summary>

```json
{
  "name": "Exported settings",
  "createdBy": "tom",
  "date": "2024-04-07",
  "firmwareName": "GRBL 1.8",
  "settings": [
    {
      "key": "$30",
      "value": "1000",
      "units": "RPM",
      "description": "Maximum spindle speed. Sets PWM to 100% duty cycle.",
      "shortDescription": "Maximum spindle speed"
    },
    {
      "key": "$31",
      "value": "0",
      "units": "RPM",
      "description": "Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.",
      "shortDescription": "Minimum spindle speed"
    },
    {
      "key": "$10",
      "value": "1",
      "units": "mask",
      "description": "Alters data included in status reports.",
      "shortDescription": "Status report options"
    },
    {
      "key": "$32",
      "value": "0",
      "units": "boolean",
      "description": "Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.",
      "shortDescription": "Laser-mode enable"
    },
    {
      "key": "$11",
      "value": "0.200",
      "units": "millimeters",
      "description": "Sets how fast Grbl travels through consecutive motions. Lower value slows it down.",
      "shortDescription": "Junction deviation"
    },
    {
      "key": "$12",
      "value": "0.002",
      "units": "millimeters",
      "description": "Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.",
      "shortDescription": "Arc tolerance"
    },
    {
      "key": "$13",
      "value": "0",
      "units": "boolean",
      "description": "Enables inch units when returning any position and rate value that is not a settings value.",
      "shortDescription": "Report in inches"
    },
    {
      "key": "$0",
      "value": "33.333",
      "units": "microseconds",
      "description": "Sets time length per step. Minimum 3usec.",
[... 224 lines cut ...]
G39 X0.5 Y0.5 I41 J19 Z-2 R2.5 F50
[PRB:0.000,0.000,0.000:1]
...
[PRB:0.000,0.000,0.000:1]
[MSG:HMAP start corner;0.500;63.500]
[MSG:HMAP end corner;41.500;82.500]
[MSG:HMAP control points;16]
[MSG:HMAP;0;0;0.000]
...
[MSG:HMAP;3;3;-0.028]
```

zero again:
```
G10 P0 L20 X0 Y0 Z0
[verbose] <Idle|MPos:0.500,63.500,21.816|FS:0.000,0|Pn:P>
$10=0
[verbose] <Idle|WPos:-54.667,-25.333,1.316|FS:0.000,0|Pn:P>
```

I'd expect Wpos to be 0; this matches the position shown in UGS, and G0/G1 commands certainly don't move relative to the current position (`G0 Z2`) causes a large movement in X/Y and rapid mashing of the e-stop switch :) 

I suspected that it might be a missing `mc_sync_position` at the end of `mc_build_hmap`, but adding that doesn't seem to change the behaviour.

Thanks, let me know if there's any other information i can provide or tests to perform.

tomjnixon, 2024-04-08:
found it!

The problem is that `mc_build_hmap` does not set `target` to the final position (it uses it as a temporary instead), and this is expected by the parser:

https://github.com/Paciente8159/uCNC/blob/857f96d47742e2b4a13ab5c4183eaf924576eb08/uCNC/src/core/parser.c#L1806

It's not exactly clear to me why this doesn't happen for linear probes.

There's a fix in the branch i linked above, but it's probably not the best way.

Paciente8159, 2024-04-08:
Thanks for the pointer.
A better approach to sync all systems is to call `mc_sync_position()` to sync all subsystems.

I also noticed the retraction is not working quite as expected. It should retract at least to a height of the (initial position + R) between points and is not.

Check the #680 for a fix. Let me know if it fixed the issue.

closed 2024-04-08 by Paciente8159