Paciente8159/uCNC #684: [ask] ramps 1.4 beeper and read sd card feature

yuliusliem, 2024-04-26:
i have a question how to turn on a beeper sound on ramp full graphic controller?
and why every time i try to run gcode or gnc file from a full graphic controller nothing happen? but run perfectly from console.
i use 1.9 cnc version and other but it is still nothing happen.
last question, what is isr on pin file why some negatif and some positif?

Paciente8159, 2024-04-26:
> i have a question how to turn on a beeper sound on ramp full graphic controller? and why every time i try to run gcode or gnc file from a full graphic controller nothing happen? but run perfectly [...]

Beep is not implemented...but it should be quick to do it.

I will test that. I assume you are saying you can navigate the SD card but not run the file correct?

About the pin ISR please read [this](https://github.com/Paciente8159/uCNC/wiki/AVR:-Customizing#interrupt-on-change-feature) and see this [image](https://github.com/Paciente8159/uCNC/blob/master/docs/arduino-mega-pinout-diagram.png) for reference.
It explains how to setup input ISR on AVR

Paciente8159, 2024-04-27:
Check [this](https://github.com/Paciente8159/uCNC-modules/commit/daaea0c231b0ddab52b3924e82cd9f0c92a4e854).

I've added a beep on the display encoder press.
I also fixed the version check of SD card v2 module. Make sure you are using v2 and not the pf version of the SD card module.

I tested and I was able to run a gcode file from the display.

yuliusliem, 2024-04-27:
thank you sir, i will try it.

Paciente8159, 2024-05-20:
I'm assuming this issue is solved with the latest changes.

closed 2024-05-20 by Paciente8159