Paciente8159/uCNC #686: [BUG] Y axis endstop triggered after a period of time and hold in this position

lorddag, 2024-05-20:
**Describe the bug**
 When I first turn on the uCNC there is no endstop triggered (as it should be). I can even manually trigger all the endstops and it responds as expected. After a random period of time (motors on or off, is independent),the y axis endstop is reported as triggered and is hold in the triggered prosition after that. I have to reset the board to escape.
uCNC then returns <Alarm|MPos:0.000,0.000,0.000|FS:0.000,0|Pn:Y> all the time. 
The endstop is not triggered( verified via multimeter). X endstop and Z endstop are the same, wired the same and it does not happen with them.

**What is your hardware**
Describe your hardware:
1. What's your board: Arduino mega + Ramps 1.4
3. Paste your "Grbl/µCNC settings": Files attached.
4. If this is a custom compilation tell us what configurations you changed: No. default 

**Additional context**
[cnc_hal_reset - copia.txt](https://github.com/Paciente8159/uCNC/files/15380416/cnc_hal_reset.-.copia.txt)
[boardmap_overrides - copia.txt](https://github.com/Paciente8159/uCNC/files/15380417/boardmap_overrides.-.copia.txt)
[boardmap_reset - copia.txt](https://github.com/Paciente8159/uCNC/files/15380418/boardmap_reset.-.copia.txt)
[cnc_hal_overrides - copia.txt](https://github.com/Paciente8159/uCNC/files/15380419/cnc_hal_overrides.-.copia.txt)

lorddag, 2024-05-20:
Thank you for the help and fast response.
I also thought about noise, but the y endstop is reported as triggered all the time after that. I would expect from noise a random activation but not a constant activation. It also happens without power supply to the motors, so less noisy enviroment.
I swapped endstops and the effect is always in the y axis. 

I'll enable the commands you mentioned. I'll report the data.

Paciente8159, 2024-07-02:
Closing this issue given the time passed with no evolution.
Feel free to report it again if it happens.

closed 2024-07-02 by Paciente8159