Paciente8159/uCNC #742: Amazing! Admirable! You are still updating and maintaining it.

hanke-cnc, 2024-08-23:
I tried your firmware two years ago, and I didn't expect that you are still updating and maintaining it today.

hanke-cnc, 2024-08-23:
Two years ago, I created a controller with a 7-inch capacitive touch screen, WiFi file transfer, and 3D toolpath simulation, and sold it for only $71. It has achieved a bit of success now. Since I am not good at programming, I hope to find a partner (to whom I will pay) to maintain a closed-source version of the motion controller. I plan to use the STM32H723ZGT6. I don't know what your purpose was for creating uCNC, but if you are interested, please contact me at hankecnc@gmail.com.

closed 2024-08-25 by hanke-cnc

Paciente8159, 2024-08-25:
Hi there.
I plan to continue developing and maintaining it for as long as it is enjoyable.
uCNC is as free as it gets and I have no commercial (for now at least) aspirations for it and it goes under the GPLV3 license, meaning that although it can be used in commercial applications, the modifications to the code should be made available.
I keep this project on my (very little) spare time. I don't have the time required to develop and maintain a parallel project with comercial compromises and deadlines.

This also means that I get to keep some degree of freedoom of my time and focus.

I'm opened to extend support to new harware and features if that has a more opened and general purpose application.

Thanks.