Paciente8159/uCNC #782: Nano 328p

Juank23, 2024-11-19:
Good morning, friend, download the ucnc firmware but I get this error, it is an arduino nano328p, stay tuned
[image]

Paciente8159, 2024-11-20:
Hi there! Are you sure it's a 328p and not a 328pb?

Juank23, 2024-11-20:
Good day, I have an arduino nano 328p and one 328pu, the chip is smaller but it's the same, with grbl and vscode it goes up normal in any of the 2, here I send you a picture thank you
[image]
[image]

Juank23, 2024-11-20:
I followed your instruction and now I get the following error
[image]

Paciente8159, 2024-11-20:
The thing is that the missmatch error you are having on your 328p is very odd. I never had any issue with my 328p on my arduino UNO.

I add signature issues with the Arduino Nano I got though. Here is a picture of mu Nano.
[image]
It's actually an ATMEGA328PU.

This is what I did to get it to allow upload:

1. I installed MiniCore on Arduino IDE
2. Then with the help of my regular Arduino UNO (serving as a programmer) I installed the MiniCore bootloader on the 328PU.
3. After that I used the configuration on VSCode for the atmega328pb variant and all went as expected.

Juank23, 2024-11-20:
See now what it does
[image]

Paciente8159, 2024-11-20:
In that case (and since you now installed the MiniCore (urboot) try to change the board to Atmega328P to see if it works.

Like this. Change from this:

[image]

to this

[image]

Juank23, 2024-11-21:
I followed all your instructions but it didn't work. 
[image]

Paciente8159, 2024-11-21:
I've modified the avr.ini a bit with some more clarifications about the subject.
You can look at it [here](https://github.com/Paciente8159/uCNC/pull/784).

I will also update the wiki.

Juank23, 2024-11-21:
Good morning, thanks for your help, I did everything I did but nothing worked me indiscaste, I did several tests without good results.... to the last I changed the bootloader for the arduino one and worked well, now I have another question, starts to work well but at the end of the gcode is in a state of pause and you have to cancel to start again here a screenshot ...
[image]

Paciente8159, 2024-11-21:
Control pins (Emergency stop, hold, and cycle start/resume pins) are defined by default and have a weak pull-up enabled by default. That means you need to pull them down to ground.

If you don't need them, then you can simply disable them. To do that depends on what method you are using to configure the project. Did you use the web-builder or you manually configured the project).

If you configured manually you can [disable control pins here](https://github.com/Paciente8159/uCNC/blob/3c88a955137fb40182d138bf87c0364bad32699f/uCNC/cnc_config.h#L574)

Juank23, 2024-11-21:
my friend I'm doing it manually in vscode, but I followed your instructions and still the same, I just got a couple of plates that I bought stm32f401 I uploaded the code without probelma but it does exactly the same, it starts the job well but when finished it stays in that state until it is given to restart, the strange thing is that at the end of the gcode I remove M30 and ends the process to continuucacion I send you the example gcode thank you

Translated with www.DeepL.com/Translator (free version)
[1.zip](https://github.com/user-attachments/files/17850383/1.zip)

Paciente8159, 2024-11-21:
Ahhh that.
That is the M2 or M30 behaviour. When you send a M2 or M30 the machine will enter lock mode preventing further code from running. This is to prevent accidental code execution after a program finnishes.

You do not need to reset the board. An unlock command ($X) will unlock the machine.

Juank23, 2024-11-21:
is there any way to remove that function in the code so that it terminates the process without having to use that command ($x) thanks

Paciente8159, 2024-11-21:
I've added [an option in the cnc_config.g file](https://github.com/Paciente8159/uCNC/pull/785).
Check line 327. Patch will be merged in master in a few minutes.

Juank23, 2024-11-21:
I copied all the content but it is still the same
[image]
[image]

Paciente8159, 2024-11-21:
Ok. Will check it tomorrow.
Thanks

Juank23, 2024-11-21:
Friend, thank you very much

Paciente8159, 2024-11-22:
I [found this](https://github.com/grbl/grbl/issues/681)

I don't have a board with me now but in a few hours I will flash Gnea/Grbl an check what is the default behavior for M2.
I will modify the option to make it compatible. Changes to the default µCNC might also be applied.

Paciente8159, 2024-11-22:
I've tweaked M2/M30 a bit to match LinuxCNC behavior in #786 .
I prefer to leave µCNC default behavior to lock the machine, ignoring all code after M2/M30 until an unlock command is issued.
You can emulated Grbl command by enabling [option `DISABLE_ENDPROGRAM_LOCK` in the cnc_config.h file](https://github.com/Paciente8159/uCNC/blob/9adf983bf4ed40feafbb03b5798f1f2574cdbe6a/uCNC/cnc_config.h#L327)

Paciente8159, 2024-11-22:
[3 Files were modified.](https://github.com/Paciente8159/uCNC/pull/786/files)
Just download the master branch again. It's better that way. Less error prone.

Paciente8159, 2024-11-23:
I will take a look and report you back

Paciente8159, 2024-11-23:
Ok I've detected a problem were the parser was not updating the feed and spindle correctly if modified in the canned cycle block.
[Here is the fix. Please try the code in this branch and let me know how it goes.](https://github.com/Paciente8159/uCNC/tree/canned-cycle-feed-spindle-update)

Not that your GCode is also incorrect.
```
N190 G80
N200 X0. Y-10.639
```

[According to Linux CNC it's an error to program any axis words with G80 active.](https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g80)
The correct code should be:

```
N190 G80
N200 G0 X0. Y-10.639
```

Juank23, 2024-11-26:
good day, I modified the gcode as you indicated but the crazy movements continue, I had not noticed that the motorers are always hooked and I tried with $4=1 it unhooks them but they do not work they remain off thanks

Paciente8159, 2024-11-27:
I compared to the motions performed with LinuxCNC and the G82 seems to match.
Note that the code you send yields several errors on LinuxCNC and does not comply with the RS274NGC standard.
One of the issues as stated are the axis commands with the G80 motion mode active.
I'm pushing the PR. If you encounter any issues let me know.

closed 2024-11-27 by Paciente8159