Paciente8159/uCNC #798: Homing single axis

MarvD97, 2025-01-08:
I am currently switching from GRBL to µCNC. In GRBL, one can enable single axis homing, so that using the command `$HX` only the x-Axis is homed which is important in my application. 
I could not find such an option in µCNC. Is it available or is there a workaround?

Paciente8159, 2025-01-08:
This feature is not available but it's possible to add it by creating an [custom grbl command extension module](https://github.com/Paciente8159/uCNC-modules/blob/master/Examples/extended_grbl_cmd.c)

It should not be too hard. I'm a bit tight with work now but I might be able to pull that off during the weekend.

Paciente8159, 2025-01-11:
[This](https://github.com/Paciente8159/uCNC-modules/pull/87) still needs propper testing. I may not have time to test it this weekend, but should be something in these lines.

MarvD97, 2025-01-13:
Thank you, I am currently having some hardware issues but I will test it soon with my hardware.

Paciente8159, 2025-01-14:
I've reviewed the module and made the necessary fixes.
Tested on the virtual environment and seems to be working fine.
Let me know if you find any issue.

MarvD97, 2025-01-15:
I've tested it on my hardware (5 axis) and it worked without any issues. Thanks!

Paciente8159, 2025-01-15:
Ok. I'm closing this as solved with [https://github.com/Paciente8159/uCNC-modules/pull/87](https://github.com/Paciente8159/uCNC-modules/pull/87)
Let me know if you run into any issue.

closed 2025-01-15 by Paciente8159