Paciente8159/uCNC #806: [BUG] Strange build for AVR-UNO

gwiethaus, 2025-01-18:
**Describe the bug**
When compiling for Arduino UNO, it generates several errors such as undeclared constant or undeclared function.

**What is your hardware**
Describe your hardware:
1. Arduino UNO and CNC shield v3
2. uCNC Version 2025-01-16
3. Windows 10
4. Visual Studio Code / Platformio

**Additional context**
I used the uCNC configuration generator from the website. Attached are the files and versions that were used.
Also attached is the terminal response file with the errors.

[uCNC config.zip](https://github.com/user-attachments/files/18462453/uCNC.config.zip)

[uCNC-master_20250116.zip](https://github.com/user-attachments/files/18462455/uCNC-master_20250116.zip)

[Terminal_log.txt](https://github.com/user-attachments/files/18462457/Terminal_log.txt)

[platformio.zip](https://github.com/user-attachments/files/18462461/platformio.zip)

[Folder Structures.zip](https://github.com/user-attachments/files/18462471/Folder.Structures.zip)

Paciente8159, 2025-01-18:
Hi there.
It's not a bug. You simply have incomplete definitions of the PWM pins. Also note that you are trying to use PWM were your don't have hardware PWM pins available.
Look at the [UNO Pinout](https://github.com/Paciente8159/uCNC/blob/master/docs/arduino-uno-pinout-diagram.svg)
The only pins that can do PWM are 3, 5, 6, 9, 10, and 11.

Paciente8159, 2025-01-20:
There was an issue on the Web Builder (for AVR specifically) that prevented the PWM channel to be correctly loaded from the default configuration. This issues was fixed yesterday.