Paciente8159/uCNC #811: [BUG] Bad CMake Executable

gwiethaus, 2025-01-20:
**Describe the bug**

Sorry for my English
After updating and generating the new version of uCNC using the website to generate the config, when compiling for Arduino UNO, an error was generated in CMAKE with the message described in the screenshot. And CMake works properly for other programs.

**What is your hardware**
1. Arduino UNO
2. uCNC Builder [ucnc_build.json](https://github.com/user-attachments/files/18482836/ucnc_build.json)
3. Visual Studio code / Platformio 3.3.4
4. Last Version uCNC 2025-01-20

What am I doing wrong now?

**Screenshots**

[image]

[image]

[image]

Paciente8159, 2025-01-20:
Thanks. I'll take a look tomorrow

Paciente8159, 2025-01-21:
Ok several notes on your problem.

First you still had the PWM0 pin definition incomplete.

Next although you replaced the configuration files, you still need to select the correct environment for PlatformIO.
You need to click here and select the correct environment (you are using the Default that tries to compile for several different boards)

[image]

Now about the UNO board. The program size for that MCU is near the limits of what it can hold. By turning on backlash compensation you will need to sacrifice something else, or the code will simply not fit in the MCU flash.

Either you sacrifice the control pins (ESTOP, HOLD and RESUME) and probing (for example), or you need to sacrifice something else like storing the settings in the EEPROM (that means at power up you need to reload the settings again with a gcode file or set the default/reset settings to match your needs.

gwiethaus, 2025-01-21:
Hello! Thanks for your reply. I don't think I'll need ESTOP for emergency stop. I will need EEPROM! When the messages shown on the terminal, they didn't seem like a lack of memory, because otherwise it would indicate this at the time of compilation. I had already done this, for example, on the ESP32 WROOM-32 and there was a lack of memory, but that's not the case for the uCNC.

First of all, I'll test your suggestion directly on the Arduino Uno to see if there's a problem.

I'll do an experiment by disabling all ESTOP, RESUME, HOLD and see what happens and then I'll report back here with the result.

Regarding "First you still had the PWM0 pin definition incomplete": Now I understand what was missing. I needed to link it to a channel and a timer.

Can I disable E-Stop and enable the Probe to get the amount of memory needed?

Thank you very much for your help. I'm also trying to contribute and note bugs here if they happen to exist. It's a little difficult to understand some things, because I was used to using Grbl for Arduino UNO and here it's independent hardware, so some things change.

After everything is ready in uCNC I'll share it with some people in a CNC assembly group here in Brazil and demonstrate the versatility of uCNC in relation to the old Grbl. There is life beyond Mach3.

Thank you very much and I'll wait

Paciente8159, 2025-01-21:
Um abraço para o Brasil.

Yes in (most) MCU's you need to set the PWM pin respective channel and timer. Depends from MCU to MCU but that is usually the case.

You can disable E-STOP or the probe only if you simply undefine them. But that may not be enough. You can also define some pins by groups (like control pins, limits switch pins and the probe pin like I did). You can play around with the configuration until you get one that compiles withing the flash size of the Arduino UNO.

By default the UNO board is the only one that has some advanced options already enabled just to allow the firmware to fit on it. For example on laser machines that only need X and Y axis, disabling Z axis also results in some code shrinking.