Paciente8159/uCNC #813: [FR] New Boards ESP32 S3 e ESP32 Basic

gwiethaus, 2025-01-21:
Hello, could you implement the configurations for ESP32-S3 Wroom-1 M16R8 development boards and for the basic ESP32 Wroom-32, which are widely used? I believe it will not be difficult to do, as it already has something like the ESP32 Wemos D1 R32.

Boards:
- ESP32-S3 Wroom-1 M16R8 or M8R8
- ESP32-C3 and C2
- ESP32 Wroom-32

Paciente8159, 2025-01-21:
Well, it should be possible to compile code for those board using the web builder and adjusting a couple of files. I'll show you an example tomorrow. 

My only doubt is if the code works correctly on the board or not. I only have esp32 wroom base boards.

gwiethaus, 2025-01-23:
Thank you @Paciente8159. Later I would like to know how to do this. I believe that if you test it with the ESP32 wroom-32 it should give you an idea of ​​how to do it for other boards in the ESP32 line. You only need to change the PIO pins. I think it can be adjusted and defined using the Web Builder.

Thank you for your willingness to help.

Paciente8159, 2025-01-23:
I tested some custom builds and it's not that easy. With some small adjustments ESP32-S3 can be configured. ESP32-C3 will require more refactoring.
I guess the best approach is to treat these as different MCU families (as they actually are).
That will require adding those architectures to the core code. I'll add this to the TODO list.

gwiethaus, 2025-01-24:
The work you are doing is really cool (@Paciente8159 ). It will help a lot of people. I know there must be a lot of implementations and improvements to be made, but with patience and care you will be able to do it. This work will leave a legacy. I believe it will help me in another project. Implementing this on an ESP32 and the graphical interface on another ESP32-S3. I want to develop a standalone later on, something similar to DDCSV 4.1 (http://www.ddcnc.com/?m=home&c=View&a=index&aid=186&lang=en), but for this it is necessary that uCNC has good support for the devices and more advanced resources. I believe that with time this will become a reality.

By the way, yesterday I tested the suggestions you gave me, it seems to have worked. I adjusted it in Web Builder (I only turned off the eSTOP, HOLD, RESUME features, but left the Probe on. The end of stroke and Homing were activated) and then I transferred everything to Visual Studio Code and it compiled fine. I also turned on the spindle speed and direction features.

I haven't tested it on the real machine yet, but I did some tests on the Arduino Uno with CNC Shield. I tested the outputs with the digital multimeter and it was working.

I'm trying to understand now about the Arduino Probe, which I think is a short circuit between the SCL and RST pins of the CNC shield. But it's still confusing when I use the gSender software. It seems to have some instability.

Thank you very much and hugs

closed 2025-09-17 by Paciente8159