Paciente8159/uCNC #823: Unable to Access Web Interface on µCNC v1.10.2 with ESP8266

zenggood, 2025-02-02:
I am writing to seek your assistance regarding an issue I am encountering with µCNC v1.10.2 on an ESP8266.

I have successfully compiled and uploaded the firmware to my ESP8266, and it runs perfectly. However, I am unable to access the web control interface. I have tried various build configurations, but none of them have resolved the issue. Additionally, I tested the precompiled firmware provided for the ESP8266, but the web interface still does not load.

Paciente8159, 2025-02-02:
Hi there.
Did you build the file system image and uploaded it?

[image]

zenggood, 2025-02-02:
Ah, I missed this. Thanks so much!