Paciente8159/uCNC #874: [FR]Add arc interpolation to axis A.

zenggood, 2025-07-07:
Is it easy to implement circular arc interpolation for the A axis?

zenggood, 2025-07-07:
Without any kinematics, it's a standard 4-axis CNC. The A-axis cannot achieve arc interpolation like the X, Y, and Z axes. When the A-axis has arc motion, it produces strange cutting paths.

Paciente8159, 2025-07-07:
The implementation of A, B, C axis on µCNC is done like actual rotational axis. So for example rotating the A axis to position/angle 90º should be something like A90.
The implementation for arc interpolations (G2 and G3) is implemented according to the RS274NGC and Linux CNC (using plane selection G17-XY/G18-XZ/G19-YZ). So there is no implementation for A/B/C axis as these are rotational axis.

If I'm missing something, let me know and show me an example of what you are trying to implement.

Paciente8159, 2025-07-09:
> When we run below command
> 
> G3X31.6000Y66.0000R20.0000A180.0
> 
> x and y axis move to X31.6000Y66.0000 after completion of this A axis move 180 step
> 
> When we use MACH3 using same G code A axis also move simultaneously i.e XYA move simultaneously

I'll investigate this. Thanks for the report.

Paciente8159, 2025-07-09:
I've simulated G2/G3 commands on X/Y plane with an A angle and this seems to be working as expected.
I've configured a simple cartesian 4axis machine. And from the origin ran a (G2 or G3) G2X20Y20R20A180.
The A was being interpolated with the arc.
Which microcontroller are you running with this? (I can try to see if this is a compilation/optimization issue)

Paciente8159, 2025-11-07:
Due to lack of activity I will close this issue once #892 is integrated.

zenggood, 2025-11-22:
> $$
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=99          (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=11          (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=1           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=2          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=0          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=0          (Enables homing cycle. Requires limit switches on all axes.)
< $23=3          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=200.000    (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=2000.000   (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=1.000      (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=1000       (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=78.740    (X-axis travel resolution in steps per millimeter.)
< $101=78.740    (Y-axis travel resolution in steps per millimeter.)
< $102= 116.000   (Z-axis travel resolution in steps per millimeter.)
< $103= 53.300  (A-axis travel resolution in steps per degre.)
< $110=3000.000  (X-axis maximum rate. Used as G0 rapid rate.)
< $111=3000.000  (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=2900.000  (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=1900.000  (A-axis maximum rate. Used as G0 rapid rate.)
< $120=1000.000  (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=1000.000  (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=600.000   (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=600.000   (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=300.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=300.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=300.000   (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=300.000   (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok